arendst/Tasmota #6456: MCP23017 IO & Hue Emulation

BBBits, 2019-09-19:
Can the MCP23017 IO be included in Hue Emulation Device Discovery ?

Jason2866, 2019-09-19:
At the moment no.

BBBits, 2019-09-19:
Ok, thanks.
I see that the max hue devices is limited to 15.
is this a limitation of the device hue emulation type or just a decision made for some reason ?

Jason2866, 2019-09-19:
@s-hadinger  Can you answer the question from @BBBits ?

s-hadinger, 2019-09-19:
This limit of 15 is completely arbitrary. However the max number of Relays+ light is lower. Changing this number requires also to change the Hue ID assignment. Do you see this limit as an issue?

BBBits, 2019-09-19:
If support was to be added for mcp23017 , that would add up to 16 for each address.

s-hadinger, 2019-09-23:
It looks doable to extend to 31 devices. Internal settings shows that `power` is coded on 32 bits so it should be able to support 32 Relays. 

You would still need to add mcp23017 support

s-hadinger, 2019-09-23:
I looked at it, it's easy to increase the number of devices controlled by Hue emulation to 32. Beyond that number it breaks other parts of the code. I will post a small PR in the following days.

s-hadinger, 2019-09-23:
@BBBits I posted a PR to increase Hue devices to 32.

MCP23017 doc is here: https://github.com/arendst/Sonoff-Tasmota/wiki/MCP23008-MCP23017

However, I think it is still missing the matching between Tasmota Relays and the sensor29 command. I'm not sure you can do this with rules or scripting.

closed 2019-09-23 by arendst

BBBits, 2019-09-23:
Cool , Thanks .
Dunno If I have all the skills to address the hue integration of the MCP but will certainly see If I can figure it out.

Jason2866, 2019-09-24:
Isnt a friendly name needed for each relay (output)? So there will be the next restriction...

s-hadinger, 2019-09-24:
It used to be a limit, but it changed when `SetOption68` was introduced. The first 4 devices take their names from friendlyname[1..4]. The 5th and above are named after <friendlyname4>-0<01..xx>.

But indeed there is a problem when there are more than 14 devices, it will use characters after '9' which may cause issues.

Good catch @Jason2866 ! On my TODO list

Jason2866, 2019-09-26:
Now it is up to you @BBBits to write rules or do it via scripter engine.

BBBits, 2019-09-27:
eek .. stumped.
can we create rules that generate hue devices ?

Jason2866, 2019-09-27:
@thxthx0 can you help? You have scripter experience :-)

thxthx0, 2019-09-27:
I'm afraid not nearly experienced enough... ;)

I agree with s-hadinger:
> still missing the matching between Tasmota Relays and the sensor29 command...

You could define virtual relays and control the MCP23017 outputs according via script,
so that they'll be discovered by HUE Emu - 
but there will remain the limit of 8 relays in Tasmota.

I think if anyone can help, @gemu2015 is the right address :)

gemu2015, 2019-09-28:
send sensor29 cmds on change of a power state of a tasmota defined relay

\>D
; relais 1..N power state
p1=0
p2=0
etc

\>E
; get power state
p1=pwr[1]
; if there is a change
; then set appropriate output pin
if chg[p1]>0
then
if p1>0
then
->sensor29 0,ON
else
->sensor29 0,OFF
endif
endif
etc

the fanciest solution would be to have a scripter section >H
where you could define virtual hue devices and assign them to script variables
i looked into it. it is doable but a considerable amount of work.

s-hadinger, 2019-09-28:
Thanks @gemu2015 

Hue devices follow the `devices_present` variable, which gives the overall controllable devices. This variable is computed in `GpioInit()` in `sonoff.ino`. I'm not sure whether you can change this variable with scripting **before** Hue or lights are initialized.

Jason2866, 2019-09-30:
Thanks @gemu2015 for implementing!!

See Example https://github.com/arendst/Sonoff-Tasmota/pull/6522#issuecomment-536589939

PR #6478 (2019-09-23): Support up to 32 devices for Hue emulation
## Description:

Raise the limit from 15 to 32 devices adressable via Hue emulation. This allows to use the full range of `power_t` devices (32 bits).

**Related issue (if applicable):** fixes #6456

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept  [... cut]
PR #6494 (2019-09-25): Minor fixes to Hue Emulation and Light
## Description:

Two minor fixes
- Fix wrong rounding when reporting `Channel` value (from Discord by @jziolkowski )
- Fix wrong friendly name with Hue Emulation above 13 devices

**Related issue (if applicable):** fixes #6456 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests [... cut]