arendst/Tasmota #6457: Digoo DG-PS01 smart power strip 3xplug+4xUSB with individual plug control

tlc76, 2019-09-19:
Hey all,
I am planning to get [one of these](https://www.banggood.com/DIGOO-DG-PS01-Smart-Power-Strip-Work-with-Alexa-Smart-Home-USB-Port-Remote-Control-Power-Outlet-p-1287111.html), but before I hit the "Pay" button: does anyone know if this power strip supports Tasmota? I googled it before but could not find any reference to Tasmota...
Thanks,
Cristian

Jason2866, 2019-09-19:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

schrej-zz, 2019-09-20:
Hi,
it looks like the one, that I have (Meross?), but just another brand.
I flashed mine with Tasmota without any issues.

andrethomas2, 2019-09-20:
Please note that GitHub Issues are for reporting bugs and requesting additional features.

Please search the Wiki for the information you require or visit Discord Chat where general questions and answers are handled.

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-20 by andrethomas2

dgomes, 2019-09-21:
@schrej-zz would kindly share the template you use ?

schrej-zz, 2019-09-22:
There are several templates already in the Tasmota Template DB.
I used this one, but there are several others and at least one of them should work.
https://blakadder.github.io/templates/woox_R4028.html
Joerg