arendst/Tasmota #6465: Blitzwolf SHP5 or Gosund SP112 - Case opening

DoraDeu, 2019-09-20:
I already opend the case of the plug but I have no idea how to get the PCB from the lower part of the housing. On one side it comes a little bit up but even with some force I don’t get it off.
Could somebody tell me the trick?

meingraham, 2019-09-20:
Please close this issue and search online.

andrethomas2, 2019-09-20:
Please note that GitHub Issues are for reporting bugs and requesting additional features.

Please search the Wiki for the information you require or visit Discord Chat where general questions and answers are handled.

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-20 by andrethomas2

atze09, 2023-11-14:
> I already opend the case of the plug but I have no idea how to get the PCB from the lower part of the housing. On one side it comes a little bit up but even with some force I don’t get it off. Could somebody tell me the trick?

I also only found this one,
https://www.tasmota.info/sp112-flash/

But there is only 1 screw and no screws inside the contacts.
Did you succeed in opening it?