arendst/Tasmota #6467: Unable to change ButtonTopic over mqtt on 6.6.0 version

junk-barkass, 2019-09-20:
i'm unable to change the buttontopic over mqtt. 
All other works that i changed. 
Tried like this; 
mosquitto_pub -t "cmnd/106/ButtonTopic" -m "knapp106" -u user -P password

It works if using curl to change it. 
Also works on console (webpage)

junk-barkass, 2019-09-21:
What do you need more? I do think just by sending the mosquito request will show the problem. I do not have special configuration. 

When sending
mosquitto_pub -t "cmnd/106/ButtonTopic" -m "knapp" -u user -P password
it responds on web console like this, this is not correct behavior.

21:57:20 MQT: stat/106/RESULT = {"ButtonTopic":"KNAPP106"}

meingraham, 2019-09-21:
@junk-barkass 

This pretty much explains the information we deem important to initially understand how your device is set up. We've been doing this for a while which is why we believe that information is needed.

https://github.com/arendst/Sonoff-Tasmota/blob/development/.github/ISSUE_TEMPLATE/Bug_report.md

junk-barkass, 2019-09-21:
Sorry newbie here by sending an error.. Hope doing better and better.
Compile by atom by myself.

22:43:04 CMD: backlog
22:43:04 MQT: stat/106/RESULT = {"Backlog":"Empty"}
22:43:14 CMD: backlog template
22:43:14 MQT: stat/106/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:43:30 CMD: module
22:43:30 MQT: stat/106/RESULT = {"Module":"1 (Sonoff Basic)"}
22:43:58 CMD: status 0
22:43:58 MQT: stat/106/STATUS = {"Status":{"Module":1,"FriendlyName":["Guldlampan"],"Topic":"106","ButtonTopic":"KNAPP106","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"106","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:43:58 MQT: stat/106/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"106","OtaUrl":"http://insidan.us.to/sonoff/firmware106.bin","RestartReason":"Software/System restart","Uptime":"2T03:26:09","StartupUTC":"2019-09-19T18:17:49","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":38,"SaveAddress":"F6000"}}
22:43:58 MQT: stat/106/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-19T20:08:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:43:58 MQT: stat/106/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["lj3m",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
22:43:58 MQT: stat/106/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A1","22B617CD","01003BC0"]}}
22:43:58 MQT: stat/106/STATUS5 = {"StatusNET":{"Hostname":"106-2472","IPAddress":"192.168.200.217","Gateway":"192.168.200.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.200.1","Mac":"84:0D:8E:70:09:A8","Webserver":2,"WifiConfig":4}}
22:43:58 MQT: stat/106/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.200.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7009A8","MqttUser”:”user","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:43:58 MQT: stat/106/STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 21 21:43:58 2019","Local":"Sat Sep 21 22:43:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:45","Sunset":"18:05"}}
22:43:58 MQT: stat/106/STATUS10 = {"StatusSNS":{"Time":"2019-09-21T22:43:58"}}
22:43:58 MQT: stat/106/STATUS11 = {"StatusSTS":{"Time":"2019-09-21T22:43:58","Uptime":"2T03:26:09","Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"1","Wifi":{"AP":1,"SSId":"lj3m","BSSId":"E4:F0:42:E6:0C:E2","Channel":1,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:06"}}}
22:45:56 CMD: gip
22:45:56 MQT: stat/106/RESULT = {"GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)”}

Turned on weblog4

22:51:33 RSL: Received Topic cmnd/106/ButtonTopic, Data Size 5, Data knapp
22:51:33 RSL: Group 1, Index 1, Command BUTTONTOPIC, Data knapp
22:51:33 MQT: stat/106/RESULT = {"ButtonTopic":"KNAPP106”}

Best Regards Magnus

meingraham, 2019-09-22:
Please update to the latest firmware version to ensure that this problem is still present in the latest version.

`Backlog OTAURL http://thehackbox.org/tasmota/020300/sonoff.bin; Upgrade 1`

Allow the device to update to minimal and then to the latest version. It will reboot twice to do so. Please do not interrupt it until the update is completed.

junk-barkass, 2019-09-23:
> 21 sep. 2019 kl. 23:53 skrev magnus larsson <junk.barkass@gmail.com>:
>
> Sorry newbie here by sending an error.. Hope doing better and better.
> Compile by atom by myself.
>
>
> 22:43:04 CMD: backlog
> 22:43:04 MQT: stat/106/RESULT = {"Backlog":"Empty"}
> 22:43:14 CMD: backlog template
> 22:43:14 MQT: stat/106/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
> 22:43:30 CMD: module
> 22:43:30 MQT: stat/106/RESULT = {"Module":"1 (Sonoff Basic)"}
> 22:43:58 CMD: status 0
> 22:43:58 MQT: stat/106/STATUS = {"Status":{"Module":1,"FriendlyName":["Guldlampan"],"Topic":"106","ButtonTopic":"KNAPP106","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"106","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 22:43:58 MQT: stat/106/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"106","OtaUrl":"http://insidan.us.to/sonoff/firmware106.bin <http://insidan.us.to/sonoff/firmware106.bin>","RestartReason":"Software/System restart","Uptime":"2T03:26:09","StartupUTC":"2019-09-19T18:17:49","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":38,"SaveAddress":"F6000"}}
> 22:43:58 MQT: stat/106/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-19T20:08:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
> 22:43:58 MQT: stat/106/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["lj3m",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
> [... cut ...]
> 22:45:56 MQT: stat/106/RESULT = {"GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"0 (None)”}
> [... 3 lines cut ...]
> 22:51:33 RSL: Received Topic cmnd/106/ButtonTopic, Data Size 5, Data knapp
> 22:51:33 RSL: Group 1, Index 1, Command BUTTONTOPIC, Data knapp
> 22:51:33 MQT: stat/106/RESULT = {"ButtonTopic":"KNAPP106”}
>
> Best Regards Magnus

Done the upgrade. 

10:11:13 CMD: weblog 4
10:11:13 MQT: stat/106/RESULT = {"WebLog":4}
10:11:13 CFG: Saved to flash at F9, Count 49, Bytes 3584

10:11:19 CMD: module
10:11:19 SRC: WebConsole from 192.168.200.41
10:11:19 RSL: Received Topic /module, Data Size 0, Data 
10:11:19 RSL: Group 0, Index 1, Command MODULE, Data 
10:11:19 MQT: stat/106/RESULT = {"Module":"1 (Sonoff Basic)"}
10:11:28 WIF: Checking connection...
10:11:28 WIF: Connected
10:11:36 CMD: backlog template
10:11:36 SRC: WebConsole from 192.168.200.41
10:11:36 RSL: Received Topic /backlog, Data Size 8, Data template
10:11:36 RSL: Group 0, Index 1, Command BACKLOG, Data template

10:11:36 SRC: Backlog
10:11:36 RSL: Received Topic /template, Data Size 0, Data 
10:11:36 RSL: Group 0, Index 1, Command TEMPLATE, Data 
10:11:36 MQT: stat/106/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:11:48 WIF: Checking connection...
10:11:48 WIF: Connected

10:12:46 CMD: status 0
10:12:46 SRC: WebConsole from 192.168.200.41
10:12:46 RSL: Received Topic /status, Data Size 1, Data 0
10:12:46 RSL: Group 0, Index 1, Command STATUS, Data 0
10:12:46 MQT: stat/106/STATUS = {"Status":{"Module":1,"FriendlyName":["Guldlampan"],"Topic":"106","ButtonTopic":"KNAPP106","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"106","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:12:46 MQT: stat/106/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"106","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:01","StartupUTC":"2019-09-23T09:05:45","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":49,"SaveAddress":"F9000"}}
10:12:46 MQT: stat/106/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-19T20:08:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:12:46 MQT: stat/106/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["lj3m",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
10:12:46 MQT: stat/106/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A1","22B617CD","01003BC0"]}}
10:12:46 MQT: stat/106/STATUS5 = {"StatusNET":{"Hostname":"106-2472","IPAddress":"192.168.200.217","Gateway":"192.168.200.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.200.1","Mac":"84:0D:8E:70:09:A8","Webserver":2,"WifiConfig":4}}
10:12:46 MQT: stat/106/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.200.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7009A8","MqttUser":"scriptuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:12:46 MQT: stat/106/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 23 09:12:46 2019","Local":"Mon Sep 23 10:12:46 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:49","Sunset":"18:00"}}
10:12:46 MQT: stat/106/STATUS10 = {"StatusSNS":{"Time":"2019-09-23T10:12:46"}}
10:12:46 MQT: stat/106/STATUS11 = {"StatusSTS":{"Time":"2019-09-23T10:12:46","Uptime":"0T00:07:01","Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"0","Wifi":{"AP":1,"SSId":"lj3m","BSSId":"E4:F0:42:E6:0C:E2","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:03"}}}
10:12:48 WIF: Checking connection...
10:12:48 WIF: Connected

Sent mosquito_pub ..

10:14:42 RSL: Received Topic cmnd/106/ButtonTopic, Data Size 5, Data knapp
10:14:42 RSL: Group 1, Index 1, Command BUTTONTOPIC, Data knapp
10:14:42 MQT: stat/106/RESULT = {"ButtonTopic":"KNAPP106”}

it should no have responded RUSULT knapp

//Magnus

meingraham, 2019-09-23:
@arendst 

Can you review this one. After setting the ButtonTopic to "KNAPP106” (via Console), it cannot be set to "knapp" via MQTT.

arendst, 2019-09-23:
No problem at all:
```
15:40:36 CMD: buttontopic KNAPP106
15:40:36 SRC: Serial
15:40:36 MQT: stat/wemos4/RESULT = {"ButtonTopic":"KNAPP106"}
15:40:36 CFG: Saved to flash at FB, Count 399, Bytes 4096
15:40:44 CMD: buttontopic knapp
15:40:44 SRC: Serial
15:40:44 MQT: stat/wemos4/RESULT = {"ButtonTopic":"knapp"}
15:40:44 CFG: Saved to flash at FA, Count 400, Bytes 4096
15:40:56 CMD: buttontopic 0
15:40:56 SRC: Serial
15:40:56 MQT: stat/wemos4/RESULT = {"ButtonTopic":""}
15:40:57 CFG: Saved to flash at F9, Count 401, Bytes 4096
15:41:23 CMD: buttontopic KNAPP106
15:41:23 SRC: Serial
15:41:23 MQT: stat/wemos4/RESULT = {"ButtonTopic":"KNAPP106"}
15:41:24 CFG: Saved to flash at F8, Count 402, Bytes 4096
15:41:49 SRC: MQTT
15:41:49 MQT: stat/wemos4/RESULT = {"ButtonTopic":"knapp"}
15:41:49 CFG: Saved to flash at F7, Count 403, Bytes 4096
15:42:01 HTP: Main Menu
15:42:02 HTP: Main Menu
15:42:04 HTP: Console
15:42:16 CMD: buttontopic KNAPP106
15:42:16 SRC: WebConsole from 192.168.2.1
15:42:16 MQT: stat/wemos4/RESULT = {"ButtonTopic":"KNAPP106"}
15:42:17 CFG: Saved to flash at F6, Count 404, Bytes 4096
15:42:21 SRC: MQTT
15:42:21 MQT: stat/wemos4/RESULT = {"ButtonTopic":"knapp"}
15:42:21 CFG: Saved to flash at F5, Count 405, Bytes 4096
```
What is your ``fulltopic``

arendst, 2019-09-23:
I just verified your issue! but...

You have a special case where both grouptopic and topic are set to the same topic: 106.

The issue here is that in that case MQTT commands are recognised as group commands. Some  commands are not allowed as group command as it would make the functionality of several units invalid. 

The following commands are not allowed when executed using a grouptopic as they have to be unique in one environment:
- Hostname
- MqttClient
- Topic
- ButtonTopic

Your solutions is to change GroupTopic to a different topic than any Topic. That's also where GroupTopic becomes usefull; when several units have the same GroupTopic you can control these units with just one command EXCEPT for the ones just noted,

In addition to Buttontopic

meingraham, 2019-09-23:
@arendst 

For those four commands, is this always the case? For example, if I want to restore defaults for all devices in a group. It is not possible to issue `cmnd/<grouptopic>/Hostname 1`. Or does this check only apply when setting these four device parameters to a specific value?

Mike

P.S. I'm going to add a comment in the wiki and want to make a distinction if this is the case.

junk-barkass, 2019-09-23:
Thank you so much!! Now I understand it. i Havh now changed group topic for all my devices will use it for upgrade purpose.
mark this as solved.
Best Regards Magnus

arendst, 2019-09-23:
@meingraham Clever Dicky! Alas it is always the case.

meingraham, 2019-09-23:
Makes the wiki edits easier ;-)

junk-barkass, 2019-09-23:
Im using switch topic for my ubuntu server to know witch device the request come from.

meingraham, 2019-09-23:
Understood. So, if a user doesn't have parameter replacements (%xxxx%) in the specified `FullTopic` value, should it behave like the other "topics" and not execute?

arendst, 2019-09-23:
@meingraham no, currently there are no checks for grpflag on either SwitchTopic nor FullTopic.

If a user wants to group change a FullTopic without using %prefix% and/or %topic% it's his choice he'll messes up his environment. I won't add checks for that. Too many situations to check.

I'll add the grpflag to SwitchTopic now.

junk-barkass, 2019-09-23:
> @junk-barkass <https://github.com/junk-barkass> not an issue here.

i think so yes, for not mess up peoples mind that’s the best to do so. -Do nothing.

meingraham, 2019-09-23:
@arendst - wiki now reflects as discussed.

junk-barkass, 2019-09-23:
You are doing a great job!! love Tasmota so much.

junk-barkass, 2019-09-23:
Great!!

closed 2019-09-23 by arendst