arendst/Tasmota #6470: SONOFF (e.g. TH10): switch control via temp (internal)

g6094199, 2019-09-21:
**Describe the solution you'd like**  

He!
Would it be possible to have  internal threshold settings for the temp in the web-if (like the timer setting) where we can set to switch on/off the power socket on special temperatures? Most of the time i use the sonoff without/out of reach of my openhab. 

Sash

meingraham, 2019-09-21:
Please always check the wiki before opening an issue. If this device has internal temperature monitoring, this is already possible. Check you device specifications.

Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

andrethomas2, 2019-09-21:
Closing this issue as it has been answered. Please refer to additional links for support provided in previous post(s).

closed 2019-09-21 by andrethomas2