arendst/Tasmota #6471: Timer ignores checkboxes

bakman2, 2019-09-22:
### BUG DESCRIPTION
Timer setup to turn on at 02:00 and turn off at 05:00 every _other_ day.
Checkboxes are being ignored and it fires every day.

On:
[image]

Off:
[image]

See grafana output, turns on daily.
[image]

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW2
- [x] Tasmota binary firmware version number used: 6.6.0
  - [x] Pre-compiled

- [x] Flashing tools used: nodemcupyflasher
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
12:10:39 CMD: Backlog Template; Module; GPIO
12:10:39 MQT: devices/livingroom/boiler/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:10:39 MQT: devices/livingroom/boiler/stat/RESULT = {"Module":"43 (Sonoff Pow R2)"}
12:10:40 MQT: devices/livingroom/boiler/stat/RESULT = {"GPIO":"Not supported"}
```
- [x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  n/a
- [x] Provide the output of this command: ``Status 0``:
```
12:11:13 CMD: status 0
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"devices/livingroom/boiler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T21:44:36","StartupUTC":"2019-09-19T12:26:37","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":197,"SaveAddress":"F7000"}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["APAC02",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","280500000100068000000000000000000000","00000001"]}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS5 = {"StatusNET":{"Hostname":"Boiler","IPAddress":"10.0.0.24","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.6","Mac":"80:7D:3A:33:36:B3","Webserver":2,"WifiConfig":4}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3336B3","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 22 10:11:13 2019","Local":"Sun Sep 22 12:11:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"07:36","Sunset":"19:49"}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS10 = {"StatusSNS":{"Time":"2019-09-22T12:11:13","ENERGY":{"TotalStartTime":"2019-08-07T17:18:15","Total":89.538,"Yesterday":2.162,"Today":2.138,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
12:11:13 MQT: devices/livingroom/boiler/stat/STATUS11 = {"StatusSTS":{"Time":"2019-09-22T12:11:13","Uptime":"2T21:44:36","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"APAC02","BSSId":"44:D9:E7:91:E1:29","Channel":9,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:03"}}}
12:11:30 MQT: devices/livingroom/boiler/tele/STATE = {"Time":"2019-09-22T12:11:30","Uptime":"2T21:44:53","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"APAC02","BSSId":"44:D9:E7:91:E1:29","Channel":9,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:11:30 MQT: devices/livingroom/boiler/tele/SENSOR = {"Time":"2019-09-22T12:11:30","ENERGY":{"TotalStartTime":"2019-08-07T17:18:15","Total":89.538,"Yesterday":2.162,"Today":2.138,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
n/a

### TO REPRODUCE
Setup timer as above.

### EXPECTED BEHAVIOUR
Timer to be respected.

arendst, 2019-09-22:
Pls check output of command ``timers``

bakman2, 2019-09-22:
> Pls check output of command `timers`

output
```
12:28:59 MQT: devices/livingroom/boiler/stat/RESULT = {"Timers":"ON"}
12:28:59 MQT: devices/livingroom/boiler/stat/RESULT = {"Timers1":{"Timer1":{"Arm":0,"Mode":0,"Time":"02:00","Window":0,"Days":"1010101","Repeat":0,"Output":1,"Action":1},"Timer2":{"Arm":0,"Mode":0,"Time":"05:00","Window":0,"Days":"1010101","Repeat":0,"Output":1,"Action":0},"Timer3":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer4":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:28:59 MQT: devices/livingroom/boiler/stat/RESULT = {"Timers2":{"Timer5":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer6":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer7":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer8":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:28:59 MQT: devices/livingroom/boiler/stat/RESULT = {"Timers3":{"Timer9":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer10":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer11":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer12":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}
12:28:59 MQT: devices/livingroom/boiler/stat/RESULT = {"Timers4":{"Timer13":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer14":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer15":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0},"Timer16":{"Arm":0,"Mode":0,"Time":"00:00","Window":0,"Days":"0000000","Repeat":0,"Output":1,"Action":0}}}

```

Jason2866, 2019-09-22:
Timers are not activated. You have to check `Arm`

bakman2, 2019-09-22:
> Timers are not activated. You have to check `Arm`

Digging further, somewhere deep in my node-red flows there was also a schedule activated. 
Unbelievable. My apologies.

closed 2019-09-22 by bakman2