arendst/Tasmota #6474: Help with Rules.

dragonflyuk, 2019-09-23:
I've set a rule on one of my tasmota TH16 devices 

RULE ON DS18B20#Temperature>70 DO POWER ON ENDON ON DS18B20#Temperature<70 DO POWER OFF ENDON

However this is controlling a pump, with the inevitable effect that cold water is pumped into the system, and we go into a pump on\pump off cycle every few minutes.
I've been trying to set a timer so that the pump must stay on for atleast 30mins, but I've don't seem to be able to get it right.

Can anyone give me any pointers, timers seem to really confuse me.

dragonflyuk, 2019-09-23:
This is my latest incarnation, could this work?

RULE 
   ON RULES#TIMER=1 DO VAR1 0 ENDON 
   ON VAR1#1 DO BREAK 
   ON DS18B20#Temperature>70 DO POWER ON; VAR1 1;  RULETIMER1 1800; ENDON 
   ON DS18B20#Temperature<70 DO POWER OFF ENDON

effelle, 2019-09-23:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

dragonflyuk, 2019-09-23:
I Shall try that thanks, however it looks like the BREAK in the above aren't having an effect, and follwing rules are been executed.

andrethomas2, 2019-09-23:
Closing this issue as it has been answered. Please refer to additional links for support provided in previous post(s).

closed 2019-09-23 by andrethomas2