arendst/Tasmota #6475: Pressure Units and Altitude Correction

dennyfmn, 2019-09-23:
**Have you looked for this feature in other issues and in the wiki?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

Options for Pressure readings do not include units inHg, commonly used in the US.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

SetOption24 has pressure options for hPa and mmHg. Would it be  possible to add inHg to the options?

Also, chips like the Bosch BME280 report pressure referenced to absolute zero. It would be nice to be able to specify an offset correction for your local altitude so that the readings correspond to the local weather services. For example, my local altitude in Burnsville, MN USA is 950' above sea level. From a correction table at:

http://novalynx.com/manuals/bp-elevation-correction-tables.pdf
ELEVATION CORRECTION TABLES FOR BAROMETRIC PRESSURE SENSORS
Correction from table = 34.3mb, 1.016 InHg

If pressure were displayed in inHg and I was able to supply a correction factor of 1.016, the BME280 outputs would agree with our local weather services.

I presume it would require a new SetOption command.

Many thanks!

arendst, 2019-09-23:
Ever tried command ``Altitude`` ?

dennyfmn, 2019-09-23:
Thanks for the tip! Entering my altitude in meters I now get:

BME280-76 Temperature	78.9°F
BME280-76 Humidity	47.3%
BME280-76 Pressure	736.5 mmHg
BME280-76 SeaPressure	762.9 mmHg

BME280-77 Temperature	77.7°F
BME280-77 Humidity	47.2%
BME280-77 Pressure	736.8 mmHg
BME280-77 SeaPressure	763.2 mmHg

If there could be a Pressure unit choice for inHg, that would be great!

jziolkowski, 2019-09-23:
The pressure unit is stored as a binary value, which effectively limits it to mmHg or hPa. Adding more units would need extra work and it breaks some reverse compatibility regarding commands (because effectively this command would have to be renamed)

dennyfmn, 2019-09-23:
Thanks for looking into this. Maybe it can be done in the future.

closed 2019-09-23 by dennyfmn