arendst/Tasmota #6476: Android apk for Tasmota

araczkowski, 2019-09-23:
Hi Tasmota Community,

I'm representing a project AI-Speaker.com, we are working on voice controlled automation platform that allow to be controlled locally and respect privacy. As a part of our project we are delivering devices based on esp8266 thats works on Tasmota software (our fork https://github.com/sviete/AIS-Tasmota). 

Thank you for this great software!

We would like to give something back to you to :) we think that Android app to discover and control the Tasmota devices can be something which can help to use Tasmota(especially for the new users and for the end users with less technical knowledge).

The main author of the app is my colleague Kamil @l0kers 
We choiced the same license as Tasmota GNU GPL 3.0 https://github.com/sviete/AIS-devices/blob/master/LICENSE

[image]

The main features in the app are:

1. Scan WiFi networks and add the new device (discover the ESP AP via vendor in MAC address):
https://youtu.be/d7uW1mUzRlU

2. Discover the MQTT gate service (if it’s advertising via mDNS). Allow to set the mqtt broker for the device:
https://youtu.be/gMUoayJpOOY

3. Scan network to discover devices in network mDNS + add them to app:
https://youtu.be/AQnpPhmJD2M

You can grap signed app from here https://github.com/sviete/AIS-devices/releases/tag/v1.0
or build it yourself in Android studio. I'm currently releasing it on Google Play - the publication is pending... for Google approval.

Within a few weeks we have a plan to improve the appearance and refine the details.
I hope my colleague Kuba will find a time to help us improve the user experience in this app.

**If authors of Tasmota agree?** 
We can:
- Name and brand devices in the app as Tasmota (use the new icon and logo from Tasmota) 
- Add the detail description how to use the app in the Tasmota Wiki 

@arendst @ascillato @andrethomas @Jason2866
Please let me know what do you think?

And of course please let us know if you want to add or improve something in the app. 
We (@l0kers and @araczkowski) will do our best to make it better, we are open for pull requests.

jziolkowski, 2019-09-23:
Note that in next release, mDNS will be disabled by default (as in not even compiled in) so you might explore other options for device discovery.

Jason2866, 2019-09-23:
App doesnt work with my Samsung Galaxy Tab S2 :-(

araczkowski, 2019-09-23:
@jziolkowski  We are prepared for this already, currently we have bonjour and ip scanner.
https://github.com/sviete/AIS-devices/tree/master/app/src/main/java/pl/sviete/dom/devices/net

araczkowski, 2019-09-23:
@Jason2866 do you have some log, logcat via adb maybe?
 (the app will be in Google Play soon you will be able to send us a log)

jziolkowski, 2019-09-23:
How will the application primarily connect with the devices? There are numerous reports showing that (ab)using HTTP/webserver leads to increased LoadAvg and device stability issues. Tasmota is designed to be used mainly via MQTT.

Jason2866, 2019-09-23:
@araczkowski  Sorry no log. On Android platform i am a "just user"

araczkowski, 2019-09-23:
@jziolkowski The main goal of the app is to not control all the devices via http. But be able to add the device to your local wifi network then you can add it to mqtt broker (via this app). The app is a kind of tool to help you with parring and scanning the local network for devices. 
The mqtt broker is still needed if you wish add device to it (and the mobile phone is not a good cadidate to be a mqtt broker).

jziolkowski, 2019-09-23:
I don't understand how this app would help with adding the device to wifi (because it must already be connected for your app to discover it), and you don't add any devices to MQTT brokers.

jziolkowski, 2019-09-23:
So if you say you don't want to control the device, what's the point of discovering it in the first place?

araczkowski, 2019-09-23:
@effelle 

Part of our users are people with vision problems, they were some problems to use the pure Tasmota. 
Mainly because the pairing process (add device to wifi and to mqtt gateway) was too difficult for them. To make the process more easy we changed mainly 3 things:

**1. multi-press and hold functionality**
in ouer devices we are using only the one method of pairing (the AP / Wi-Fi Manager)
you can press the button from 3 to 7 times or hold it longer the Wi-Fi Manager (WifiConfig 2) will start
https://github.com/sviete/AIS-Tasmota/blob/b201ce2c9a05788b4879cf311a51f6063d4373ad/sonoff/support_button.ino#L267

**2. mqtt discover the gateway host**
some people were not able to keep the local mqtt gateway ip static. They are changing connection on gateway from wifi to eth, or the rooter is giving new ip to gateway and they don't know how to deall wit this.  So we decided to handle this problem. 
In our solution the mqtt gateway is informing our public service about  current local IP (after each change).  If the device can't reach the local mqtt host and the mqtt hostname starts with dom- (our name) we are asking our service for the local ip for mqtt gateway 
https://github.com/sviete/AIS-Tasmota/blob/b201ce2c9a05788b4879cf311a51f6063d4373ad/sonoff/xdrv_02_mqtt.ino#L161

**3. colors and other minor changes** 
mainly in my_user_config

on this video you can see how you can add device to a mqtt gate without monitor/display via remote with phisical buttons (this way is sutable for people with vision problems) https://www.youtube.com/watch?v=MnnVwIhimew
(sorry this video is only in polish)

jziolkowski, 2019-09-23:
1) can be easily done by configuring buttons as switches with proper switchmode and rules.

2) If someone assigned a dynamic IP to their MQTT broker then should not touch home automation, because clearly has no idea what he's doing.

effelle, 2019-09-23:
I agree. Not having a static IP for a broker doesn't make much sense.

Jason2866, 2019-09-23:
Or is the main goal to controlTasmota devices via http without any mqtt stuff installed?

jziolkowski, 2019-09-23:
I understand the part where you help with the initial AP configuration part by connecting to the sonoff- wifi for the user and "taking over" the configuration. That's helpful (at least for some)

andrethomas, 2019-09-23:
Hi

I'm guessing the use case is when you're not using any form of automation software and want to communicate from the APP directly to the devices via MQTT (I hope and not HTTP)?

Will this be limited to Android or are you planning on rolling out for iOS also?

Since the main reason for using a separate fork to maintain operationally acceptable functionality for your device(s) and the mobile app, it may be feasible to incorporate such changes into the main development branch even if such changes are elective during the build process to try and keep things as uniform as possible.

araczkowski, 2019-09-23:
@jziolkowski 

> I don't understand how this app would help with adding the device to wifi

I will try to expalin, step by step:
1. from menu you can choice "New device"
[image]

2. on next screen you will see the instruction how to put the device into the paring mode
[image]

3. select the devices from discovered list  (discover the ESP AP in paring mode via vendor in MAC address):
[image]

4. give the name for the device
[image]

5. wifi credentials

[image]

jziolkowski, 2019-09-23:
Read my last comment, I understand this part ;) (after a while, that is)

Jason2866, 2019-09-23:
@araczkowski  Thx for the explanation. Finally i got it!
Happy to test if a working version for my Tablet is there ;-)

jziolkowski, 2019-09-23:
A bit of a niche product, but maybe it will be popular among less-technically oriented users. Have fun with the development and good luck!

araczkowski, 2019-09-24:
hi, the app is alive at Google Play
https://play.google.com/store/apps/details?id=pl.sviete.dom.devices

According to Google 8,443 Android devices are supported by this app

[image]

and Samsung Galaxy Tab S2 is on the supported list
[image]

@Jason2866 please try to install the app from Google Play, let us know in case of problems. Thx :)

araczkowski, 2019-09-24:
@andrethomas 
> Will this be limited to Android or are you planning on rolling out for iOS also?

No such plans cause we don't have any experience with iOS. As I know to develop iOS app you'll need to be running OS X. We are using Linux to code.

Jason2866, 2019-09-24:
Still doesnt work on my Galaxy Tab.

araczkowski, 2019-09-24:
> doesnt work

@Jason2866  can you describe little more :)

Jason2866, 2019-09-24:
Thats all
[image]

araczkowski, 2019-09-24:
minSdkVersion 23
targetSdkVersion 28 (this is the minimum version acceptable by Google for new app on Google Play)

I just tried to open it on emulator (Nexus 6 API 23) and .... something is wrong, checking this

[image]

effelle, 2019-09-24:
Is the same error I got on my Nexus 7.

araczkowski, 2019-09-24:
@effelle I think yes - please try to upgrade (details below)

@Jason2866 
I figure out the problem, some resources were marked for specific SDK
Roll-out of the improved release 1.0.2 via Google Play is in progress

You can grab new signed by me (the same signature as from Google Play) version from here:
https://github.com/sviete/AIS-devices/releases/tag/1.0.2

It works on emulator now for older SDK, 
Thanks for bringing this issue up :) 
let me know in case of any other issue 

[image]

effelle, 2019-09-24:
OK. Some considerations:
1. Now it's working fine on my Nexus 7;
2. On first run, of all my Tasmota devices just 9 out of 21 are selectable. A second scan shown all of them;
3. If I have more than a relay connected to the device, just the first one is switchable on main panel, but for my understanding the app is not intended for control, since it offer the ability to open the webUI from the menu.
4. Since I don't use mDNS, the app can't see my broker then can't add device to it.

Jason2866, 2019-09-24:
Now it does work with my Galaxy Tab S2. Same findings as @effelle already mentioned.

araczkowski, 2019-09-24:
@effelle @Jason2866 

thx for info, cool :)

We are going to improve the look and some details in this app (add a picture how to quick press the button four times until the green LED blinks) then continue with the adding process to wifi. 
The goal is to make this configuration process so simple that even user without knowledge about network can do this.

Maybe we will find a solution to scan local network for MQTT gate on the standard MQTT port 1883 (or allow to add a gate manually to selected area).

Anyway I hope this app can be handy for initial configuraion (at least for some Tasmota users)

Here 
https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration

Instead of 

> If you don't have access to your router you can find your newly flashed device with an IP scanner:

we can use this app to do this part.

**Please let me know if you have any idea how we can make this app more useful for Tasmota users.**

meingraham, 2019-09-24:
Once the Tasmota device is connected to Wi-Fi, this should be able to find it as well - http://tasmota.simplethings.work/

araczkowski, 2019-09-25:
nice approach :)
this is a proof that for end users without knowledge about network and programing (probably more than 95% of population) this kind of tools are needed/helpful

andrethomas2, 2019-10-03:
Thanks for all the information and inputs shared - I suggest any further discussion be directed to the relevant GitHub page https://github.com/sviete/AIS-devices for this utility.

closed 2019-10-03 by andrethomas2

ascillato, 2019-10-03:
@araczkowski if you want, you can add this information to the wiki. Thanks

araczkowski, 2019-10-03:
Thanks @ascillato , 
OK, we will do this :)
but first we need to  improve the graphic (to have screens up to date  in Tasmota wiki)
My friend / the designer is ill now, so I hope we will start to work on app look next week.
The Tasmota has great API and deserve good looking app for device searching and configuration! (for less technical users) :)

ascillato, 2019-10-03:
:smile: :+1:

sushantr5, 2021-04-19:
Is the code repository publicly accessible? I am looking for similar app but with little bit different feature like integration of AWS Iot Core. I would like to contribute.

aybhalala, 2021-05-05:
@araczkowski Is the repository moved to some other location?

araczkowski, 2021-05-05:
Hi.
Yes, we sold this app to our client with all copyrights. As far as I know, they have decided to shut down the source and only use this app with their cloud service.
Sorry, life...

clach04, 2024-06-29:
> Is the code repository publicly accessible? I am looking for similar app but with little bit different feature like integration of AWS Iot Core. I would like to contribute.

https://github.com/vaginessa/AIS-devices maybe a snapshot ?

sfromis, 2024-06-29:
Checking forks of the one you found there does seem to indicate a "common inheritance". Very old code, of course....

weirdal3333, 2025-11-30:
> Hi. Yes, we sold this app to our client with all copyrights. As far as I know, they have decided to shut down the source and only use this app with their cloud service. Sorry, life...

You should not have sold it. You could have just archived the github repo and let someone else fork it. How much did you sell it for? The software looks waaaay better than tasmotrol from what the screenshots show.