arendst/Tasmota #6479: Sonoff 4ch V2 Ghost Switching

79137913, 2019-09-23:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I have 4 Sonoff 4ch V2 with the buttons wired to my wall switchs. All four devices have an low pass filter board I made to prevent GhostSwitching one LowPassFilter for each button, but no success here.
And the boards with the stock firmware works well.

Done:
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff 4ch V2
- [ ] Tasmota binary firmware version number used: 6.6.0 core 2.3.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: nodeMCU pyflasher
- [x ] Provide the output of this command: ``Backlog Template; Module; GPIO``:

> 00:49:31 CMD: Backlog Template; Module; GPIO
> 00:49:31 SRC: WebConsole from 192.168.1.102
> 00:49:31 RSL: Topic Recibido /Backlog, Tamaño de Datos 22, Datos Template; Module; GPIO
> 00:49:31 RSL: Grupo 0, Índice 1, Comando BACKLOG, Datos Template; Module; GPIO
> 00:49:31 SRC: Backlog
> 00:49:31 RSL: Topic Recibido /Template, Tamaño de Datos 0, Datos 
> 00:49:31 RSL: Grupo 0, Índice 1, Comando TEMPLATE, Datos 
> 00:49:31 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
> 00:49:31 SRC: Backlog
> 00:49:31 RSL: Topic Recibido /Module, Tamaño de Datos 0, Datos 
> 00:49:31 RSL: Grupo 0, Índice 1, Comando MODULE, Datos 
> 00:49:31 RSL: stat/sonoff/RESULT = {"Module":"7 (Sonoff 4CH)"}
> 00:49:31 SRC: Backlog
> 00:49:31 RSL: Topic Recibido /GPIO, Tamaño de Datos 0, Datos 
> 00:49:31 RSL: Grupo 0, Índice 1, Comando GPIO, Datos 
> 00:49:31 RSL: stat/sonoff/RESULT = {"GPIO1":"0 (Ninguno)","GPIO2":"0 (Ninguno)","GPIO3":"0 (Ninguno)"}

  Configuration output here:

  

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

> 
> 00:53:57 RSL: Topic Recibido /status, Tamaño de Datos 1, Datos 0
> 00:53:57 RSL: Grupo 0, Índice 1, Comando STATUS, Datos 0
> 00:53:57 RSL: stat/sonoff/STATUS = {"Status":{"Module":7,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":6,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 00:53:57 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-ES.bin","RestartReason":"Power on","Uptime":"0T00:35:24","StartupUTC":"2019-09-23T23:18:33","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":56,"SaveAddress":"F4000"}}
> 00:53:57 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:19:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
> 00:53:57 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bunker",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
> 00:53:57 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["0000040A","0FDAE396","001783A0","22B617CD","01003BC0"]}}
> 00:53:57 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0074","IPAddress":"192.168.1.110","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:88:E0:4A","Webserver":2,"WifiConfig":1}}
> 00:53:57 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_88E04A","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
> 00:53:57 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 23 23:53:57 2019","Local":"Tue Sep 24 00:53:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:39","Sunset":"18:45"}}
> 00:53:57 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-24T00:53:57"}}
> 00:53:57 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-24T00:53:57","Uptime":"0T00:35:24","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":71,"POWER1":"OFF","POWER2":"ON","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Bunker","BSSId":"64:70:02:70:27:24","Channel":4,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:17"}}}

- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

>   00:46:04 WIF: Conectado
> 00:46:26 WIF: Probando conexión...
> 00:46:26 WIF: Conectado
> 00:46:48 WIF: Probando conexión...
> 00:46:48 WIF: Conectado
> 00:47:08 WIF: Probando conexión...
> 00:47:08 WIF: Conectado
> 00:47:20 APP: Botón3 multi-press 1 **here i didnt press the button**
> 00:47:20 SRC: Button
> 00:47:20 RSL: stat/sonoff/RESULT = {"POWER3":"ON"}
> 00:47:20 RSL: stat/sonoff/POWER3 = ON
> 00:47:20 CFG: Grabado a la flash en F4, Conteo 56, Bytes 3584
> 00:47:28 WIF: Probando conexión...
> 00:47:28 WIF: Conectado
> 00:47:33 HTP: Menú Principal
> 00:47:36 HTP: Información
>   
> 

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-24:
Try setting `ButtonDebounce`

79137913, 2019-09-24:
> 
> 
> Try setting `ButtonDebounce`

Hello! 

What is the integer value I should set? (its set on 50)

Jason2866, 2019-09-24:
Doy you have all the switches on the main board on factory default settings?

Aradzh, 2019-09-24:
I have similar problem with 4 CH PRO (first version)
I try to use ButtonDebounce with different intervals and (SetOption13 1 and SetOption1 1) - but unsuccessful - in log it is something like
00:00:04 APP: Button3 level 1-0
00:00:04 SRC: Button
when I did not press buttons. I think about low pass filter mod and in one forum I read that for T1 it helps if one wash a welding flux from board. 
Another solution disable buttons - Is there is a command to disable buttons ?

Jason2866, 2019-09-24:
@Aradzh  Use Template function. In your template just delete the buttons config from the GPIOs.
No more buttons which can do ghost acting

Aradzh, 2019-09-24:
> @Aradzh Use Template function. In your template just delete the buttons config from the GPIOs.
> No more buttons which can do ghost acting
Thanks ! Completely disable buttons using Template function.

79137913, 2019-09-24:
@Jason2866 
> Try setting `SetOption13 1` and `SetOption1 1`

Hello, Im going to proceed changing those options and start the supervision process.

> Doy you have all the switches on the main board on factory default settings?
> https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-Pro#4ch-pro-dipswitch-configuration

All the boards are with a default config, only things changed are module Sonoff 4CH(7), ButtonDebounce:200 SetOption13:1 SetOption1:1, Wifi Settings, and thats all

I really don't know what is:
> Tasmota the following settings are recommended:
> 
>     S6: 1
>     K5: all 1
>     K6: all 0
> 
> (0 and 1 are printed onto the board next to the switch names)

here is a picture of my board, it's not a PRO it's a 4ch v2
https://i.imgur.com/eySjXPL.jpg

Please answer me how do i config s6 k5 k6 and where to find the 0 or 1 next to the switch names so i can proceed with this configuration

Jason2866, 2019-09-24:
Sorry i missread it is a 4ch (not a pro). So no dip switches.
So only way without hardware mod is deleting the button config with a self created template.

Jason2866, 2019-09-24:
Original:
[image]

change to:
[image]

Jason2866, 2019-09-24:
How are the wall switches connected?

79137913, 2019-09-24:
https://i.imgur.com/QU5IIlK.jpg
Each color cable to a low pass filter connected to 3.3v and gnd from the board pins, then connected to one side of the wall switch. The other side of the wall switch is connected to GND board pin.

I've tried your template configuration and yes, no ghost switching but my wall switches don't work . So this is not an option.

If the low pass filter isn't enough i can add an optocupler board, what do you think?
https://www.banggood.com/A87-4-Channel-Optocoupler-Isolation-Module-High-And-Low-Level-Expansion-Board-For-Arduino-p-1195867.html?cur_warehouse=CN

Jason2866, 2019-09-24:
The Optocoupler should do the job.

79137913, 2019-09-24:
Nice, I'm going to try and then Comment here if it was a succes or not.

ascillato2, 2019-09-29:
Putting this issue ON HOLD until the user has time to test it.

closed 2019-09-29 by ascillato2

konzell, 2019-12-27:
I got the same problem (Ghost Switching) with an Sonoff 4CH Pro Modul, although I connected my wall switches to the buttons by relays. I already deactivated mqtt, but it is still ghost switching even if all 4 channels are off

ascillato, 2019-12-28:
As connecting wall switches to a sonoff 4ch is indeed a hardware modification, this type of issues can show up depending on the value of pull up resistors, type of wire, soldering, proximity to Mains AC, on how long is the wires used, if are twisted pair or not, if the sorrounding of the sonoff 4ch is an electrical noisy environment, etc etc. So, to have a reliable setup, it is better to use the sonoff ch4 as comes from the manufacturer and for wall switches use a device manufactured for that usecase like a shelly.

konzell, 2019-12-28:
I maybe found the solution, solution since I cleared the rf code by obening and pressing the rf switch long it did note appear again, I will observe this.

@ascillato: The wall switches are decoupled with 4 relays right next to the Sonoff (less than 30 cm cable, various relays with 8V, 12 V, 220V AC), that it was not because I could also check by deactivating the switches with a custom template and the ghost switchings occurred anyway. Nevertheless, I am happy to hear suggestions if someone knows a more suitable, cheap device.

letsgetbusy, 2020-01-02:
I at all and happy new year.
I have the same problem with a 4CH PRO. 
I have a LowPassFilter, delete RF code form the board, and i cut a PIN11 from RF IC..  Ghost switching are there..

iitbulls, 2020-04-05:
Troubled with ghost switching on 4ch pro. Problem is occuring even with MQTT disabled. Another problem i found that this occurs on only POWER1 of both the sonoff 4ch pro I have. I am unable to diagnose this and fix it, please help out