arendst/Tasmota #6481: Cannot use '*' in Web Admin password

siochs, 2019-09-24:
### BUG DESCRIPTION
Cannot use '*' in Web Admin password.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): GoSund SP1
- [x] Tasmota binary firmware version number used: 6.6.0 2019-07-06T13:10:20
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
00:03:14 CMD: Backlog Template; Module; GPIO
00:03:14 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:03:14 RSL: stat/sonoff/RESULT = {"Module":"55 (Gosund SP1 v23)"}
00:03:15 RSL: stat/sonoff/RESULT = {"GPIO":"Not supported"}
```
- [x] Provide the output of this command: ``Status 0``:
```
00:03:51 CMD: Status 0
00:03:51 RSL: stat/sonoff/STATUS = {"Status":{"Module":55,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:03:51 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:49","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":22,"SaveAddress":"F6000"}}
00:03:51 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:03:51 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["test","test-iot"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
00:03:51 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
00:03:51 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4044","IPAddress":"192.168.43.211","Gateway":"192.168.43.21","Subnetmask":"255.255.255.0","DNSServer":"192.168.43.21","Mac":"84:0D:8E:54:EF:CC","Webserver":2,"WifiConfig":4}}
00:03:51 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_54EFCC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:03:51 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:03:51 1970","Local":"Thu Jan 01 00:03:51 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:03:51 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:03:51 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:51","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:03:51 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:03:51","Uptime":"0T00:03:49","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"test","BSSId":"22:F4:78:61:C8:6E","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
Cannot provide output because device immediately reboots.
```

### TO REPRODUCE
- Flash sonoff.bin
- Connect to AP
- Setup WiFi credentials for AP1 and AP2
- Device reboots and connects to AP1
- Go to main menu -> Configuration -> Confugure Module -> GoSund SP1. Reboot.
- Go to main menu -> Configure Other -> Web Admin Password [x] -> Enter t4zm0ta* -> Save. Reboots
- Go to main menu. No password prompted.

But ommitting '*' works as expected

### EXPECTED BEHAVIOUR
- Flash sonoff.bin
- Connect to AP
- Setup WiFi credentials for AP1 and AP2
- Device reboots and connects to AP1
- Go to main menu -> Configuration -> Confugure Module -> GoSund SP1. Reboot.
- Go to main menu -> Configure Other -> Web Admin Password [x] -> Enter t4zm0ta* -> Save -> Prompt for admin / ta4zm0ta* -> Reboots
- Go to main menu. Prompt for password.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
N/A

Jason2866, 2019-09-24:
No special characters allowed for password. Dont use `*`

siochs, 2019-09-24:
Ok then. Not a bug. Thank you.

closed 2019-09-24 by siochs