arendst/Tasmota #6483: SonOff POW Ver2.0 Bad Power Measurement with no loading

tugsi, 2019-09-24:
I have two Sonoff POW Ver2.0 (not R2!), both with the same issue.
Having no load and switching on, the measurement didn't show 0, it is swaying between 0W up to over 100W, same way the  #current.
I also have a POW Ver1.0, there isn't such are failure, all is fine.

So i took a look inside and i saw, that the Ver2.0 have the CSE7759 Chip on it.

After this, I found a issue on Github on Espurna, with the same behavior.
[SonOff POW Bad Power Measurement #696](https://github.com/xoseperez/espurna/issues/696)

One Guy is telling about the different against the HLW-Chip, one important message is found in the Datasheet:

> Note: Due to the noise of CSE7759, if there is no signal input, there will be about 2Hz signal output on the CF1 pin indicating the current and voltage rms value (this signal can be judged by the CF pulse frequency to hide the movement). To ensure the correct measurement of the effective value, the effective value of the current applied to the channel V1P\V1N shall not be less than 40uv, and the effective value of the voltage applied to the channel V2P shall not be less than 480uv. When using, please consult our technical support.

But here on Tasmota I found nothing, I cannot believe, that no one has the failure but me ;-)

I used different releases of Tasmota, also the newest 6.6.0 .
But nothing changed.

Has anyone a solution for the problem?

Thanks Thomas

Jason2866, 2019-09-24:
Newest is NOT 6.6.0.  You can do a OTA to latest [version](http://thehackbox.org/tasmota/json/sonoff.bin)
Many changes since 6.6.0. Give it a try, your problem could be solved already

tugsi, 2019-09-24:
OK, you're right, I've tried the latest (6.6.0.13), but nothing changed.

meingraham, 2019-09-24:
`SetOption39` plus a thorough calibration and see if that improves things.

ascillato2, 2019-09-24:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-24 by ascillato2

tugsi, 2019-09-24:
I don't think this issue has been answered anywhere. :-(
If yes, then tell me where, please.

I do the `SetOption39` tried several numbers, also 1 too, as described in the Commands Wiki.
But nothing changed.
All the description found with `SetOption 39` is for the Pow R2, but i have the Pow R1 in Version 2.0!
If i change the configuration in Pow R2, nothing is measured.

tugsi, 2019-09-24:
I don't think this issue has been answered anywhere. :-(
If yes, then tell me where, please.

I do the `SetOption39` tried several numbers, also 1 too, as described in the Commands Wiki.
But nothing changed.
All the description is for the Pow R2, but i have the Pow R1 in Version 2.0!
If i change the configuration in Pow R2, nothing is measured.

meingraham, 2019-09-24:
`plus a thorough calibration`?

https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration

tugsi, 2019-09-25:
Yes, also made the Power calibration. With a 60W Lightbulb.
Also metering the Voltage. (I'm a electrician)
Still have the swaying of the Power.

Voltage | 234 V
Current 0.218 A
Power 7 W
Aperen Power 51 VA
Reactive Power 50 VAr
Power Factor 0.14

It seems, that the `SetOption39 1` have no function.

tugsi, 2019-09-25:
I´ve read the Code for the CSE7759 in Tasmota, 
Could it be, that I have to change the Template?
Usually the POW R1 uses the HWL-Chip, so the GPIOs are set with HWL.
If I read the code right, the Option 39 could only be active with the CSE used.
But I dont know, which GPIO with which settings.
HWL has 3 GPIOs used, changing to CSE, I only have 2 (RX and TX).