arendst/Tasmota #6488: Allow configuring arbitrary off-peak times for tariff commands

znanev, 2019-09-24:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
This feature request is not related to a real problem, but if implemented, it will improve accuracy of energy consumption reporting when using off-peak tariffs. 

**Describe the solution you'd like**  
Currently the **tariff** commands use whole hours to configure the off-peak period. Where I live (UK), off-peak period start/end time can fall on half an hour also.
Example configuration, supported in current implementation, for 8 hours off-peak period 22:00 - 06:00 (standard time) and 23:00 - 07:00 (daylight saving time):
```
tariff1 22,23
tariff2 6,7
```
**Describe alternatives you've considered**  
If the **tariff** commands support arbitrary times (not whole hours only), it should be possible to configure for example 7 hours off-peak period 22:30 - 05:30 (standard time) and 23:30 - 06:30 (daylight saving time):
```
tariff1 22:30,23:30
tariff2 5:30,6:30
```

**Additional context**  
Arbitrary times for **tariff** commands will also help account for time drift that some electricity meters exhibit. I'll give example with my meter, which now "thinks" that the time is 21:49 (when real time is in fact 23:44). So it lags about 01:55 and accounting for this time drift, configuration for the off-peak period should be:
```
tariff1 0:25,1:25
tariff2 7:25,8:25
```

meingraham, 2019-09-24:
@arendst 

Did I call it or did I call it ;-)

Except I proposed using minutes since midnight as is used in other "time" commands.

Mike

P.S. I don't really have a horse in this race as my area does not use tariffs ;-)

andrethomas, 2019-09-25:
My personal view is that this should be implemented on the home automation side (node-red etc) and not on the device itself.

arendst, 2019-09-25:
@meingraham you called it.

arendst, 2019-09-25:
I agree with @andrethomas but as I already did so much work on it I guess extending it to minutes makes it more usefull. Just waiting for the one requesting four tariffs a day...

pablozg, 2019-09-25:
> Just waiting for the one requesting four tariffs a day...

I don't know if four tariffs exists in any country, but in Spain we have up 3 tarrifs , "Punta - Standard", "Valle - Off-peak" and "Supervalle - Superoff-peak", the last one from 01:00 to 07:00. 😄 

@arendst What do you think? 😃

arendst, 2019-09-25:
The best thing to do here is abandon Spain.

znanev, 2019-09-25:
Thank you, Theo!

Just tested the new build and I was able to configure my strange start / end times for off-peak tariff.

I'm closing the issue now, because this feature request has been implemented.

closed 2019-09-25 by znanev

slavkobrzeg, 2019-11-29:
> I agree with @andrethomas but as I already did so much work on it I guess extending it to minutes makes it more usefull. Just waiting for the one requesting four tariffs a day...

Well, more tariffs per day is maybe too much, but in Poland our off-peak tariff is active two times per day. Mostly during a night but also usually for 2-3 hours around noon. I am, for example, using those two hours to top-up a bit my electric car. I'm using tasmota to count off-peak consumption over the night, but it is not accurate right now, because my charges during a day is counted as peak time.
So is this possible to set more periods for off-peak tariff?

TakissX, 2022-10-16:
Another solution is to trigger. (After the electrical panels have an LED to indicate when the reduced tariff is activated. )
So I have configured a “sonoff mini” as a detector. And with scripts from Domoticz that I work with, I activate the devices that I want to work only during the hours of the reduced tariff.

marsic-84, 2024-06-17:
Command "EnergyTotal 777777" resets standart tariff to 777.777 kWh
How to reset off-peak tariff??