arendst/Tasmota #6489: Curtain Motor Tuya

Greefon, 2019-09-25:
Hi everyone!
I bought this WiFi Tuya Motor - https://ru.aliexpress.com/item/32957276089.html?spm=a2g0s.9042311.0.0.11d033edYrVfyc 
Successfully flashed it with FlashESP8266 tool as the OTA was not working. 
Started web-interface and my happiness gone :) as I didn't get how to configure the GPIOs against available functions/component. I started to discover the board itself and made a disappointing  conclusion. The ESP8266 is connected to the main board by 4 contacts: VCC, GND, RX and TX.
It is not directly connected to the relays or sensors. It's tied-up to the STM 8S003F3P6 controller which was hidden right under the ESP.
Now I faced a situation that I can't resolve: how to move next? 
I see GND, RES, VCC, SWIN (holes) contacts for the STM controller to upload/download, but I doubt that it's not protected in any way to for example read-out contained software.
As ESP was overwritten there is no way I can sniff the RX/TX communication from it, neither read-out software from ESP itself.
So I'm stuck. I thinking of buying another the same motor, but can't predict whether any of ways described above will work.
Please advice - how to get it work, any ideas would be appreciated. 
Thank you!
[image]
[image]
[image]
[image]
[image]
[image]

meingraham, 2019-09-25:
[TuyaMCU](../wiki/Commands#TuyaMCU). Also read the linked article which steps through the process of determining the proper setup (if the device is currently supported).

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-25 by Greefon

leifnel, 2019-12-26:
If this is resolved, it would be nice to have the result "backported" to here or wiki instead of having to look for tidbits in a discord archive.