arendst/Tasmota #6491: Moes QS-WiFi-D01 Dimmer not dimming when controlled via Alexa

thomas-lentz, 2019-09-25:
### BUG DESCRIPTION
When controlling the Wifi Dimmer with Alexa it is not possible to set the dimming. 

- On/Off via Alexa works fine
- Console input works with dimming (& On/Off)
- WebUI also works with dimming (& On/Off)

Personal Opinion: Missing SerialSend5 sending dim value to Dimmer MCU.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Moes QS-Wifi-D01-Triac
- [x] Tasmota binary firmware version number used: Version 6.6.0(sonoff) BuildDateTime 2019-09-24 22:20:45
 - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino IDE
- [x] Flashing tools used: Arduino IDE
- [x] Provide the output of command:  ``Backlog Template; Module; GPIO``:
  ```
  10:25:19 CMD: Backlog Template; Module; GPIO
  10:25:19 RSL: stat/sonoff/RESULT = {"NAME":"WiFi-Dimmer","GPIO":[0,148,0,149,0,0,0,0,0,42,37,0,0],"FLAG":0,"BASE":18}
  10:25:19 Script: performs "SerialSend5 FF555805DC0A"
  10:25:19 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
  10:25:19 Script: performs "Dimmer 40.00"
  10:25:19 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":40}
  10:25:20 RSL: stat/sonoff/RESULT = {"Module":"0 (WiFi-Dimmer)"}
  10:25:20 Script: performs "SerialSend5 FF555805DC0A"
  10:25:20 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
  10:25:20 Script: performs "Dimmer 40.00"
  10:25:20 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":40}
  10:25:20 RSL: stat/sonoff/RESULT = {"GPIO":"Not supported"}
  10:25:20 Script: performs "SerialSend5 FF555805DC0A"
  10:25:20 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
  10:25:20 Script: performs "Dimmer 40.00"
  10:25:20 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":40}
  10:25:27 RSL: tele/sonoff/STATE = {"Time":"2019-09-25T10:25:27","Uptime":"0T00:25:17","Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"MyWifi","BSSId":"AA:BB:CC:DD:EE:FF","Channel":1,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:05"}}
  10:25:27 RSL: tele/sonoff/SENSOR = {"Time":"2019-09-25T10:25:27","COUNTER":{"C1":0}}
  ```
- [x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  No rules but script:
  https://github.com/arendst/Sonoff-Tasmota/files/3476824/WiFi-Dimmer-Script-v0.3.txt
   ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  10:28:04 CMD: status 0
  10:28:04 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Dimmer-Sleeping"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
  10:28:04 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:27:54","StartupUTC":"2019-09-25T09:00:10","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":136,"SaveAddress":"F4000"}}
  10:28:04 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-24T22:20:45","Boot":7,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
  10:28:04 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyWifi","MyWifi2"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00C08009","280500000100060000000000000000000000","00000000"]}}
  10:28:04 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":601,"Free":400,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","07CAE396","001F83A0","22B617CD","01003BC0"]}}
  10:28:04 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-8167","IPAddress":"192.168.28.77","Gateway":"192.168.28.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.28.1","Mac":"84:0D:8E:B6:BF:E7","Webserver":2,"WifiConfig":4}}
  10:28:04 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B6BFE7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
  10:28:04 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 25 09:28:04 2019","Local":"Wed Sep 25 10:28:04 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:40","Sunset":"18:43"}}
  10:28:04 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-25T10:28:04","COUNTER":{"C1":0}}}
  10:28:04 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-25T10:28:04","Uptime":"0T00:27:54","Heap":23,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"MyWifi","BSSId":"AA:BB:CC:DD:EE:FF","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  10:30:55 CMD: weblog 4
  10:30:55 RSL: stat/sonoff/RESULT = {"WebLog":4}
  10:30:55 Script: performs "SerialSend5 FF555805DC0A"
  10:30:55 SRC: Rule
  10:30:55 RSL: Received Topic /SerialSend5, Data Size 12, Data FF555805DC0A
  10:30:55 RSL: Group 0, Index 5, Command SERIALSEND, Data FF555805DC0A
  10:30:55 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
[... 66 lines cut ...]
10:42:01 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
10:42:01 Script: performs "Dimmer 28.00"
10:42:01 SRC: Rule
10:42:01 RSL: Received Topic /Dimmer, Data Size 5, Data 28.00
10:42:01 RSL: Group 0, Index 1, Command DIMMER, Data 28.00
10:42:01 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":28}
10:42:01 CFG: Saved to flash at F8, Count 156, Bytes 3584
10:42:02 UDP: Packet (133)
10:42:04 HTP: Hue API (/b6bfe7/lights/191626865)
10:42:04 HTP: Hue Result ({"state":{"on":true,"bri":71,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Dimmer-Sleeping","modelid":"LCT007","uniqueid":"84:0d:8e:b6:bf:e7:00:11-1","swversion":"5.50.1.19085"})
```

3) Controlled via Alexa:
```
10:40:50 HTP: Hue API (/b6bfe7/lights/191626865/state)
10:40:50 HTP: Hue POST args ({"bri":128})
10:40:50 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":50}
10:40:50 HTP: Hue Result ([{"success":{"/lights/1/state/bri":128}}])
10:40:50 CFG: Saved to flash at FA, Count 154, Bytes 3584
10:40:50 HTP: Hue API (/b6bfe7/lights/191626865)
10:40:50 HTP: Hue Result ({"state":{"on":true,"bri":128,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Dimmer-Sleeping","modelid":"LCT007","uniqueid":"85:0e:8e:b6:bf:e7:00:11-1","swversion":"5.50.1.19085"})
```
You can refresh the WebUI after setting the dim value via Alexa and then see the requested value displayed in the slider setting, but it is never send via Serialsend5 to the MCU of the dimmer.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-25:
Try latest development version. There where a change in HueEmulation.
Didnt remember exactly what was fixed. 
So update first, before hunting something maybe already solved

shantur, 2019-09-25:
@thomas-lentz : You can try to upgrade to latest development version and configure dimmer using `TuyaMcu` command. No scripts and then try HueEmulation.

https://github.com/arendst/Sonoff-Tasmota/wiki/MCU-Based-Tuya-Dimmers-and-Switches

https://github.com/arendst/Sonoff-Tasmota/wiki/TuyaMCU

thxthx0, 2019-09-25:
@thomas-lentz
dimming works with the script and HUE Emu,
 
the issue was solved here: https://github.com/arendst/Sonoff-Tasmota/issues/6085#issuecomment-517977908

Plz update Tasmota to the latest dev version 6.6.0.14  as suggested.

thxthx0, 2019-09-25:
@shantur 
This dimmer uses "non standard" serial commands,
and switching is controlled by pulse recognition.

https://github.com/arendst/Sonoff-Tasmota/issues/6491

I don't think it's controllable by TuyaMCU atm,
but it would be great if it were possible in the future.

Jason2866, 2019-09-25:
@thxthx0  Thx for the info. Since i do not have such a device i was not "on track"

thomas-lentz, 2019-09-27:
Okay, the hint in [issue 6085 (comment)](https://github.com/arendst/Sonoff-Tasmota/issues/6085#issuecomment-517977908) did the trick. 
Now the script and the rules are invoked after a change via hue/Alexa.
Thank you!

closed 2019-09-27 by thomas-lentz