arendst/Tasmota #6493: YTF IR Bridge flashed but can not configure IR ...

Eddie-BS, 2019-09-25:
Hi,

I did succesfully flash an YTF IR bridge with tuys convert (ran on RPI)
the YTF comes online as basic sonoff device and I can start configuring it.
I did change the device type to YTF IR Bridge (62) and the device reboots.
After that I do not have the 

GPIO | Component | Description
-- | -- | --
04 | LED1i (52) | Blue LED - Link status
05 | IRrecv (51) | IR Receiver
13 | Button1 (17) | Button
14 | IRsend (8) | IR Transmitter

setting to configure, it shows :

[image]

Am I missing something ?

regards,
Eddie

arendst, 2019-09-25:
You're missing nothing. The GPIO's used by the device are invisible.

Eddie-BS, 2019-09-25:
thanks Theo,
any suggestions in how to configure this further to send and receive signals ?

meingraham, 2019-09-25:
@Eddie-BS 

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Eddie-BS, 2019-09-25:
@meingraham tnx, I will close the issue and try to find a solution myself.

closed 2019-09-25 by Eddie-BS