arendst/Tasmota #6496: Reset WiFi config using button

abhitheawesomecoder, 2019-09-26:
Erase/Reset WiFi config using button press

Jason2866, 2019-09-26:
Yes, possible

abhitheawesomecoder, 2019-09-26:
How any supporting link?

Jason2866, 2019-09-26:
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

abhitheawesomecoder, 2019-09-26:
I checked Wiki but it's big and was not able to find that's why asking help
from community

abhitheawesomecoder, 2019-09-26:
Erase/Reset WiFi config using button press (sonoff RF bridge powered by tasmota)

andrethomas, 2019-09-26:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage#4-short-presses

andrethomas2, 2019-09-26:
Closing this issue as it has been answered. Please refer to additional links for support provided in previous post(s).

closed 2019-09-26 by andrethomas2

abhitheawesomecoder, 2019-09-27:
Andre Thomas Thanks so much for your help