arendst/Tasmota #6497: Sonoff Tasmota 6.6.0 with Blitzwolf SHP5 Battery = 0 in MQTT messages

zebons, 2019-09-27:
> **GUIDE**  
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>   
> Please DO NOT OPEN AN ISSUE:  
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING

I Flashed a Blitzwolf SHP5 module with Sonoff Tasmota Classic build 6.6.0 and forward onfos to Domoticz via MQTT Messages. All is working fine except battery level set to 0 in MQTT messages.
`10:49:47 MQT: domoticz/in = {"idx":35,"nvalue":0,"svalue":"51;115.7","Battery":0,"RSSI":9}`
 As this module is powered by AC current, we 'd get a battery level to 255 (no battery used) or a real value (just having an info if the module is operational or not).
As a result I get a warning in Domoticz for low battery...

Notice that I also flash a Sonoff S26 and don't have that issue as you can see below : 
Sonoff S26 MQTT Message 10:54:08 MQT: domoticz/in = {"idx":13,"nvalue":0,"svalue":"","Battery":200,"RSSI":9}
 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Blitzwolf SHP5
- [X] Tasmota binary firmware version number used: Sonoff Classic 6.6.0
- [X] Pre-compiled
- [X] Flashing tools used: ESPEasyFlasher
- [X] Provide the output of this command: ``Backlog Template; Module; GPIO``:
` 10:31:17 MQT: domoticz/in/RESULT = {"NAME":"SHP5","GPIO":[56,145,56,146,0,22,0,0,0,0,21,0,18],"FLAG":0,"BASE":18}`
  Configuration output here:
- [X] Provide the output of this command: ``Status 0``:
  STATUS 0 output here:
```
  10:32:42 CMD: STATUS 0
10:32:42 MQT: domoticz/in/STATUS = {"Status":{"Module":0,"FriendlyName":["Blitzwolf","Sonoff2"],"Topic":"in","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:32:42 MQT: domoticz/in/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T01:33:36","StartupUTC":"2019-09-27T07:59:06","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":151,"SaveAddress":"F6000"}}
10:32:42 MQT: domoticz/in/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-classic)","BuildDateTime":"2019-07-06T13:11:57","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:32:42 MQT: domoticz/in/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXXXXXX","XXXXXXXX"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060100000000000000000000","00000000"]}}
10:32:42 MQT: domoticz/in/STATUS4 = {"StatusMEM":{"ProgramSize":489,"Free":512,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","6010A392","009003A8","200217C6","010000C0"]}}
10:32:42 MQT: domoticz/in/STATUS5 = {"StatusNET":{"Hostname":"Blitzwolf SHP5","IPAddress":"XXX.XXX.XXX.XXX","Gateway":"XXX.XXX.XXX.XXX","Subnetmask":"XXX.XXX.XXX.XXX","DNSServer":"XXX.XXX.XXX.XXX","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
10:32:42 MQT: domoticz/in/STATUS6 = {"StatusMQT":{"MqttHost":"XXX.XXX.XXX.XXX","MqttPort":XXXX,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BE3A6","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:32:42 MQT: domoticz/in/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 27 09:32:42 2019","Local":"Fri Sep 27 10:32:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
10:32:42 MQT: domoticz/in/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:32:42 MQT: domoticz/in/STATUS10 = {"StatusSNS":{"Time":"2019-09-27T10:32:42","ENERGY":{"TotalStartTime":"2019-09-26T12:20:16","Total":0.113,"Yesterday":0.101,"Today":0.013,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
10:32:42 MQT: domoticz/in/STATUS11 = {"StatusSTS":{"Time":"2019-09-27T10:32:42","Uptime":"0T01:33:36","Vcc":0.193,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":3,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
   Console output here:
  
```
10:49:47 MQT: domoticz/in = {"idx":35,"nvalue":0,"svalue":"51;115.7","Battery":0,"RSSI":9}
10:49:47 MQT: domoticz/in = {"idx":30,"nvalue":0,"svalue":"241","Battery":0,"RSSI":9}
10:49:47 MQT: domoticz/in = {"idx":32,"nvalue":0,"svalue":"0.218","Battery":0,"RSSI":9}
```

meingraham, 2019-09-27:
`sonoff-basic` is only for initial OTA flash or specifically for the Sonoff Basic line of devices.

> If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. 

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

zebons, 2019-09-27:
It's not a sonoff-basic release but a sinoff-classic version.
I first load the basic firmware but upgrade to classic when I see that Domoticz need that minimal version.
My Sonoff 26 module has also the classic version and works fine.

meingraham, 2019-09-27:
Classic/Basic - my oversight.

The rest of my post applies.

> If your Tasmota version is not the latest from **_the development branch_**, please update your device before submitting your issue.

> close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4)

zebons, 2019-09-27:
It's  the latests stable buid release. I don't want to compile a version from the development branch !
It's working fine for the Sonoff S26 with the same stable release !
If you tell me that is corrected in the development branch I can wait but please don't give those rapid response totally unusefull. I take time to test and give you the maximum of informations please take a little time to analyze my issue before your response (I can wait...)

meingraham, 2019-09-27:
Development is stable.

zebons, 2019-09-27:
Very surprising !
If you don't take into account issues on the latest stable release, I don't know how it'd be corrected in future release even in the development branch ! 
Is there dayly  (or weekly) build from the development branch or must we compile the latest sources  ?  Is there a tutorial to do so ?

arendst, 2019-09-27:
What you also could have done is read the changelog of the development branch. You would find a solution to your issue in version 6.6.0.1.

Lucky for you the development binaries are build after every commit so I suggest you start reading the README.md file to find the link to these binaries.

Happy weekend.

meingraham, 2019-09-27:
Development binaries - since it is a recommendation in the Issue template, the link to the repository is provided in the template as well.

@zebons NOW, please close this issue.

closed 2019-09-27 by zebons