arendst/Tasmota #6498: Enable forwarding entire console output to MQTT [MQTLog?]

jziolkowski, 2019-09-27:
Making TDM, I've been trying to find out if entire device management could be done _without_ using WebUI; the simple answer is no, because of console. While a lot of data can be requested via MQTT, some debugging information is available only there. Hence my suggestion, to add the possibility to forward console to MQTT.

Why? Because for a tool that's not meant to be used as a main interface to Tasmota, webserver takes up a reasonably amount of resources.

As it will naturally increase MQTT chatter a lot, I have a few thoughts how to avoid it.
- first of all, by default (unless specified otherwise during compilation), the feature is disabled. Those who don't need it won't even know it's there.
- one option is to have the console respond on a completely different, hardcoded prefix, ex. cons/, so MQTT clients, home automation software etc can simply ignore the messages
- another option, as an alternative to the above, reply topic different than the usual RESULT, ex. CONSOLE, which also can be ignored from parsing by other apps.

This is just a concept, as I'm not proficient enough (or at all?) in C++ to even attempt making a draft or a PR, but @shantur said that this should be easy to implement.

Jason2866, 2019-09-27:
@arendst  what resources would be free if webserver is not used?

shantur, 2019-09-27:
Another option (which @jziolkowski won't like it) can be Websockets. That could give Tasmota another way to communicate to. In my opinion, one good thing about using Websocket for configuration management is that Websockets can be used without any MQTT broker setup.

Jason2866, 2019-09-27:
@shantur  +1 dislike Websockets ;-)

jziolkowski, 2019-09-27:
Indeed, there are plenty alternatives. But at the same time, Tasmota which is meant to be used primarily via MQTT, doesn't present all the (vital) data via this transport.

arendst, 2019-09-27:
Without the webserver you'll lose Wemo/Hue emulation too.

Just a compile without define USE_WEBSERVER gives a rough idea of the impact.

jziolkowski, 2019-09-27:
You can't have everything, huh? :)

I assume there is no way to have a "lightweight" webserver for emulations?

jziolkowski, 2019-09-27:
However, what about the core of the issue? Sending log data throught MQTT?

arendst, 2019-09-27:
Log data must be possible is most likely the only info currently not send through MQTT.

What MQTT tool are you planning to use to handle these messages? I use Mqtt-Spy and it's not up to it.

jziolkowski, 2019-09-27:
My application, TDM: https://github.com/jziolkowski/tdm
Under heavy development now.

[image]

arendst, 2019-09-27:
How about piping/publishing the console output to a topic called console. 

So if a normal topic would be ``stat/sonoff1/power`` it would send to ``stat/sonoff1/console``.  

Or worse ``cons/sonoff1`` which breaks with previous syntax...

arendst, 2019-09-27:
Well it's actually logging output that will be piped/published. All other messages or send to mqtt anyway.

arendst, 2019-09-27:
A matter of adding MqttLog to SeriaLog, Syslog and WebLog...

shantur, 2019-09-27:
I would prefer `stat/sonoff1/console`

arendst, 2019-09-27:
But it's not really a console as it will only pipe logging information.

jziolkowski, 2019-09-27:
My thoughts exactly. MQTTLog with log levels like other commands.

Console sounds better, but indeed it is log output, so stat/sonoff/log makes more sense

arendst, 2019-09-27:
I'll make a POC

jziolkowski, 2019-09-27:
So if I understand correctly, on this topic would be only the log data, not the command outputs etc?

arendst, 2019-09-27:
Correct. The command outputs were always send as MQTT and will continue to do so.

jziolkowski, 2019-09-27:
That was quick :) Slow day? ;)

ascillato2, 2019-10-02:
Closing this issue as the feature has been added. Thanks :)

jziolkowski, 2019-10-02:
No, this is not added :) Theo only added some boilerplate code and moved on to bigger topics. But the feature is not there.

arendst, 2019-10-02:
This is all you will get.

jziolkowski, 2019-10-02:
Oh damn, I tested this earlier and it wasn't working. My bad, sorry Theo.

It seems it works out of the box :) Fantastic, thank you.

[image]

jziolkowski, 2019-10-03:
One thing is missing: MQTTLog in STATUS3

jziolkowski, 2019-10-04:
Thanks a lot!

closed 2019-10-04 by jziolkowski