arendst/Tasmota #6499: enabling mqtt stops serialsend to work

fragolinux, 2019-09-27:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
### BUG DESCRIPTION
_i've one of these LCTECH 2 relays module, configured, put in rules as per guide down here, all fine... https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay

i put pulsetime 10 on both relays (i want to connect it to a remote to open a gate and a lamp), and still no problems... as soon as i put in mqtt server and user info and the devices reboots, relays stop to work... i see in console it sends the serial commands just as before, and i even try them even manually always in console, no way, i've to revert to a previous config backup to have relays working again... even removing mqtt info or disabling it completely will not restore relays functionality, only restore config or full reconfigure after "reset 0" works... I'm on tasmota 6.6.0.14 and tried 6.6.0 too_

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): LCTECH 2 relay board
- [X] Tasmota binary firmware version number used: 6.6.0 and 6.6.0.14
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: espeasy flasher
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
15:06:10 CMD: Backlog Template; Module; GPIO
15:06:11 MQT: stat/cancello/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:06:11 MQT: stat/cancello/RESULT = {"Module":"0 (Generic)"}
15:06:11 MQT: stat/cancello/RESULT = {"GPIO0":"21 (Relay1)","GPIO1":"0 (None)","GPIO2":"22 (Relay2)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO9":"0 (None)","GPIO10":"0 (None)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  ```
- [X] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
15:07:13 CMD: Backlog Rule1; Rule2; Rule3
15:07:13 MQT: stat/cancello/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":378,"Rules":"on System#Boot do Baudrate 9600 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon"}
15:07:13 MQT: stat/cancello/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":416,"Rules":"on Power2#State=1 do SerialSend5 A00201A3 endon on Power2#State=0 do SerialSend5 A00200A2 endon"}
15:07:14 MQT: stat/cancello/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  ```
- [X] Provide the output of this command: ``Status 0``:
  ```
15:06:25 CMD: status 0
15:06:25 MQT: stat/cancello/STATUS = {"Status":{"Module":0,"FriendlyName":["cancello","faro"],"Topic":"cancello","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:06:25 MQT: stat/cancello/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:56","StartupUTC":"2019-09-27T13:40:29","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":23,"SaveAddress":"F5000"}}
15:06:25 MQT: stat/cancello/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:06:25 MQT: stat/cancello/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shark-HOME2","SharkAzzo2"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00C08009","280500000100060000000000000000000000","00000000"]}}
15:06:25 MQT: stat/cancello/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
15:06:26 MQT: stat/cancello/STATUS5 = {"StatusNET":{"Hostname":"cancello-0726","IPAddress":"172.30.0.131","Gateway":"172.30.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.4","Mac":"84:0D:8E:A7:E2:D6","Webserver":2,"WifiConfig":4}}
15:06:26 MQT: stat/cancello/STATUS6 = {"StatusMQT":{"MqttHost":"mrshark.ddns.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A7E2D6","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:06:26 MQT: stat/cancello/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 27 14:06:26 2019","Local":"Fri Sep 27 15:06:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:43","Sunset":"18:38"}}
15:06:26 MQT: stat/cancello/STATUS10 = {"StatusSNS":{"Time":"2019-09-27T15:06:26"}}
15:06:26 MQT: stat/cancello/STATUS11 = {"StatusSTS":{"Time":"2019-09-27T15:06:26","Uptime":"0T00:25:57","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Shark-HOME2","BSSId":"E4:8D:8C:CA:F7:0D","Channel":1,"RSSI":98,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
15:08:11 CMD: weblog 4
15:08:11 MQT: stat/cancello/RESULT = {"WebLog":4}
15:08:11 CFG: Saved to flash at F4, Count 24, Bytes 3584
15:08:16 CMD: power1 1
15:08:16 SRC: WebConsole from 10.0.0.27
15:08:16 RSL: Received Topic /power1, Data Size 1, Data 1
15:08:16 RSL: Group 0, Index 1, Command POWER, Data 1
15:08:16 MQT: stat/cancello/RESULT = {"POWER1":"ON"}
15:08:16 MQT: stat/cancello/POWER1 = ON
15:08:16 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
15:08:16 SRC: Rule
15:08:16 RSL: Received Topic /SerialSend5, Data Size 8, Data A00101A2
15:08:16 RSL: Group 0, Index 5, Command SERIALSEND, Data A00101A2
15:08:16 MQT: stat/cancello/RESULT = {"SerialSend":"Done"}
15:08:16 SRC: PulseTimer
15:08:16 MQT: stat/cancello/RESULT = {"POWER1":"OFF"}
15:08:17 MQT: stat/cancello/POWER1 = OFF
15:08:17 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
15:08:17 SRC: Rule
15:08:17 RSL: Received Topic /SerialSend5, Data Size 8, Data A00100A1
15:08:17 RSL: Group 0, Index 5, Command SERIALSEND, Data A00100A1
15:08:17 MQT: stat/cancello/RESULT = {"SerialSend":"Done"}
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
i just followed guide on 1st link in this issue (plus pulsetime set to 7 for both relays), it all works until i put in mqtt server, user and password, and even removing this infos would not restore relays functionality, only a config restore would do...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
just work as before, and sending out mqtt info and react to mqtt IN messages...

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-10-03:
Hi,

Sorry, I couldn't reproduce your issue. Please, can you update your device to latest Tasmota version http://thehackbox.org/tasmota/pre-2.6/sonoff.bin ?

fragolinux, 2019-10-03:
done, exactly same behaviour... tell what you need as log, but i started with 6.6.0, upgraded to minimal, then to the firmware you said, tested relays, they worked... put in mqtt host, user, pass and topic, it rebooted, stop, relays don't trigger anymore, even though in console i can see the commands received, it publishes topics to my mqtt server and reacts to commands sent via mqtt, i see it shows the serialsend, but no relays clicking sound at all...

`18:42:29 MQT: stat/cancello/STATUS = {"Status":{"Module":0,"FriendlyName":["cancello","faro"],"Topic":"cancello","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:42:29 MQT: stat/cancello/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:42","StartupUTC":"2019-10-03T17:37:47","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":35,"SaveAddress":"F7000"}}
18:42:29 MQT: stat/cancello/STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(3db6f97-sonoff)","BuildDateTime":"2019-10-03T16:10:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
18:42:29 MQT: stat/cancello/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Shark-HOME2","SharkAzzo2"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00C08009","280500000100060000005AFF000000000000","00000000"]}}
18:42:29 MQT: stat/cancello/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:42:29 MQT: stat/cancello/STATUS5 = {"StatusNET":{"Hostname":"cancello-0726","IPAddress":"192.168.1.118","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:A7:E2:D6","Webserver":2,"WifiConfig":4}}
18:42:29 MQT: stat/cancello/STATUS6 = {"StatusMQT":{"MqttHost":"mrshark.ddns.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A7E2D6","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:42:29 MQT: stat/cancello/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 03 17:42:29 2019","Local":"Thu Oct 03 18:42:29 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:52","Sunset":"18:26"}}
18:42:29 MQT: stat/cancello/STATUS10 = {"StatusSNS":{"Time":"2019-10-03T18:42:29"}}
18:42:29 MQT: stat/cancello/STATUS11 = {"StatusSTS":{"Time":"2019-10-03T18:42:29","Uptime":"0T00:04:42","UptimeSec":282,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Shark-HOME2","BSSId":"74:36:6D:30:9C:69","Channel":4,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:07"}}}
18:42:38 MQT: stat/cancello/RESULT = {"POWER1":"ON"}
18:42:38 MQT: stat/cancello/POWER1 = ON
18:42:38 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
18:42:38 MQT: stat/cancello/RESULT = {"SerialSend":"Done"}
18:42:38 MQT: stat/cancello/RESULT = {"POWER1":"OFF"}
18:42:38 MQT: stat/cancello/POWER1 = OFF
18:42:38 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
18:42:38 MQT: stat/cancello/RESULT = {"SerialSend":"Done"}`

meingraham, 2019-10-03:
@fragolinux 

Your GPIO configuration doesn't show and Tx/Rx components configured.

fragolinux, 2019-10-07:
oh, don't know how to do that... the module is the one in 1st link i put, it's premade, if i go in device config in tasmota web gui i see relay1 on gpio0 and relay2 on gpio2, that's it, all the remaining is done via rules, as in that guide... and the serial communication just works, but only till i put in the mqtt info...

meingraham, 2019-10-07:
@fragolinux 

You followed this procedure? 
https://github.com/arendst/Sonoff-Tasmota/wiki/LC-Technology-WiFi-Relay

fragolinux, 2019-10-07:
@meingraham that's exactly the guide i followed, as stated in opening post... and, i repeat, that's just RIGHT, relays work as expected... unless you put in mqtt info...

meingraham, 2019-10-07:
The template you posted: `{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}`

The template **_in the article_**:
`{"NAME":"LC Technology 4CH Relay","GPIO":[52,255,17,255,255,255,255,255,21,22,23,24,255],"FLAG":0,"BASE":18}`

The relays are `21,22,23,24`

In your case you should two relays `21,22,0,0`

fragolinux, 2019-10-07:
how is it possible the template command does not report the correct gpios? I setup them via the CONFIGURE MODULE, not the CONFIGURE TEMPLATE menu option...

[image]

[image]

[image]

fragolinux, 2019-10-07:
i run the "template 255" right now, just in case... i find confusing that there are 2 areas pretty much similar to do same things, why not combining them? Anyway, this didn't solve...

```
21:51:43 CMD: template
21:51:43 MQT: stat/cancello/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:56:10 CMD: template 255
21:56:10 MQT: stat/cancello/RESULT = {"NAME":"Merged","GPIO":[21,255,22,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:57:58 CMD: template
21:57:58 MQT: stat/cancello/RESULT = {"NAME":"Merged","GPIO":[21,255,22,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

```

meingraham, 2019-10-07:
You have modules and templates crossed... but I don't believe this is what's causing the issue.

Let's start over. Perform `Reset 6`. Then go to `Configure`, `Configure Module`, and select module `Generic (18)` and save. Once it restarts, go back to configure module and set GPIO0 & GPIO2 as your relays. and save. Then enter the rule and enable the rule.

Let's start with that and see what happens when you have MQTT configured and you try toggling the relays.

fragolinux, 2019-10-07:
i already did that EXACTLY, as you (or Jason) suggested me on discord about 10 days ago, before opening this issue, as you said at that time :)
and i redid that other times (changing various firmware versions): all works until i put in mqtt info :(

meingraham, 2019-10-07:
OK... I correspond with dozens of users. Sometimes my old mind can't keep up.

But your configuration is still crossed. You have a Module 0 (which comes from having a template). I'm suggesting eliminating at least one variable.

Then my next question is trying to understand better why you believe SerialSend stops working. It looks like it is ("Done"). But what is not working is that the relay is not responding to the SerialSend. I just need for us to be explicit and specific. Remember, I can't see your environment so you need to be as detailed and specific in "painting the picture" for me so that I can "see" it and try to suggest next steps.

fragolinux, 2019-10-07:
sure, tomorrow i'll redo everything again, no problem... thanks for your help :)

yes, that's correct, what's not working is the relay not responding to serial commands, true, sorry if i reported it wrongly... if there's a more complete log i can share, just tell me how

fragolinux, 2019-10-08:
so: i did "reset 5" (as reset 6 will keep mqtt info, and i've to get rid of them)
then:
```
backlog module 0; gpio0 21; gpio2 22; seriallog 0
Rule1 on System#Boot do Backlog Baudrate 9600; SerialSend5 0 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon on Power2#State=1 do SerialSend5 A00201A3 endon on Power2#State=0 do SerialSend5 A00200A2 endon
backlog rule1 1; pulsetime1 7; pulsetime2 7
```
all working, on firmware 6.6.0.15 (the 2 pulsetime are my additions, to just turn relays on for less than a second)... then i put in mqtt server, used and pass (which are correct, i can see on my broker device registering), and that's the end of my module working... i can see the toggle buttons act as expected, turning on for a fraction of a second, and serial info sent out, but this time they do not trigger the relay as before....
```
12:21:52 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
12:21:52 MQT: stat/sonoff/POWER1 = ON
12:21:52 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
12:21:52 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
12:21:53 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
12:21:53 MQT: stat/sonoff/POWER1 = OFF
12:21:53 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
12:21:53 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
12:21:55 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
12:21:55 MQT: stat/sonoff/POWER2 = ON
12:21:55 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
12:21:55 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
12:21:56 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
12:21:56 MQT: stat/sonoff/POWER2 = OFF
12:21:56 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
12:21:56 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
```

meingraham, 2019-10-08:
@fragolinux 

I edited the article and consolidated it into one as it all fits in one rule set. Simplifies a bit.

Can you turn on `weblog 4` and capture the logs again?

Thanks.

Mike

fragolinux, 2019-10-08:
done (after putting in mqtt info):
```
14:25:55 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
14:25:55 MQT: stat/sonoff/POWER1 = ON
14:25:55 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
14:25:55 SRC: Rule
14:25:55 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00101A2"
14:25:55 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
14:25:56 SRC: PulseTimer
14:25:56 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
14:25:56 MQT: stat/sonoff/POWER1 = OFF
14:25:56 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
14:25:56 SRC: Rule
14:25:56 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00100A1"
14:25:56 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
14:26:01 WIF: Checking connection...
14:26:01 WIF: Connected
14:26:06 SRC: WebGui from 10.0.0.27
14:26:06 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
14:26:06 MQT: stat/sonoff/POWER2 = ON
14:26:06 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
14:26:06 SRC: Rule
14:26:06 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00201A3"
14:26:06 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
14:26:07 SRC: PulseTimer
14:26:07 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
14:26:07 MQT: stat/sonoff/POWER2 = OFF
14:26:07 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
14:26:07 SRC: Rule
14:26:07 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00200A2"
14:26:07 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
14:26:21 WIF: Checking connection...
14:26:21 WIF: Connected
```

fragolinux, 2019-10-08:
done, this is after reset 5, put gpio in, disabled seriallog, enter and enabled rule, stop, no other mods other than "weblog 4" as requested, and relays are working now (no mqtt yet, obviously):

1. 15:38:48 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":259,"Rules":"on System#Boot do Backlog Baudrate 9600; SerialSend5 0 endon on Power1#State=1 do SerialSend5 A00101A2 endon on Power1#State=0 do SerialSend5 A00100A1 endon on Power2#State=1 do SerialSend5 A00201A3 endon on Power2#State=0 do SerialSend5 A00200A2 endon"}
2. 15:38:51 CMD: weblog 4
3. 15:38:51 RSL: stat/sonoff/RESULT = {"WebLog":4}
4. 15:38:52 CFG: Saved to flash at FA, Count 10, Bytes 4096
5. 15:38:54 WIF: Checking connection...
6. 15:38:54 WIF: Connected
7. 15:38:56 HTP: Main Menu
8. 15:38:58 SRC: WebGui from 10.0.0.27
9. 15:38:58 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
10. 15:38:58 RSL: stat/sonoff/POWER1 = ON
11. 15:38:59 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
12. 15:38:59 SRC: Rule
13. 15:38:59 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00101A2"
14. 15:38:59 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
15. 15:38:59 CFG: Saved to flash at F9, Count 11, Bytes 4096
16. 15:39:01 SRC: WebGui from 10.0.0.27
17. 15:39:01 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
18. 15:39:01 RSL: stat/sonoff/POWER1 = OFF
19. 15:39:02 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
20. 15:39:02 SRC: Rule
21. 15:39:02 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00100A1"
22. 15:39:02 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
23. 15:39:02 CFG: Saved to flash at F8, Count 12, Bytes 4096
24. 15:39:03 SRC: WebGui from 10.0.0.27
25. 15:39:03 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
26. 15:39:03 RSL: stat/sonoff/POWER2 = ON
27. 15:39:03 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
28. 15:39:03 SRC: Rule
29. 15:39:03 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00201A3"
30. 15:39:03 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
31. 15:39:04 CFG: Saved to flash at F7, Count 13, Bytes 4096
32. 15:39:06 SRC: WebGui from 10.0.0.27
33. 15:39:06 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
34. 15:39:06 RSL: stat/sonoff/POWER2 = OFF
35. 15:39:06 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
36. 15:39:06 SRC: Rule
37. 15:39:06 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00200A2"
38. 15:39:06 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
39. 15:39:06 CFG: Saved to flash at F6, Count 14, Bytes 4096
40. 15:39:14 WIF: Checking connection...
41. 15:39:14 WIF: Connected

meingraham, 2019-10-08:
OK, so the power action occurs and Tasmota is clearly saving the state (several strewn throughout the second log)
`15:38:59 CFG: Saved to flash at F9, Count 11, Bytes 4096`

And these are not present when MQTT is active.

OK, I think we've given the investigation a thorough run through. It's time for @arendst to take a look.

Mike

arendst, 2019-10-08:
Nice excercise Mike and @fragolinux but alas, I cannot reproduce.

Your latest conclusion regarding the ``Saved to flash`` message is caused by the short pulsetime. If the relay is toggled within a second the end result is no change within a second and the settings update code won't find a difference in settings and won't save anything to flash. So @fragolinux failed to set pulsetime in the last weblog 4 dump.

I have no LCTEch relay so I can only monitor the serial traffic from the Tx/Rx line and in my case the codes happily appear with or without MQTT connected:

```
18:17:53 WIF: Checking connection...
18:17:53 WIF: Connected
18:18:08 CMD: power2 2
18:18:08 SRC: WebConsole from 192.168.2.1
18:18:08 MQT: stat/wemos4/RESULT = {"POWER2":"ON"}
18:18:08 MQT: stat/wemos4/POWER2 = ON
18:18:08 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
18:18:08 SRC: Rule
18:18:08 MQT: stat/wemos4/RESULT = {"SerialSend":"Done"}
18:18:09 SRC: PulseTimer
18:18:09 MQT: stat/wemos4/RESULT = {"POWER2":"OFF"}
18:18:09 MQT: stat/wemos4/POWER2 = OFF
18:18:09 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
18:18:09 SRC: Rule
18:18:09 MQT: stat/wemos4/RESULT = {"SerialSend":"Done"}
18:18:13 WIF: Checking connection...
18:18:13 WIF: Connected
```
and serial:
```
a0 02 01 a3                                                                                       ..£                            
a0 02 00 a2                                                                                       ..¢                            
```

fragolinux, 2019-10-08:
i'll try again with mqtt and no pulsetime (which is missing in latest weblog as per precise request @meingraham ), and as you can see in my post of a few hours ago, even there serial output is sent out: https://github.com/arendst/Sonoff-Tasmota/issues/6499#issuecomment-539468749

fragolinux, 2019-10-08:
here it is, latest run, before and after setting mqtt info, with weblog 4 and no pulsetime, i waited "manually" to turn relays on and off a few seconds between the 2 states:

before:
```
17:32:30 SRC: WebGui from 192.168.1.57
17:32:30 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
17:32:30 RSL: stat/sonoff/POWER1 = ON
17:32:30 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
17:32:30 SRC: Rule
17:32:30 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00101A2"
17:32:30 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
17:32:30 CFG: Saved to flash at F6, Count 22, Bytes 4096
17:32:32 WIF: Checking connection...
17:32:32 WIF: Connected
17:32:33 SRC: WebGui from 192.168.1.57
17:32:33 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
17:32:33 RSL: stat/sonoff/POWER2 = ON
17:32:33 RUL: POWER2#STATE=1 performs "SerialSend5 A00201A3"
17:32:33 SRC: Rule
17:32:33 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00201A3"
17:32:33 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
17:32:33 CFG: Saved to flash at F5, Count 23, Bytes 4096
17:32:36 SRC: WebGui from 192.168.1.57
17:32:36 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
17:32:36 RSL: stat/sonoff/POWER1 = OFF
17:32:36 RUL: POWER1#STATE=0 performs "SerialSend5 A00100A1"
17:32:36 SRC: Rule
17:32:36 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00100A1"
17:32:36 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
17:32:36 CFG: Saved to flash at F4, Count 24, Bytes 4096
17:32:39 SRC: WebGui from 192.168.1.57
17:32:39 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
17:32:39 RSL: stat/sonoff/POWER2 = OFF
17:32:39 RUL: POWER2#STATE=0 performs "SerialSend5 A00200A2"
17:32:39 SRC: Rule
17:32:39 CMD: Group 0, Index 5, Command "SERIALSEND", Data "A00200A2"
17:32:39 RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
17:32:39 CFG: Saved to flash at FB, Count 25, Bytes 4096
17:32:45 CMD: status 0
17:32:45 SRC: WebConsole from 192.168.1.57
17:32:45 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:32:45 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:32:45 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:01:42","StartupUTC":"2019-10-08T16:31:03","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":25,"SaveAddress":"FB000"}}
17:32:45 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(3db6f97-sonoff)","BuildDateTime":"2019-10-03T16:10:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
17:32:45 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*********","*********"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","280500000100060000005A64000000000000","00000000"]}}
17:32:45 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:32:45 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0726","IPAddress":"192.168.1.109","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:A7:E2:D6","Webserver":2,"WifiConfig":4}}
17:32:45 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A7E2D6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:32:45 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 08 16:32:45 2019","Local":"Tue Oct 08 17:32:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:59","Sunset":"18:15"}}
17:32:45 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-08T17:32:45"}}
17:32:45 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-08T17:32:45","Uptime":"0T00:01:42","UptimeSec":102,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*********","BSSId":"74:36:6D:30:9C:69","Channel":13,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}}
```

after:
```
17:33:45 RUL: SYSTEM#BOOT performs "Backlog Baudrate 9600; SerialSend5 0"
17:33:45 SRC: Rule
17:33:45 CMD: Group 0, Index 1, Command "BACKLOG", Data "Baudrate 9600; SerialSend5 0"
17:33:45 SRC: Backlog
17:33:45 CMD: Group 0, Index 1, Command "BAUDRATE", Data "9600"
17:33:46 MQT: stat/sonoff/RESULT = {"Baudrate":9600}
17:33:46 SRC: Backlog
17:33:46 CMD: Group 0, Index 5, Command "SERIALSEND", Data "0"
17:33:46 MQT: stat/sonoff/RESULT = {"SerialSend":"Done"}
17:33:46 APP: Boot Count 4
17:33:46 CFG: Saved to flash at F7, Count 29, Bytes 4096
17:33:53 SRC: WebGui from 192.168.1.57
17:33:53 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
17:33:53 MQT: stat/sonoff/POWER1 = ON
17:33:53 RUL: POWER1#STATE=1 performs "SerialSend5 A00101A2"
[... 20 lines cut ...]
17:34:00 CFG: Saved to flash at F4, Count 32, Bytes 4096
17:34:03 SRC: WebGui from 192.168.1.57
17:34:03 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
17:34:03 MQT: stat/sonoff/POWER2 = OFF
[... cut ...]
17:34:11 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 08 16:34:11 2019","Local":"Tue Oct 08 17:34:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:59","Sunset":"18:15"}}
17:34:11 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-08T17:34:11"}}
17:34:11 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-08T17:34:11","Uptime":"0T00:00:36","UptimeSec":36,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*********","BSSId":"74:36:6D:30:9C:69","Channel":13,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```

arendst, 2019-10-08:
Can you explain the correct hardware setup? How do you power the relays?

fragolinux, 2019-10-08:
just for test, please be patient as i want to help in debug this, i tried espurna on same device (the stm relay version), and setup up both pulsetime of 0.7 and mqtt and it works, so it's not a problem of the device itself... just i'd like to have everything with a single firmware for easier maintenance... i can send and receive mqtt data from nodered and all info are received by both firmwares, but with tasmota it does not trigger the relays
[image]

fragolinux, 2019-10-08:
> Can you explain the correct hardware setup? How do you power the relays?

sure :)
there's a 5.5mm female barrel connector to the 2 power inputs terminals, and i use an usb 2 male barrel connector adapter to power it, as in image... the 2 esp-01 are flashed 1 with tasmota and 1 with espurna, just for easy test

[image]

arendst, 2019-10-08:
And the power is coming from?

fragolinux, 2019-10-08:
now from a 6 port 40w shared usb multiport (2.4A max per single port, only 3 devices connected), a few hours ago from a 2A usb power adapter

Jason2866, 2019-10-09:
Mhhh, just to rule out gpio 0 and gpio 2 issues. Since the assigned relais are just virtual dummy ones...
Assign relais1 to Gpio 12 and relais2 to Gpio 13.  Nothing else assigned to any other Gpio.
Is the behaviour the same?

Jason2866, 2019-10-09:
Out of ideas... Since i dont have the device, cant help. Tried same as Theo did, worked for me too.

fragolinux, 2019-10-09:
i know, did the same on a wemos and all fine there... as said, serial commands are sent out but not executed anymore on my device...

well, if anybody has more ideas, even in the future and want me to test on the real device, let me know...

i'll just swap the esp01 module, but for now i'll go with espurna for this device: i want to connect it to a rolling code remote to trigger remotely a flood lamp and the motorized gate where i leave my car...

thanks for everything guys, help much appreciated :)

closed 2019-10-09 by ascillato2

fragolinux, 2019-11-16:
just to update, i retested everything with tasmota.bin 7.0.0.4, same behaviour as before: all working till i put in mqtt info...

fragolinux, 2019-11-16:
just to update, i retested everything with tasmota.bin 7.0.0.4, same behaviour as before: all working till i put in mqtt info... @Jason2866

dmelliott20, 2019-12-21:
@fragolinux Did you get to the bottom of this, I'm having same issue with the same hardware

oscill8ory, 2020-01-04:
I'm experiencing this issue with a 4CH module as well.

oscill8ory, 2020-01-04:
> I'm experiencing this issue with a 4CH module as well. Running Tasmota 8.1.0.

Actually, I think I've resolved it now. Setting the baud rate to 115200 on startup (instead of 9600 suggested in the rules [here](https://github.com/arendst/Tasmota/wiki/LC-Technology-WiFi-Relay)) seems to have fixed it.

fragolinux, 2020-01-11:
confirmed: changing baud rate from 9600 to 115200 solved the mqtt issues, device is working good, using tasmota-lite.bin file :)