arendst/Tasmota #6500: Energy metering always reports off-peak power when using tariffs

znanev, 2019-09-27:
### BUG DESCRIPTION
Energy metering always reports off-peak power when using tariffs. Total power is OK, but standard tariff is always 0 (irrespective of time of day or tariffs configured hours):

```"Total":[3.152,0.097,0.000]```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf SHP2
- [x] Tasmota binary firmware version number used: 6.6.0.14(2d1bc97-sonoff)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:

``` 
18:52:26 MQT: blitzwolf-shp2-2/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:52:26 MQT: blitzwolf-shp2-2/stat/RESULT = {"Module":{"45":"BlitzWolf SHP"}}
18:52:26 MQT: blitzwolf-shp2-2/stat/RESULT = {"GPIO1":{"0":"None"},"GPIO3":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules are NOT used
  ```
- [x] Provide the output of this command: ``Status 0``:
```
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS = {"Status":{"Module":45,"FriendlyName":["Back Light"],"Topic":"blitzwolf-shp2-2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/json/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T23:31:09","StartupUTC":"2019-09-26T18:23:22","Sleep":50,"CfgHolder":4617,"BootCount":25,"SaveCount":387,"SaveAddress":"F7000"}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.14(2d1bc97-sonoff)","BuildDateTime":"2019-09-25T16:02:36","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["**MASKED**",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["400A8009","280500000100060000005AFF000000170700","00000228"]}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS5 = {"StatusNET":{"Hostname":"blitzwolf-shp2-2-2797","IPAddress":"192.168.100.109","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"FF:FF:FF:FF:FF:FF","Webserver":2,"WifiConfig":5}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_33EAED","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 27 17:54:31 2019","Local":"Fri Sep 27 18:54:31 2019","StartDST":"Sun Mar 31 01:00:00 2019","EndDST":"Sun Oct 27 02:00:00 2019","Timezone":99,"Sunrise":"06:56","Sunset":"18:50"}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-09-27T18:54:31","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.152,0.097,0.000],"Yesterday":0.088,"Today":0.058,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:54:31 MQT: blitzwolf-shp2-2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-09-27T18:54:31","Uptime":"0T23:31:09","UptimeSec":84669,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}}
  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
07:23:37 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-27T07:23:37","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.110,0.056,0.000],"Yesterday":0.088,"Today":0.017,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
07:28:29 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T07:28:29","Uptime":"0T12:05:07","UptimeSec":43507,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:28:29 MQT: blitzwolf-shp2-2/stat/RESULT = {"POWER":"ON"}
07:28:29 MQT: blitzwolf-shp2-2/stat/POWER = ON
07:28:32 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-27T07:28:32","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.110,0.056,0.000],"Yesterday":0.088,"Today":0.017,"Power":11,"ApparentPower":11,"ReactivePower":0,"Factor":1.00,"Voltage":245,"Current":0.046}}
07:28:37 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T07:28:37","Uptime":"0T12:05:15","UptimeSec":43515,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:04"}}

07:29:31 CMD: tariff
07:29:31 MQT: blitzwolf-shp2-2/stat/RESULT = {"Tariff":{"Off-Peak":{"STD":"00:29","DST":"01:29"},"Standard":{"STD":"07:29","DST":"08:29"},"Weekend":"OFF"}}

...

07:33:37 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T07:33:37","Uptime":"0T12:10:15","UptimeSec":43815,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:33:37 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-27T07:33:37","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.112,0.058,0.000],"Yesterday":0.088,"Today":0.018,"Period":2,"Power":20,"ApparentPower":32,"ReactivePower":25,"Factor":0.63,"Voltage":244,"Current":0.130}}
07:38:37 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T07:38:37","Uptime":"0T12:15:15","UptimeSec":44115,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:38:37 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-27T07:38:37","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.113,0.059,0.000],"Yesterday":0.088,"Today":0.020,"Period":2,"Power":20,"ApparentPower":32,"ReactivePower":25,"Factor":0.64,"Voltage":243,"Current":0.132}}
07:43:37 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T07:43:37","Uptime":"0T12:20:15","UptimeSec":44415,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:28:37 MQT: blitzwolf-shp2-2/tele/STATE = {"Time":"2019-09-27T09:28:37","Uptime":"0T14:05:15","UptimeSec":50715,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"**MASKED**","BSSId":"FF:FF:FF:FF:FF:FF","Channel":4,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:04"}}
[... cut ...]
```
power on
```

3. Note the reported values in _Total_ for SENSOR reports:

```
"Total":[3.152,**0.097**,**0.000**]
```

The third value (for standard tariff - _tariff2_) is always 0, even when standard tariff is currently active.

### EXPECTED BEHAVIOUR
When standard tariff (_tariff2_) is active, third value in _"Total"_ should be incremented instead of second value.

### SCREENSHOTS
Not applicable

### ADDITIONAL CONTEXT
Access point name and MAC address have been masked for privacy purposes.

pablozg, 2019-09-29:
It's really strange, as you can see in my case the counters works fine:

````
07:59:51 MQT: Casa/PB/sdm120/stat/RESULT = {"Tariff":{"Off-Peak":{"STD":"22:00","DST":"23:00"},"Standard":{"STD":"12:00","DST":"13:00"},"Weekend":"OFF"}}
07:55:45 MQT: Casa/PB/sdm120/tele/SENSOR = {"Time":"2019-09-29T07:55:45","ENERGY":{"TotalStartTime":"2019-09-27T00:00:00","Total":[681.772,113.992,567.783],"Yesterday":6.371,"Today":2.444,"ExportActive":[160.947,70.433,90.514],"Period":1.04,"Power":275.10,"ApparentPower":284.93,"ReactivePower":77.60,"Factor":0.96,"Frequency":50.00,"Voltage":238.4,"Current":1.26,"ImportActive":681.773,"ImportReactive":247.692,"ExportReactive":7.542,"PhaseAngle":0.00}}
````

Can you provide the "EnergyReset" command output?
````
07:58:08 MQT: Casa/PB/sdm120/stat/RESULT = {"EnergyReset":{"Total":681.782,"Yesterday":6.371,"Today":2.454,"Usage":[113.881,567.783],"Export":[70.433,90.514]}}
````

znanev, 2019-09-29:
Strange, indeed. Here's the output from `EnergyReset`:

```
09:30:49 MQT: blitzwolf-shp2-2/stat/RESULT = {"EnergyReset":{"Total":3.328,"Yesterday":0.091,"Today":0.013,"Usage":[0.000,0.000],"Export":[0.000,0.000]}}
```

znanev, 2019-09-29:
In addition, here's the output of the telemetry message just before issuing the `EnergyReset` command above:

```
09:28:37 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-29T09:28:37","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.328,0.274,0.000],"Yesterday":0.091,"Today":0.013,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

Note `Today`'s value: 0.013

I turned the socket on with a load connected for some time now, here's the value of the same telemetry message some minutes later:

```
09:43:37 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-29T09:43:37","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.332,0.278,0.000],"Yesterday":0.091,"Today":0.017,"Period":2,"Power":20,"ApparentPower":32,"ReactivePower":24,"Factor":0.64,"Voltage":242,"Current":0.131}}
```

The value of `Today` is now 0.017, also note that the value for tariff1 in `Total` has been increased from 0.274 to 0.278, although the increase should have been for tariff2 now.

And the output from `EnergyReset` command again:

```
09:45:42 MQT: blitzwolf-shp2-2/stat/RESULT = {"EnergyReset":{"Total":3.333,"Yesterday":0.091,"Today":0.017,"Usage":[0.000,0.000],"Export":[0.000,0.000]}}
```

pablozg, 2019-09-29:
Can you do a restart using the button in webui after have some consumption and do again the EnergyReset command?

I say this because the EnergyReset command reads the stored values, and tasmota by default only store the energy values at midnight or after a restart.

znanev, 2019-09-29:
This is the output from the `EnergyReset` command, followed immediately after I restarted the device from the WebUI:

```
10:14:11 MQT: blitzwolf-shp2-2/stat/RESULT = {"EnergyReset":{"Total":3.335,"Yesterday":0.091,"Today":0.020,"Usage":[0.281,0.000],"Export":[0.000,0.000]}}
```

Also the telemetry message:

```
10:14:13 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-29T10:14:13","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.335,0.281,0.000],"Yesterday":0.091,"Today":0.020,"Period":0,"Power":21,"ApparentPower":32,"ReactivePower":25,"Factor":0.63,"Voltage":244,"Current":0.133}}
```

Usage is in line with what is seen in "Total", but again it only uses tariff1 (the active tariff now should be tariff2). This is my current tariff configuration:

```
10:16:44 MQT: blitzwolf-shp2-2/stat/RESULT = {"Tariff":{"Off-Peak":{"STD":"00:29","DST":"01:29"},"Standard":{"STD":"07:29","DST":"08:29"},"Weekend":"OFF"}}
```

And local time is now 10:16 BST (British Summer Time - UTC+1)

arendst, 2019-09-29:
It might be there is in error in calculating the off-peak period; in holland it starts at 23:00, at your location it's after 00:00.

I'll have to check the formula used but I'm quite sure something is wrong there.

To be continued.

arendst, 2019-09-29:
Should be fixed now.

znanev, 2019-09-29:
That was quick, Theo, thanks!

Just updated the firmware to commit `4c21e46`, here is the output of the telemetry message and also `EnergyReset` command when the device restarted:

```
12:19:00 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-29T12:19:00","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.355,0.301,0.000],"Yesterday":0.091,"Today":0.040,"Power":19,"ApparentPower":30,"ReactivePower":23,"Factor":0.63,"Voltage":244,"Current":0.122}}
12:19:02 MQT: blitzwolf-shp2-2/stat/RESULT = {"EnergyReset":{"Total":3.355,"Yesterday":0.091,"Today":0.040,"Usage":[0.301,0.000],"Export":[0.000,0.000]}}
```

Turned on the socket and here is the output now:

```
12:27:36 MQT: blitzwolf-shp2-2/tele/SENSOR = {"Time":"2019-09-29T12:27:36","ENERGY":{"TotalStartTime":"2019-06-26T21:16:03","Total":[3.358,0.301,0.003],"Yesterday":0.091,"Today":0.043,"Period":2,"Power":21,"ApparentPower":32,"ReactivePower":25,"Factor":0.63,"Voltage":243,"Current":0.133}}
12:27:37 MQT: blitzwolf-shp2-2/stat/RESULT = {"EnergyReset":{"Total":3.358,"Yesterday":0.091,"Today":0.043,"Usage":[0.301,0.000],"Export":[0.000,0.000]}}
```

So the electricity consumed is definitely reported on the standard (tariff2) tariff (0.003 so far). I'll do another quick test tomorrow when the off-peak tariff kicks in and will report back.

znanev, 2019-09-30:
OK, just tested the off-peak tariff and it reports as expected too.

Now that off-peak tariff is no longer active (after 8:29 local time), consumption is accounted on the standard tariff again, which is also the expected behaviour.

So, closing this issue now.

Thank you, Theo for the quick fix!

closed 2019-09-30 by znanev