arendst/Tasmota #6501: TreatLife Dimmer

jabbo357, 2019-09-28:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Treatlife dimmer does not go up to 100%.  Only 2 lights are on the faceplate when command "dimmer 100" is given.  The dimmer will dim down, but it is like its thinking that 100% is about 30% of lighting.  

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ___TreatLife Dimmer__
- [ x] Tasmota binary firmware version number used: __6.6.0___
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ x] IDE / Compiler used: __Tuya-Convert___
- [x ] Flashing tools used: __Tuya-Convert___
- [ x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  02:16:55 CMD: Backlog Template; Module; GPIO
02:16:55 SRC: WebConsole from 10.0.0.12
02:16:55 RSL: Received Topic /Backlog, Data Size 22, Data Template; Module; GPIO
02:16:55 RSL: Group 0, Index 1, Command BACKLOG, Data Template; Module; GPIO
02:16:56 SRC: Backlog
02:16:56 RSL: Received Topic /Template, Data Size 0, Data 
02:16:56 RSL: Group 0, Index 1, Command TEMPLATE, Data 
02:16:56 MQT: stat/first_set_of_lights/RESULT = {"NAME":"TreatLife","GPIO":[255,255,255,255,255,255,0,0,255,255,255,255,255],"FLAG":0,"BASE":54}
02:16:56 SRC: Backlog
02:16:56 RSL: Received Topic /Module, Data Size 0, Data 
02:16:56 RSL: Group 0, Index 1, Command MODULE, Data 
02:16:56 MQT: stat/first_set_of_lights/RESULT = {"Module":"54 (Tuya Dimmer)"}
02:16:56 SRC: Backlog
02:16:56 RSL: Received Topic /GPIO, Data Size 0, Data 
02:16:56 RSL: Group 0, Index 1, Command GPIO, Data 
02:16:56 MQT: stat/first_set_of_lights/RESULT = {"GPIO0":"0 (None)","GPIO1":"107 (Tuya Tx)","GPIO2":"0 (None)","GPIO3":"108 (Tuya Rx)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  02:17:58 CMD: Backlog Rule1; Rule2; Rule3
02:17:58 SRC: WebConsole from 10.0.0.12
02:17:58 RSL: Received Topic /Backlog, Data Size 19, Data Rule1; Rule2; Rule3
02:17:58 RSL: Group 0, Index 1, Command BACKLOG, Data Rule1; Rule2; Rule3
02:17:58 SRC: Backlog
02:17:58 RSL: Received Topic /Rule1, Data Size 0, Data 
02:17:58 RSL: Group 0, Index 1, Command RULE, Data 
02:17:58 MQT: stat/first_set_of_lights/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
02:17:58 SRC: Backlog
02:17:58 RSL: Received Topic /Rule2, Data Size 0, Data 
02:17:58 RSL: Group 0, Index 2, Command RULE, Data 
02:17:58 MQT: stat/first_set_of_lights/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
[... 161 lines cut ...]
02:39:22 RSL: Group 0, Index 1, Command DIMMER, Data 91
02:39:22 MQT: stat/first_set_of_lights/RESULT = {"POWER1":"ON","Dimmer":91}
02:39:22 TYA: Send dim skipped value=197
02:39:22 TYA: RX Packet: "55aa0307000802020004000003e804"
02:39:22 TYA: RX Dim State=232
02:39:22 TYA: RX Packet: "55aa0307000802020004000003e804"
02:39:22 TYA: RX Dim State=232
02:39:23 CFG: Saved to flash at F4, Count 51, Bytes 3584
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

When using dimmer 100, the dimmer should make the light go 100%

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-28:
Use latest development version Tuya handling has completely changed since 6.6.0
What you try is possible.
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

andrethomas2, 2019-09-28:
Closing this issue as it has been answered. Please refer to additional links for support provided in previous post(s).

closed 2019-09-28 by andrethomas2

nass623, 2019-10-10:
I have the same issue reported here but with a moes ds01 dimmer. I think it is a rebrand of this dimmer. I am trying to find out how the issue was solved as I have tried different versions of the bin files and I cannot get the dimmer to properly set the brightness level. I have looked online and cannot find what the solution is. Can you please help me? Thanks.

broken, 2019-11-11:
@jabbo357 @nass623 I have the same problem. Did either of you get this working?