arendst/Tasmota #6505: Wrong output index after updating from 6.5.0 (Release) to 6.6.0 (Release)

AIexBV, 2019-09-28:
### BUG DESCRIPTION
I am using this item [RGB Plug](https://blakadder.github.io/templates/rgbpow.html) and my usecase is very similar to that one [button-with-single-press-double-press-and-hold](https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#button-with-single-press-double-press-and-hold).

After updating from 6.5.0 (Release) to 6.6.0 (Release) the single press function stopped working, since it affects power2 instead of power1 suddenly.

### REQUESTED INFORMATION
The malfunction appears after setting Option11 from 0 to 1.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): see above
- [x] Tasmota binary firmware version number used: 6.6.0 rel.
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: n/a
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  `stat/sonoff/RESULT = {"NAME":"RGB Smart Plug","GPIO":[37,0,39,0,38,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}`
  Configuration output here:
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
12:24:24 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
12:24:24 RSL: stat/sonoff/POWER1 = OFF
12:24:25 APP: Button1 multi-press 1
12:24:25 SRC: Button
12:24:25 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
12:24:25 RSL: stat/sonoff/POWER1 = ON
12:24:37 CMD: SetOption1 1
12:24:37 RSL: stat/sonoff/RESULT = {"SetOption1":"ON"}
12:24:43 APP: Button1 multi-press 1
12:24:44 SRC: Button
12:24:44 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
12:24:44 RSL: stat/sonoff/POWER1 = OFF
12:24:44 APP: Button1 multi-press 1
12:24:45 SRC: Button
12:24:45 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
12:24:45 RSL: stat/sonoff/POWER1 = ON
12:25:14 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
12:25:14 RSL: stat/sonoff/POWER1 = OFF
12:25:16 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
12:25:16 RSL: stat/sonoff/POWER1 = ON
12:25:33 CMD: setoption11 1
12:25:33 RSL: stat/sonoff/RESULT = {"SetOption11":"ON"}
12:25:34 RSL: tele/sonoff/SENSOR = {"Time":"2019-09-28T12:25:34","ENERGY":{"TotalStartTime":"2019-09-28T12:07:49","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":231,"Current":0.000}}
12:25:38 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
12:25:38 RSL: stat/sonoff/POWER2 = ON
12:25:40 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
12:25:40 RSL: stat/sonoff/POWER2 = OFF

### EXPECTED BEHAVIOUR
The single press should affect power1, but not power2. If there is a intended change in the behavior, the Rule Cookbook should be updated.

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-28:
Please try latest development version. If this is a bug, there will be no change in release version
All enhancements and fixes are done in development. A development build will become the next release version.

AIexBV, 2019-09-28:
I tried the latest dev version already. Unfortunately no change.

Jason2866, 2019-09-28:
Mhh, there is no relay2 in this device. So you must have defined a virtual relays for some use case.
Led function and behaviour is changed since release 6.5.0. See release notes and changelog.
Wiki is always explaining latest development version.

AIexBV, 2019-09-29:
I did not define a virtual relay. I have just applied the matching template for that item and after that, set up the example mentioned above (single-/double-/long-press). Led´s are not involved at all in that example. So changes in Led function should not apply here.
Trying a "_reset 6_" and reconfigure didn´t help either.

After playing around, I found that skipping "_SetOption11 1_" from the example single- and double-press-function works again, but not the hold function. Watching the console, I see that the event "_button1#state=3_" will never be fired.

Any further ideas?

ascillato2, 2019-10-01:
In order to help you, you need to share more information. Please, post the output of `status 0` command

AIexBV, 2019-10-02:
06:59:08 CMD: Status 0
06:59:08 MQT: /MqttPlugLed2/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Heizluefter","Sonoff2"],"Topic":"MqttPlugLed2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"1T18:26:27","StartupUTC":"2019-09-30T11:32:41","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":110,"SaveAddress":"F6000"}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T14:26:16","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan_tt","wlan_72_tt5"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0002800B","140500000100060000000000000000000000","00000000"]}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":557,"Free":444,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS5 = {"StatusNET":{"Hostname":"MqttPlugLed2-6586","IPAddress":"192.168.1.142","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:23:19:BA","Webserver":2,"WifiConfig":4}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"raspi77","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2319BA","MqttUser":"DVES_USER","MqttCount":73,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 02 05:59:08 2019","Local":"Wed Oct 02 06:59:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:50","Sunset":"18:28"}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-02T06:59:08","ENERGY":{"TotalStartTime":"2019-09-28T15:13:01","Total":0.462,"Yesterday":0.141,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
06:59:08 MQT: /MqttPlugLed2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-02T06:59:08","Uptime":"1T18:26:27","Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Dimmer":0,"Color":"0,0,0","HSBColor":"0,0,0","Channel":[0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"wlan_tt","BSSId":"18:D6:C7:F9:D2:02","Channel":11,"RSSI":58,"LinkCount":69,"Downtime":"0T00:03:45"}}}

arendst, 2019-10-02:
This is an expected behaviour with latest version.

Code regarding single/double press has changed for devices with two outputs only to accomodate devices with only one button but two outputs (either one relay and one USB port (SHP5) or one relay and one multi color led (bilikay)).

ascillato2, 2019-10-02:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-02 by ascillato2

AIexBV, 2019-10-02:
@ascillato2 - Actually I don't see the answer above. What is the intended approach to make the hold function working again on this device?