arendst/Tasmota #6506: Wemos D1 Mini and HC SR501 PIR Motion Sensor should publisch state to mgtt broker like a Relais/ Power State

smartboart, 2019-09-28:
Hello,

i would like to publisch the state of the pir sensor connected to wemo d1 mini powered by tasmota 6.6.0 directly via mqtt to my broker ( iobroker ) without any time delay like Teleperiod..
I would like to pulish the states like the relais ( Power) states emidiatly without time delay...

Is that posible without marrying  the Sensor input to a relays output? ( like Switch 2 to Relay 2 without rule)

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- x[ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __WemoD1Mini___
- [x ] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [x ] Self-compiled
    - [ x] IDE / Compiler used: __USB Ftdi___
- [X ] Flashing tools used: _node mcu pi flasher_& Atom___
- [ x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
{"Module":"18 (Generic)"}

{"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"21 (Relay1)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"10 (Switch2)","GPIO15":"0 (None)","GPIO16":"0 (None)"}

- [ x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
MQT: stat/sonoff/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":399,"Rules":"on switch2#state=1 do publish stat/sonoff/PIR2 ON endon on switch2#state=0 do publish stat/sonoff/PIR2 OFF endon"}
  
  
- [x ] Provide the output of this command: ``Status 0``:
  STATUS 0 output here:
  
 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemo-Eingang"],"Topic":"sonoff","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}

- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  16:34:11 CMD: weblog 4
16:34:11 SRC: WebConsole from 192.168.1.27
16:34:11 RSL: Received Topic /weblog, Data Size 1, Data 4
16:34:11 RSL: Group 0, Index 1, Command WEBLOG, Data 4
16:34:11 MQT: stat/sonoff/RESULT = {"WebLog":4}
16:34:23 RUL: SWITCH2#STATE=1 performs "publish stat/sonoff/PIR2 ON"
16:34:23 SRC: Rule
16:34:23 RSL: Received Topic /publish, Data Size 19, Data stat/sonoff/PIR2 ON
16:34:23 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff/PIR2 ON
16:34:23 MQT: stat/sonoff/PIR2 = ON
16:34:28 RUL: SWITCH2#STATE=0 performs "publish stat/sonoff/PIR2 OFF"
16:34:28 SRC: Rule
16:34:28 RSL: Received Topic /publish, Data Size 20, Data stat/sonoff/PIR2 OFF
16:34:28 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff/PIR2 OFF
16:34:28 MQT: stat/sonoff/PIR2 = OFF
16:34:28 WIF: Checking connection...
16:34:28 WIF: Connected
16:34:48 WIF: Checking connection...
16:34:48 WIF: Connected

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-28:
Not without a rule, as you've done.

andrethomas, 2019-09-28:
Maybe exploring the various switchmode's as documented here https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches#switchmode could be useful - I use this to monitor the state of a magnetic door sensor.

Jason2866, 2019-09-28:
Take a look in wiki. This example you can easily change
https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#enable-a-pir-switch-only-at-night

smartboart, 2019-09-28:
> 
> 
> Maybe exploring the various switchmode's as documented here https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches#switchmode could be useful - I use this to monitor the state of a magnetic door sensor.

Sorry, this will not help, i think my goalwas not clear...It is working with my rule like i have shoen in my request, but it is not working emidiatly . It is telling my broaker the state only in the teleperiod time, not emediatly...Soit is no setupissue with the switchmode or switchtopic...
Myteleperiod is on standart value 300...Even 10  / 10 seconds would be to long for a motion detector... Soi nee a rule whicht publish the state tomy broker direct after motion detection...how can this rule look like?
Rule1 on switch1#state=1 do publish stat/sonoff/PIR1 ON endon on switch1#state=0 do publish stat/sonoff/PIR1 OFF endon
this is not doing what i expect. i need a detection directly in my mqtt broker. Like the information when i switch on a relay...this information i have also directlyin my broker..

So i would be very gratefull for a rule example

andrethomas, 2019-09-28:
> So i would be very gratefull for a rule example

Sorry, I only know the other way to do it :)

smartboart, 2019-09-28:
ok, this willwork with a dummy relais without using a gpio...So i have a seperate / one more state in iobroker .Asolution with the Switch State only would be nicer...But with this rule it works for me  after i have change the Switch state in allmy iobroker skripts to Power...

Jason2866, 2019-09-28:
Glad you find a working solution. There will be no change in Tasmota code.
Please close.
Thx!

andrethomas, 2019-09-28:
With this

[image]

and `switchmode1 1`

I get this when the signal goes LOW:

```
19:37:09 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
19:37:09 MQT: stat/sonoff/POWER = OFF
```

And this when the signal goes HIGH

```
19:37:12 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
19:37:12 MQT: stat/sonoff/POWER = ON
```

No rule necessary :)

smartboart, 2019-09-28:
> With this
>
> [image]
>
> and `switchmode1 1`
>
> I get this when the signal goes LOW:
>
> ```
> 19:37:09 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
> 19:37:09 MQT: stat/sonoff/POWER = OFF
> ```
>
> And this when the [...]

yes as a result in the gui / command window...But not at the MQTT Broker

andrethomas, 2019-09-28:
> yes as a result in the gui / command window...But not at the MQTT Broker

Then I am not sure why I notifications from my fridge being opened from my automation software listening to `stat/fridge1/RESULT` and send me a telegram when it gets `msg.payload.POWER === "OFF"` or `msg.payload.POWER === "ON"` ?

smartboart, 2019-09-28:
> Then I am not sure why I notifications from my fridge being opened from my automation software listening to `stat/fridge1/RESULT` and send me a telegram when it gets `msg.payload.POWER === "OFF"` or [...]

maybe it is an issue with the sonoff adapter from iobroker..i did not changed to the mqtt adapter so far, because of the changes i have to make in 100 of scripts...
But this info here is exactly what i was serarching for...

https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#distinguish-switch1-and-switch2-without-the-use-of-relay1-and-relay2

smartboart, 2019-09-28:
Thanks for help.. if anybody else has the same issue and find another solution, please let me know...

closed 2019-09-28 by smartboart