arendst/Tasmota #6508: Add `0x` to hex values of IrReceived output

CWempe, 2019-09-28:
**Is your feature request related to a problem? Please describe.**  
When trying to send IR commands with my Tuya IR blaster it turned out I forgot to use `0x` in front of the hex data value.

`{"Protocol":"COOLIX","Bits":24,"Data":"0xB21F28","Repeat":3}`

instead of

`{"Protocol":"COOLIX","Bits":24,"Data":"B21F28","Repeat":3}`

I did not knew this, because I copied the result from the received ir signal of the original remote.

e.g.:

`{"Protocol":"COOLIX","Bits":24,"Data":"B27BE0","DataLSB":"4DDE07","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"off",........}}`

**Describe the solution you'd like**  
I think the `IrReceived` output should contain the `0x` in front of the hex values, so the user can copy the output and use it as input.

Another feature or bug:
Since `0x` seems to be mandatory, the command should not be accepted without it.
But I got `{"IRSend":"Done"}` anyway.

Should I create a separate issue?

I know the wiki says to use `0x` and I just irgnored it. :(

CWempe, 2019-09-28:
It looks like this is a duplicate of https://github.com/arendst/Sonoff-Tasmota/issues/5431 and was fixed in https://github.com/arendst/Sonoff-Tasmota/commit/d7715d1d2ad8f837bd7e588a77f27dcc63fd57a1.

But why is the issue still present in 6.0.0.14?

ascillato, 2019-09-29:
6.0.0.14 is very old. You should use latest that is 6.6.0.14. please update or use the precompiled bins from http://thehackbox.org/tasmota/json

Thanks

meingraham, 2019-09-29:
I believe 6.0.0.14 was a typo.

s-hadinger, 2019-09-29:
The missing 0x is definitely a bug of 6.6.0.14. Thanks for reporting. 

With regards to error message when parsing commands, I reluctant to improve it because it often mean significant code increase. With 0x missing it is currently parsed as an integer, so probably converted to 0.

arendst, 2019-09-29:
Inputting is a feature!

The code checks for 0x. If it's present it handles the rest as hex input. If not present it handles it as deckmal input.

An inbuilt feature of function strtol and provides legacy support too.

arendst, 2019-09-29:
Regarding output, there is a flag/option to report either decimal or hexadecimal.

s-hadinger, 2019-09-29:
I just tried and here is what I get:

```
08:45:46 CMD: IRhvac {"Vendor":"Coolix","Power":"On","Mode":"Hot","FanSpeed":3,"Temp":22.5}
08:45:46 MQT: stat/sonoff/IR1/RESULT = {"IRHVAC":{"Vendor":"COOLIX","Model":1,"Power":"on","Mode":"auto","Celsius":"on","Temp":22.5,"FanSpeed":"medium","SwingV":"off","SwingH":"off","Quiet":"off","Turbo":"off","Econo":"off","Light":"off","Filter":"off","Clean":"off","Beep":"off","Sleep":-1}}
08:45:46 MQT: tele/sonoff/IR1/RESULT = {"Time":"2019-09-29T08:45:46","IrReceived":{"Protocol":"COOLIX","Bits":24,"Data":"0xB25F78","DataLSB":"0x4DFA1E","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"on","Mode":"auto","Celsius":"on","Temp":22,"FanSpeed":"medium","SwingV":"off","SwingH":"off","Quiet":"off","Turbo":"off","Econo":"off","Light":"on","Filter":"off","Clean":"on","Beep":"off","Sleep":-1}}}
```

The 0x is present in the `IrReceived` using sonoff-ir. @CWempe Could you please send a complete log?

CWempe, 2019-09-29:
@meingraham 
Yes, it was a typo. I am running `"6.6.0.14(55ab963-ir)`.

@s-hadinger 
Here is the mqtt output with two recorded button presses from the original remote.

```
tele/ir02/LWT Online
cmnd/ir02/POWER (null)
tele/ir02/INFO1 {"Module":"YTF IR Bridge","Version":"6.6.0.14(55ab963-ir)","FallbackTopic":"cmnd/DVES_F20241_fb/","GroupTopic":"sonoffs"}
tele/ir02/INFO2 {"WebServerMode":"Admin","Hostname":"ir02","IPAddress":"192.168.1.37"}
tele/ir02/INFO3 {"RestartReason":"Software/System restart"}
stat/ir02/RESULT {"POWER":"OFF"}
stat/ir02/POWER OFF
tele/ir02/STATE {"Time":"2019-09-29T12:46:03","Uptime":"0T00:00:14","UptimeSec":14,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"CWempe","BSSId":"34:31:C4:14:76:3B","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:04"}}
tele/ir02/RESULT {"Time":"2019-09-29T12:46:14","IrReceived":{"Protocol":"COOLIX","Bits":24,"Data":"B23F20","DataLSB":"4DFC04","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"on","Mode":"cool","Celsius":"on","Temp":20,"FanSpeed":"max","SwingV":"off","SwingH":"off","Quiet":"off","Turbo":"off","Econo":"off","Light":"on","Filter":"off","Clean":"on","Beep":"off","Sleep":-1}}}
tele/ir02/RESULT {"Time":"2019-09-29T12:46:19","IrReceived":{"Protocol":"COOLIX","Bits":24,"Data":"B27BE0","DataLSB":"4DDE07","Repeat":0,"IRHVAC":{"Vendor":"COOLIX","Model":-1,"Power":"off","Mode":"off","Celsius":"on","Temp":2.0,"FanSpeed":"unknown","SwingV":"off","SwingH":"off","Quiet":"off","Turbo":"off","Econo":"off","Light":"on","Filter":"off","Clean":"on","Beep":"off","Sleep":-1}}}
```

@arendst 
Are you refering to `SetOption58` or `SetOption58`?
Or do you mean something else?

btw:  
I now get why there is no "warning" when i do not use `0x` for the input. Because the firmware allows decimal and therefore input without `0x` can be valid.

So my second feature request can be ignored.

PR #6512 (2019-09-29): Fix 6508
## Description:

Fix missing `0x` on Data fields.
Also removed the timestamp in IrReceived MQTT message (was there by mistake in the first place).

**Related issue (if applicable):** fixes #6508 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/ [... cut]
closed 2019-09-29 by arendst