arendst/Tasmota #6510: Support for "all on/off" button

NdK73, 2019-09-29:
Hello.

I have succesfully flashed a socket with 5 buttons for 4 controllable relays (plus 2 always-on USB ports).

I could easily configure buttons corresponding to the relays to toggle 'em, but the first button, that should provide global on/off (short press) or "enter config" (long press) remains unconfigured and unused.

Is it possible to recover original function?

Tks.

meingraham, 2019-09-29:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

NdK73, 2019-09-29:
I did not follow troubleshooting template because I submitted this as a feature request :)

**Have you looked for this feature in other issues and in the wiki?**
Yes. Couldn't find anything applicable. Maybe I searched for the wrong terms, but I think I used the ones a tasmota-newbie (like me) would use.

**Is your feature request related to a problem? Please describe.**
No... Unless you consider a missing feature a problem.

**Describe the solution you'd like**  
Having a "config button" or "multi-action button"  menu item that when pressed would turn all the relays on or off.

**Describe alternatives you've considered**  
I'm already working on my own firmware and Tasmota is just an intermediate step I use to be able to flash my own.

**Additional context**  
The device is this: https://www.amazon.it/gp/product/B07L3FMYSQ - as you can see there's a button to individually toggle each socket and a bigger one to globally turn on/off all the sockets. A long press on the master button enters config mode,

andrethomas, 2019-09-29:
> I'm already working on my own firmware and Tasmota is just an intermediate step I use to be able to flash my own.

A feature request on something used as an intermediate step? :)

jziolkowski, 2019-09-29:
Would be great if you tried to read the wiki, because such command is already there.
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#power0
You need rules to achieve this mixed buttons behavior.

andrethomas2, 2019-09-29:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-29 by andrethomas2

NdK73, 2019-09-29:
Tks. I always try to find what could be useful for others, too.
I'll have a glance at rules, but it's surely not worth the effort for my case.