arendst/Tasmota #6514: upgrading firmware from sonoff-basic bricks the sonoff mini in diy mode

magiva, 2019-09-29:
I flashed the sonoff mini in DIY mode OTA with the DIY tool and sonoff-basic.bin
I set the device up and then, to check i was on the latest version, i clicked upgrade firmware
it upgraded the firmware, now the device is unresponsive and is not on the wifi, nor is it looking for the diy network.
Maybe upgrade should be disabled or a warning for mini-DIY devices

Jason2866, 2019-09-29:
Have you done as described in wiki? https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrading#device-settings

ascillato2, 2019-09-29:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-29 by ascillato2

magiva, 2019-09-29:
nothing is possible as it isn't accessible.
i could solder some pins to comm points but this issue is more that the
upgrade option shouldn't be available for minis that were flashed from diy
mode ota with the sonoff tool.

br
Mark-Andrew

meingraham, 2019-09-30:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY#post-installation
```
3. If you flashed sonoff-basic.bin, it is recommend that you upgrade to the firmware and Core variant 
that is needed for your device and use case (e.g., sonoff.bin). You must perform this update using the 
local File upload OTA. Do not use a web OTA for this step. Download the firmware file from the 
repository to your computer.

Some users have reported that upgrading via web OTA from sonoff-basic.bin to another binary has 
resulted in an unresponsive device which has required a wired flash to recover.
```

magiva, 2019-09-30:
yes that is the correct conclusion
i hadn't read that

br
Mark-Andrew