arendst/Tasmota #6517: SwitchMode Behavior Mismatch

F-OBrien, 2019-09-29:
### BUG DESCRIPTION
SwitchMode 3 `PUSHBUTTON` behaves like SwitchMode 6 `PUSHBUTTONHOLD_INV` (excluding the long press functionality)
SwitchMode 4 `PUSHBUTTON_INV` behaves like SwitchMode 5 `PUSHBUTTONHOLD` (excluding the long press functionality)
The functionality of SwitchMode 5 and 6 is as described in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#switchmode) but SwitchMode 3 and 4 are the opposite of the wiki descriptions.
SwitchMode 3 and 4 behave as described in the code comments in `support_switch.ino`. Mode 3 `// Toggle with pushbutton to Gnd` Mode 4 `// Toggle with releasing pushbutton from Gnd`
SwitchMode 5 and 6 have the same code comment `// Toggle with pushbutton to Gnd` so it is not clear what the intended functionality is.
  
### REQUESTED INFORMATION

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1 Mini
- [x] Tasmota binary firmware version number used: Master and 6.6.0.14 Build Date & Time | 2019-09-29T13:03:09
  - [x] Pre-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
   Configuration output here:
 ```20:51:37 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:51:38 RSL: stat/sonoff/RESULT = {"Module":{"0":"Generic"}}
20:51:38 RSL: stat/sonoff/RESULT = {"GPIO0":{"53":"Led2"},"GPIO1":{"0":"None"},"GPIO2":{"9":"Switch1"},"GPIO3":{"0":"None"},"GPIO4":{"22":"Relay2"},"GPIO5":{"52":"Led1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"157":"LedLink"},"GPIO16":{"21":"Relay1"}}```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```

### TO REPRODUCE
Use SwitchMode 3, 4, 5 & 6 and compare behavior with wiki descriptions.

### EXPECTED BEHAVIOUR
PUSHBUTTON and PUSHBUTTONHOLD should have similar behavior
PUSHBUTTON_INV and PUSHBUTTONHOLD_INV should have similar behavior

ascillato2, 2019-10-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-09 by ascillato2

F-OBrien, 2019-10-10:
@ascillato2 I'm not sure why this was labeled as a question as it wasn't intended as one. But if you guys wish to ignore this that's up to you.

The fact is switch mode 3 and 6 are equivalent and mode 4 and 5 are equivalent while naming convention in the code and the wiki would suggest mode 3 and 5 should be equivalent and mode 4 and 6 should be equivalent.

The modes work, just not as described so it's not a major bug it may just be confusing to new users as it confused me.

meingraham, 2019-10-10:
@F-OBrien 

Can you make the suggested changes to the wiki?

F-OBrien, 2019-10-11:
@meingraham to have the Wiki reflect the current behavior I think it should read as follows:

________________
`SwitchMode 3`

`Set push-button mode (0 = TOGGLE, 1 = ON (_default_))`

Tasmota will send a TOGGLE command when the button is **pressed (closing** the circuit). When **the button is released (opening** the circuit) nothing will happen. Default state is ON and when pressed it's OFF. (This trigger is known as **[falling-edge](https://en.wikipedia.org/wiki/Signal_edge))**

`SwitchMode 4`

`Set inverted push-button mode (0 = OFF (_default_), 1 = TOGGLE)`

Tasmota will send a TOGGLE command when the button is **released (opening** the circuit). When the button is **pressed (closing** the circuit) nothing will happen. Default state is OFF and when pressed it's ON. (This trigger is known as **[rising-edge](https://en.wikipedia.org/wiki/Signal_edge))**
______________
Mode 4 triggers when the switch goes from low (pressed) to high (not pressed) but the switch state goes from ON (1) to OFF (0). If you consider the button press it is rising edge but the switch state it is falling edge. As the code is based on `PRESSED`/`NOT_PRESSED` I'll assume the wiki considers switch action and not ON/OFF state. 
______________
Alternatively I think if line 167 in `support_switch.ino`
`if ((PRESSED == button) && (NOT_PRESSED == Switch.last_state[i])) {` 
and line 182
`if ((NOT_PRESSED == button) && (PRESSED == Switch.last_state[i])) {` 
were swapped the switch modes would behave as described in the wiki but that would cause people switches to behave slightly differently than they currently do, if they are using those switch modes.

zwinsch, 2019-10-12:
This was confusing me too in the beginning - but I think it is more or less by intent. It is due to, that in Modes 5+6 either a Toggle OR a Hold is triggert, so the sort press trigger is on opposite edge trigggtert as the longpress/hold is triggert by.
Else you would generate first a toggle and after a while a  hold on the same action... this in fact could be helpfull in some circumstances - but chanes to that behavior would screw up most of the rules already created for switch-handling...

Here is what I did to implement a behavior similiar to the existing switches for my shades
(sort press = start/stop, long press= run until switch is released aggain) - It took me quiet a while to come to that simple solution, which does not even need a variable:

Setoption32 4

Rule1
on Switch1#state do Power1 2 endon
on Power1#state=1 do SwitchMode1 0 endon
on Power1#state=0 do SwitchMode1 5 endon
on Switch2#state do Power2 2 endon
on Power2#state=1 do SwitchMode2 0 endon
on Power2#state=0 do SwitchMode2 5 endon