arendst/Tasmota #6518: enable access of DS18x20 sensor ID in rules and/or scripts

th-s-op, 2019-09-29:
I am using a number of DS18B20 sensors. As in the non-legacy driver the sensor names are "DS18B20-n", n numbered from 1 upwards and sorted by ID/Address, I have no reliable way of accessing an individual sensor in rules/scripts. I can look for the ID in the MQTT/Status message and start using the generated Sensor name, but once I detach a sensor and/or attach a new one, I could accidentally access the wrong sensor.

Feature request: allow access to the ID of a DS18x20 Sensor in a Script and/or enable to optionally use the ID as the sensor name in a Rule.

Example:
currently: "on DS18B20-3#Temperature do var1 %value% endon"
and I want to use "on 041780358DFF#Temperature do ..."

As I want to handle Sensor data within the Tasmota device and I am NOT using MQTT/Home Automation middleware, the solution to use the MQTT status message doesn't work for me.

effelle, 2019-09-29:
I understand your request, but I'm not sure will be possible without changing the name from `DS18B20-n` to something like `DS18B20-041780358DFF`. 
To get a bigger picture and for the sake of knowledge, `ID`is a subsensor not a sensor:
```
/tele/SENSOR = {"DS18B20-3":{"Id":"041780358DFF","Temperature":16.6},"TempUnit":"C"}
```
Meaning the selectable key used to trigger a rule is the name of the sensor (`DS18B20-3`) and THEN the subsensor (`#Temperature`). ID will not change state, then is not usable as a trigger.
My 2c.

@arendst Will this name format break compatibility with legacy for multiple DS18x20 configuration? 
This will also solve a issue with sensors created on HA integration too where every Dallas will create an ID as a sensor by `HASS_DISCOVER_SENSOR_ANY`.

th-s-op, 2019-09-30:
Alternatively, if changing the names breaks compatibility and having that with a compiler switch is not an option, is there a chance to access the ID/address from within a Tasmota script? Like
`
>T
temp=DS18B20-1#Temperature
id=DS18B20-1#ID
`
This would be a feasible alternative, too as I could assign the right real sensors within a switch/case statement.

effelle, 2019-09-30:
I'n not using Tasmota scriping language, just rules. 
Since is really more flexible solution than rules, perhaps you could find a suitable example on the [cookbook](https://github.com/arendst/Sonoff-Tasmota/wiki/Script-Cookbook).

gemu2015, 2019-10-03:
it is possible but laborious to do it in scripter

first replace the '-' char with '_' by SetOption64 1

example:

then this script sorts the Ids to temperatures from t1 to t3
\>D
v1=0
v2=0
v3=0
s1=""
s2=""
s3=""
t1=0
t2=0
t3=0

; fetch 3 ds sensors with their ids and temperature
\>T
s1=DS1820_1#Id
v1=DS1820_1#Temperature
s2=DS1820_2#Id
v2=DS1820_2#Temperature
s3=DS1820_3#Id
v3=DS1820_3#Temperature

; now select the sensor temperatures by id, the ids must be known
if s1=="041780358DFF"
then
t1=v1
endif
if s2=="041780358DFF"
then
t1=v2
endif
if s3=="041780358DFF"
then
t1=v3
endif

if s1=="041780358DFC"
then
t2=v1
endif
if s2=="041780358DFC"
then
t2=v2
endif
if s3=="041780358DFC"
then
t2=v3
endif

if s1=="041780358DFD"
then
t3=v1
endif
if s2=="041780358DFD"
then
t3=v2
endif
if s3=="041780358DFD"
then
t3=v3
endif

; now we have the "right" values in t1 to t3 assigned to IDs

th-s-op, 2019-10-03:
@gemu2015 that's the solution I was looking for! I did not find the accessibility of the sensor ID in a Script by using an expression like "DS1820_1#Id", so I assumed it's not possible.
Thank you very much! I'll close the issue although I leave it to the developers if it's not a good idea to provide a way to explicitly handle an event by sensor ID in Rules as this really is the only reliable way of identifying one of many sensors.

closed 2019-10-03 by th-s-op