arendst/Tasmota #6519: PulseTime command output reformat

jziolkowski, 2019-09-29:
As a fan of consistency, I suggest to align the PulseTime command with the likes of Var/Mem and Ruletimer.

PulseTime without relay index and value returns a JSON of all set PulseTimes. 
PulseTime without relay index but with value is the same as PulseTime1 <value> (as it is right now).
Used with relay index, obviously, returns JSON for that relay only, and sets it if used with value.

On the currently returned JSON payload:
The structure is inconsistent, because the key is the set value of PulseTime, ergo dynamic.

    [23:44:04] 4ch2/stat/RESULT {"PulseTime1":{"100":{"Active":"46"}}}

If someone wants to poll for the current set value, one has to assume that the key is the value, which is against logic. I suggest a slightly modified, and more readable (and consistent) syntax:

    [23:44:04] 4ch2/stat/RESULT {"PulseTime1":{"Set": "100", "Remaining":"46"}}

In this syntax, the keys are constant, and the whole payload is very clear and readable.

Pardon my ignorance if I'm wrong, but I'm not aware of any tools for Tasmota that use this information, so no backwards compatibility would be broken should my request be applied in the code base :)

Cheers,
JZ

ascillato, 2019-10-03:
I agree with your observation. It should be consistent.

effelle, 2019-10-03:
@jziolkowski Jacek, 
sadly creating a new, global dictionary for pulsetime it's more complicated than i thought. I have to improve my skills. :P
Anyway, I've pull a PR with the new format for `pulsetimeX`. See #6519 for more information. 

F.

jziolkowski, 2019-10-03:
[image]

Fantastic! Thanks!

closed 2019-10-03 by jziolkowski

Apollon77, 2020-01-27:
Hey All here, donyou thought no one uses this feature and that’s why did a breaking change?! Bad news: at least ioBroker (See linked issue) did support it and is now broken. :-(

It is very bad to base such decisions on assumptions!!

ascillato, 2020-01-27:
Hi, sorry about your frustation but was not an unilateral desicion. See that this public issue was opened, it received feedback and after that, it was implemented. So, there wasn't assumptions, it has been asked and then the change was performed. This change was only in the JSON output of pulsetimer command. It is still a valid parseable JSON, so it shouldn't make any issue in home automation softwares.

Tasmota is in continuous development. On every update, we encourage people on reading the changelog. No one have replied a breaking change on this before you. 

Anyway, in your linked issue, the commands `PulseTime_set` and `PulseTime_remaining` were never Tasmota commands, so seems that for using those, a forked version of Tasmota was required ? Or those are commands related to the IOBROKER adapter ? So, seems that your issue is not Tasmota itself but the adapter. Anyway, if you need help, please provide us with console outputs, commands used and how the IOBROKER adapter works in order to have information to work on. Thanks.

jziolkowski, 2020-01-27:
IoBroker guys and users had 4 months to fix their parser and to adapt for what is now a stable version. This change was implemented 5 production releases ago. I fail to see any validity to your claims after literally a third of a year. Ask IOBroker devs (and its community) why didn't _they_ fix that.