arendst/Tasmota #6520: KNX: please increment the Group Addresses to Receive Data size

jkx, 2019-09-29:
By default KNX Tasmota accept up to 10 group addresses to receive data.
For a single relay, we usually need 3 addresses. one to accept read request, one to receive on/off commands and one for toggle. 

For a Sonoff 4channels, that give us 12 group addesses, but 10 is the default max. So we are unable to use this kind of config for 4 chanels devices. 

I tried to modify the source code as described in  xdrv_11_knx.ino

// sonoff.h 
```
const uint8_t MAX_KNX_GA = 16;  
const uint8_t MAX_KNX_CB = 16;   
```

// esp-knx-ip.h
```
#define MAX_CALLBACK_ASSIGNMENTS  16
#define MAX_CALLBACKS             16
#define MAX_CONFIGS               32

```
After changing this lines, the whole config isn't stored anymore. Every "save config" (even the module config) on the sonoff isn't saved. I'm unable to upgrade the firmware OTA. 

Thanks for the help.

ascillato, 2019-09-29:
Hi,

Could you be more specific? How is your config that you need 3 KNX slots for every relay? I just use one KNX slot for every relay. You can read/write/toggle on the same address.

Can you share your config and setup?

jkx, 2019-09-29:
Here a sample config, for a single relay.

screenshot

4 / 0 / 13 -> Output 1  is mandatory (same as send group) to be able to 
request a read value
4 / 0 / 14 -> Output 1  to send ON/OFF. If you use the same 4/0/13, you 
will broadcast a KNX state that isn't the real state.
4 / 0 / 15 -> Output 1 Toggle , just to toggle.

Thanks

ascillato, 2019-09-29:
Ok, but why you separate them in 3 different slots? It works in the same slot.

jkx, 2019-09-29:
I have some buttons paired w/ the ON/OFF commands (1bits) and some w/ 
the toggle, so at leat
2 are needed.

The last one come from one bug in esp-knx-ip. It need a slot to receive 
the READ request for the state.

Check out issue 3165 that describe this.

https://github.com/arendst/Sonoff-Tasmota/issues/3165

Without this, my automation won't able to query the state off the 
devices at boot.

Thanks

ascillato, 2019-09-29:
For 1 output, your config should look like in the wiki:

[image]

And you can use the same address for **Data to send to group addresses** and for **Group Addresses to receive data from**

(the loop avoiding in KNX is just not using the same address for output and input in more than 1 device)

Tasmota answer the state when a read request arrives to the "**group address
to receive data**", so you don't need any extra slot.

The screenshot is one of my devices. My home automation system (Home Assistant) makes a read request when it boots and with that config, Tasmota answers the state by KNX.

In your config you are duplicating slots. There is no need for that.

ascillato, 2019-09-29:
> The last one come from one bug in esp-knx-ip. It need a slot to receive
> the READ request for the state.

That is not a bug. It is the way it works. Tasmota answer the state when a read request arrives to the "group address to receive data"

ascillato, 2019-09-29:
Please, share a screenshot of your full KNX config

jkx, 2019-09-30:
[image]

ascillato2, 2019-09-30:
Please, try without using 4/0/14 and 4/0/15

Those are just duplicated. You can use just 4/0/13 for controlling, reading state and for informing its state.

jkx, 2019-09-30:
If I remove the 4/0/14, I will have the same address for command and the the state. This can work, but if the device is disconnected (wifi is dead or anything else), when someone on the bus (for example an knx switch) send an "ON". The homeautomation will receive this "ON" too. And will think the device is "ON", even if the device is disconnected or not working properly. 
This is the main reason why in KNX state and command address aren't the same. (this is exactly what #3165 is about).

Look at this cEMI, if I push a KNX btn. If you have both GA set to 4/0/13. You will not able to distinguish the two cEMI (the command & the state change).

`2019-09-30 21:50:30 i7 __main__[5195] DEBUG Receive: <KNXcEMI at 0x7f9452ba4a90 pa=6.6.6 ga=4/0/15 code=0x29 cmd=KNXCommand.WRITE data=bytearray(b'\x01')>
2019-09-30 21:50:30 i7 __main__[5195] DEBUG Receive: <KNXcEMI at 0x7f945061f3d0 pa=4.0.2 ga=4/0/13 code=0x29 cmd=KNXCommand.WRITE data=bytearray(b'\x01')>
`

For the 4/0/15, I have an KNX push button linked to this, not doable :)

You can use HomeAssistant with only 1 GA but :
1) It will send command and never can really trust the state of the device without sending an bunch of READ, or only trusting internal state. 

2) To toggle the light, it will use the internal state (or query the current state) before sending an ON or OFF. I can do that cause I don't use HA, it will put the whole house in dark if HA stops working. And that's really bad the WAF. 

Thanks.

ascillato, 2019-09-30:
Ok, I understand your use case. You want to have separated the address for control and for state, making the 10 available slots not enough for you. That is Ok.

I'm adding this request to my **to do** list for the KNX driver, but will not do this soon as this adds more RAM usage and will require to do some more optimizations to the code in order to not affect other Tasmota features.

> it will put the whole house in dark if HA stops working

That doesn't happen in a home automation well done. Every smart device can control itself. If your home assistant brokes, all devices will continue to communicate between them by KNX as long as your wifi works. A Home Assistant instance should have just telesupervision (panels, databases, etc) and very high level automations.

ascillato2, 2019-09-30:
Added the label (Requested Feature - Hold Over) for future reference.

@jkx Thanks for sharing your ideas

closed 2019-09-30 by ascillato2

roelbroersma, 2025-04-22:
I would also like to ask if there is any change of having more than 10 group addresses... (in the meantime.. we're almost 5 and a half years later now)  Am I the only one who thinks this is not enough?

ascillato, 2025-04-23:
@roelbroersma I did the driver and I agree with you that 10 group addresses are not enough in some cases. Unfortunately, I don't have the time right now to work on this, but you are free to modify the driver and provide a pull request we can review. If you can do that, would be very appreciated. Thanks.

roelbroersma, 2025-06-02:
> [@roelbroersma](https://github.com/roelbroersma) I did the driver and I agree with you that 10 group addresses are not enough in some cases. Unfortunately, I don't have the time right now to work on [...]

I am working on a PR. So happy to help you and bring this further!

Instead of **10**, let's make it **50**.  In this way, people can also do:   modbus rtu -> berry-script -> knx
(for smart meters, if you do not want to use the 'Smart Meter Interface')
Of just have a bunch of sensors connected..

As far as I can see, it's about:

**1.** 
**tasmota/include/tasmota.h**
```
const uint8_t MAX_KNX_GA = 50;              // Max number of KNX Group Addresses to read that can be set
const uint8_t MAX_KNX_CB = 50;              // Max number of KNX Group Addresses to write that can be set
```

**2.**
**/tasmota/tasmota_xdrv_driver/xdrv_11_knx.ino**
```
#define MAX_KNX_GA             50            // Max number of KNX Group Addresses to read that can be set
#define MAX_KNX_CB             50            // Max number of KNX Group Addresses to write that can be set
```

**3.**
**/lib/default/headers/esp-knx-ip.h**
```
#define MAX_CALLBACK_ASSIGNMENTS  50 // [Default 10] Maximum number of group address callbacks that can be stored
#define MAX_CALLBACKS             50 // [Default 10] Maximum number of callbacks that can be stored
#define MAX_CONFIGS               100 // [Default 20] Maximum number of config items that can be stored
```

**4.**
**/lib/lib_div/esp-knx-ip-0.5.2/src/esp-knx-ip.h**
```
#define MAX_CALLBACK_ASSIGNMENTS  50 // [Default 10] Maximum number of group address callbacks that can be stored
#define MAX_CALLBACKS             50 // [Default 10] Maximum number of callbacks that can be stored
#define MAX_CONFIGS               100 // [Default 20] Maximum number of config items that can be stored
// AS A BONUS, ALSO SET THE BELOW TO 100
#define MAX_FEEDBACKS             100 // [Default 20] Maximum number of feedbacks that can be shown
```

This file has this comment:

/**
 * CONFIG
 * All MAX_ values must not exceed 255 (1 byte, except MAC_CONFIG_SPACE which can go up to 2 bytes, so 0xffff in theory) and must not be negative!
 * Config space is restriced by EEPROM_SIZE (default 1024).
 * Required EEPROM size is 8 + MAX_GA_CALLBACKS * 3 + 2 + MAX_CONFIG_SPACE which is 552 by default
 */
#define EEPROM_SIZE               1024 // [Default 1024]
#define MAX_CONFIG_SPACE          0x0200 // [Default 0x0200] Maximum number of bytes that can be stored for custom config

I hope to make the comment more clear/usefull:

According to math rules, the comment means: 8 + (MAX_GA_CALLBACKS * 3) + 2 + MAX_CONFIG_SPACE
**8** = 1 byte  (first byte/overhad)
**MAX_GA_CALLBACKS * 3** = 3 bytes for every Group Address Callback
**2** = 2 bytes (for checksum or whatever) overhead
MAX_CONFIG_SPACE = 0x200 (in hex) = 512 bytes in decimal

So for 50 GA callbacks, that would be: 8 + (50 × 3) + 2 + 512 = 8 + 150 + 2 + 512 = **672 bytes** which fits in the 1024 bytes of EEPROM size.
Maybe this went wrong in a previous attempt to raise the number of GAs ?

Do you see any other things which should be changed or that come into your mind?  I will also compile and test it on a physical device.