arendst/Tasmota #6521: Request for support MS5611

SimplyKot, 2019-09-30:
I can't connect MS5611 t0 mu Sonoff Basic
Can you implement support for MS5611?
It's a fine working solution: https://github.com/jarzebski/Arduino-MS5611

Jason2866, 2019-09-30:
See https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#Can-you-add-this-unsupported-sensor-to-Tasmota

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

andrethomas, 2019-09-30:
It may be easier and quicker to use a chip/module for which there is already support... BMP085, BMP180, BMP280, BME280, and BME680 - Relatively inexpensive, reliable and easy to find.

closed 2019-09-30 by ascillato2