arendst/Tasmota #6525: TLS for MQTT in pre-built binaries

schildbach, 2019-09-30:
I would like to encourage you to provide at least one pre-built binary with support for MQTT via TLS. I know the 8266 is very RAM limited, but for example the '-basic' image has 18 kB free RAM on my hardware and according to the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/TLS#memory-usage) TLS would only need 5.4 kB. Imho it would be waste to not provide TLS in such cases.

Note that for basic security only public key pinning is needed. So the overhead of checking a certificate trust chain can be avoided.

**Have you looked for this feature in other issues and in the wiki?**  
Yes
**Is your feature request related to a problem? Please describe.**  
No
**Describe the solution you'd like**  
See above
**Describe alternatives you've considered**  
An alternative would be separating LANs, but this wouldn't provide as much security and it's not usually supported by home routers.
Another alternative would be self-building Tasmota, but this requirement would keep TLS away from 99% of Tasmota users.
**Additional context**  
n/a

majherek, 2019-09-30:
It would be great ;-)

meingraham, 2019-09-30:
With [Gitpod](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) now available, it is very simple to compile the firmware. This gives users the option to not load features they do not yet still obtain features that they do. What some consider essential, is not to others. With over 4 million Tasmota downloads, the opinion of this is quite broad.

andrethomas, 2019-09-30:
I agree with @meingraham

schildbach, 2019-09-30:
@meingraham Note I deliberately did **not** suggest adding another pre-built binary variant.

meingraham, 2019-10-01:
> I deliberately did not suggest adding another pre-built binary variant.
:question: :question:

As we discuss new features in Tasmota and how to best fit them in the available program space and iRAM, iRAM is actually the most precious resource. So, when we are weighing "default" features, the feature "market share" that users are using and asking about is part of the equation. The next is iRAM. Then finally program space in flash memory. When considering all these points, TLS does not rank as high up as several other features. Thus the decision to make TLS a user compiled feature by providing a means to generate the firmware without installing any software and compiling in under 5 minutes.

Mike

Jason2866, 2019-10-01:
The sonoff-basic version is build to be maximal weight light for devices which have a weak power supply. This version is NOT intended to get more features.
There is no current version which has free resources for TLS.

ascillato2, 2019-10-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-01 by ascillato2

schildbach, 2019-10-01:
I'd like to add that security is not a feature from a users perspective. So users will almost never vote for it, rather they take it as granted. It's the duty of us developers to always have security in mind, for each feature.

meingraham, 2019-10-01:
The TLS feature is available for a user to make use of if they desire to use it. Compiling the firmware is likely the easiest part of the setup.

A user decides whether to set up TLS on their Tasmota devices or not. When they do, it's almost inevitable that we receive inquiries when they attempt it. Thus we get a sense of how many users are using the feature. True, it's not a direct measurement. Neither are the others. But the mechanism is the same. So, in that sense we are comparing "pings" about particular features to understand "most used" features.

Mike

arendst, 2019-10-01:
Also, TLS needs a complete change of wifi library so just adding it as an "on the fly" option won't work.

adverserath, 2019-11-20:
I just spent hours trying to compile in tls for mqtt on the sonoff basic. I can't get the TLSKey command to return. I'm following the guide too, which mentions nothing of wifi libraries.
Anyway, could someone point me in the right direction for help? Would greatly appreciate it.

jeroenst, 2021-05-25:
I also would like tasmota with precompiled TLS

Jason2866, 2021-05-25:
You can find Tasmota with TLS enabled here: https://github.com/Jason2866/Tasmota-specials

Nikoolayy1, 2021-07-03:
Just to mention I submited a bug under Tasmota-specials but it could be a tasmota bug on itself:

$$$$$$$$$$$$$$$$$$$$$$$

First thanks for the precompiled tasmota software.

I installed TASMOTA with TLS support on Sonoff basic smart switch and when clicking enable TLS with GUI ''MQTT TLS'' option it does not work as the device reboots and option is cleared for some reason even after enabling it before the reboot (as enabling it causes the reboot). The workaround is from the TASMOTA console to set the option with "SetOption103 1".

https://tasmota.github.io/docs/Commands/

$$$$$$$$$$$$$$$$$$$$$$

ventilator, 2022-10-18:
My ESP8266 trips over a too large message since the MQTT broker does not support fragmentation. Any precompiled firmware with TLS active out there with a larger buffer?
Currently the ESP8266 connects but is unable to publish its larger messages.

`13:50:09.600 MQT: TLS connected in 1642 ms, max ThunkStack used 4468
13:50:09.603 MQT: MFLN not supported by TLS server
13:50:09.605 MQT: Server fingerprint: xxx
13:50:09.607 MQT: Connected
13:50:10.078 MQT: xxxx/s/LWT = Online (retained)
13:50:10.100 MQT: xxxx/s/cmnd/POWER = 
13:50:10.104 MQT: Subscribe to xxxx/s/cmnd/#
13:50:10.222 MQT: tasmota/discovery/C4DD572EF3F8/config = {"ip":"xx.xx.xx.xx","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-xxxx-5112","mac":"xxxxx","md":"Gosund SP1 v23","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.2.0.1","t":"tasmota_xxxx","ft":"xxxx/s/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
13:50:10.234 MQT: Message too large
`

arendst, 2022-10-18:
There is no buffer size configuration:

Tasmota:
```
  if (!MqttClient.beginPublish(topic, plength, retained)) {
//    AddLog(LOG_LEVEL_DEBUG, PSTR(D_LOG_MQTT "Connection lost or message too large"));
    return false;
  }
  uint32_t written = MqttClient.write(payload, plength);
  if (written != plength) {
    AddLog(LOG_LEVEL_DEBUG, PSTR(D_LOG_MQTT "Message too large"));
    return false;
  }
  MqttClient.endPublish();
```
PubSubClient:
```
boolean PubSubClient::write(uint8_t header, uint8_t* buf, uint16_t length) {
    uint16_t rc;
    uint8_t hlen = buildHeader(header, buf, length);

    rc = _client->write(buf+(MQTT_MAX_HEADER_SIZE-hlen),length+hlen);

// Start Tasmota patch
//    lastOutActivity = millis();

    if (rc != 0) {
        lastOutActivity = millis();
    }
// End Tasmota patch

    return (rc == hlen+length);
}
```
If it fails it is due too Print class (inherited by PubSubCLient) buffers....