arendst/Tasmota #6527: Ability to trigger Button HOLD action while SetOption1=0

schildbach, 2019-09-30:
I want to have a secondary action on my buttons. In the smartphone world, a long-press (aka hold in the Tasmota jargon) is pretty standard. However in Tasmota it seems HOLD is only possible when disabling the rescue multi-button assignments (3\*press and more) via SetOption1=1. I don't want to disable those, and I can't think of a reason why HOLD and e.g. 3\*press is mutually exclusive.

So my suggestion is to **enable HOLD if just ButtonTopic is set and drop the SetOption1 requirement**.

----

**Have you looked for this feature in other issues and in the wiki?**  
yes, it's documented: https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage#changing-default-button-functionality
**Is your feature request related to a problem? Please describe.**  
Yes, my usecase is: I want ot switch off all lights in my home with a single longpress on one of my light buttons. The HOLD should be published via MQTT and e.g. Node-Red would contain the logic to switch off all lights.
**Describe the solution you'd like**  
See above
**Describe alternatives you've considered**  
I don't see an alternative except maybe buying sonoffs with two buttons.
**Additional context**  
n/a

ascillato, 2019-09-30:
Hi,

You can assign up to 3 different behaviours to every button by using rules.

Please, check the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#button-with-single-press-double-press-and-hold

Thanks.

schildbach, 2019-09-30:
@ascillato The examples set SetOption1 to 1, which I do not want.

ascillato2, 2019-10-11:
Adding the label Requested Feature (hold over) until someone wants to add this feature. Thanks for sharing your ideas.

closed 2019-10-11 by ascillato2