arendst/Tasmota #6529: {"Command":"Unknown"} only when on version 6.6.0.14

gajotnt, 2019-10-01:
### BUG DESCRIPTION
After updating to latest version (6.6.0.14) several of my devices are reporting unknow commands, if i revert them to version 6.6.0.3 the problem goes away. Doesnt happen with the minimal 6.6.0.14 version. Did not try anyother versions between 6.6.0.3 and 6.6.0.14.

```
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:31 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:32 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
23:03:32 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
```

### REQUESTED INFORMATION

- [ ] Device used Sonoff T1
- [ ] Tasmota binary firmware version number used: 6.6.0.3 and 6.6.0.14
  - [ ] Pre-compiled, tried core 2.5.2 and stage core 2.6

- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
22:58:30 CMD: Backlog Template; Module; GPIO
22:58:31 MQT: stat/CozQuintal/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:58:31 MQT: stat/CozQuintal/RESULT = {"Module":{"30":"Sonoff T1 3CH"}}
22:58:31 MQT: stat/CozQuintal/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}
```

- [ ] Provide the output of this command: ``Status 0``:

  STATUS 0 output here:
```
  22:59:29 CMD: status 0
22:59:29 MQT: stat/CozQuintal/STATUS = {"Status":{"Module":30,"FriendlyName":["CozQuintal","Sonoff2","Sonoff3"],"Topic":"CozQuintal","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:59:29 MQT: stat/CozQuintal/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:52","StartupUTC":"2019-09-30T23:57:37","Sleep":50,"CfgHolder":4617,"BootCount":117,"SaveCount":6572,"SaveAddress":"FB000"}}
22:59:29 MQT: stat/CozQuintal/STATUS2 = {"StatusFWR":{"Version":"6.6.0.14(d7d0316-minimal)","BuildDateTime":"2019-09-30T20:10:46","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
22:59:29 MQT: stat/CozQuintal/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["TNT","indebuurt2"],"TelePeriod":600,"Resolution":"558180C0","SetOption":["00002009","280500000100060000005AFF000000000000","00000001"]}}
22:59:29 MQT: stat/CozQuintal/STATUS4 = {"StatusMEM":{"ProgramSize":372,"Free":628,"Heap":31,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","80082381","041003A2","20021702","010000C0","00000001"],"Drivers":"1,2,6,24","Sensors":"4"}}
22:59:29 MQT: stat/CozQuintal/STATUS5 = {"StatusNET":{"Hostname":"CozQuintal","IPAddress":"192.168.1.66","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:8A:70:18","Webserver":2,"WifiConfig":3}}
22:59:29 MQT: stat/CozQuintal/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.75","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8A7018","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:59:29 MQT: stat/CozQuintal/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 30 23:59:29 2019","Local":"Mon Sep 30 22:59:29 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-01:00"}}
22:59:29 MQT: stat/CozQuintal/STATUS10 = {"StatusSNS":{"Time":"2019-09-30T22:59:29"}}
22:59:29 MQT: stat/CozQuintal/STATUS11 = {"StatusSTS":{"Time":"2019-09-30T22:59:29","Uptime":"0T00:01:52","UptimeSec":112,"Vcc":3.460,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"TNT","BSSId":"18:E8:29:97:03:8B","Channel":6,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### TO REPRODUCE
Only does this on a couple of them... so maybe its really some defect with the sonoff T1, but wierd that its only showing when using the 6.6.0.14 version.

ascillato, 2019-10-01:
You are on minimal version. You need to update to latest full Tasmota version. Mínimal is an interim version for freeing flash memory for updating to a bigger version.

Please, use sonoff.bin

ascillato2, 2019-10-01:
Closing this issue as it is **duplicated**. Sorry.

If you need further assistant on updating, please use the Tasmota support chat. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

gajotnt, 2019-10-01:
Maybe I didn't explain myself well.
With sonoff-classic.bin (6.6.0.3) no unknown command error. 
With sonoff-minimal.bin (6.6.0.14) no unknown command error.
With sonoff.bin / sonoff-classic.bin / sonoff-sensors.bin (6.6.0.14) I get the unknown command error on some sonoffs.

meingraham, 2019-10-01:
We are keying off of
> 22:59:29 MQT: stat/CozQuintal/STATUS2 = {"StatusFWR":{"Version":"6.6.0.14(d7d0316-minimal)","BuildDateTime":"2019-09-30T20:10:46","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}

i.e., `"Version":"6.6.0.14(d7d0316-minimal)"`

You did not post any verbose (log level `4`) log output. What command is being sent to the device which results in the `RESULT`?

gajotnt, 2019-10-01:
Sorry. 
So it seams its receiving commands from other devices from domoticz. The wierd thing is only a few of them (4 or 5 cant recall the precise number) are giving me these unknow commands, i have about 20 sonoffs in my house btw.

Weblog 4:
```
08:31:05 SRC: MQTT
08:31:05 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 271, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082098",
   "idx" : 98,
   "name" : "Waste",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "0",
   "svalue2" : "130768.6",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "255",
   "svalue2" : "2680433.2",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "354",
   "svalue2" : "5021881.0",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "255",
   "svalue2" : "2680433.2",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "354",
   "svalue2" : "5021881.0",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "255",
   "svalue2" : "2680433.2",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "354",
   "svalue2" : "5021881.0",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:08 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "255",
   "svalue2" : "2680433.2",
   "unit" : 1
}

08:31:08 SRC: MQTT
08:31:08 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:09 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "354",
   "svalue2" : "5021881.0",
   "unit" : 1
}

08:31:09 SRC: MQTT
08:31:09 MQT: stat/WcCorredor/RESULT = {"Command":"Unknown"}
08:31:09 WIF: Checking connection...
08:31:09 WIF: Connected
```

Status0:

```
08:32:52 CMD: status0
08:32:52 SRC: WebConsole from 192.168.1.75
08:32:52 MQT: stat/WcCorredor/STATUS = {"Status":{"Module":29,"FriendlyName":["WcCorredor","Sonoff2"],"Topic":"WcCorredor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:32:53 SRC: MQTT
```
This is how this switch is configured in the domoticz options page:

[image]

gajotnt, 2019-10-01:
With version 6.6.0.3
Weblog 4:

```
09:01:02 DOM: idx 96, nvalue 0
09:01:04 MQT: Received Topic domoticz/out, Data Size 237, Data {
   "Battery" : 100,
   "RSSI" : 2,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140A5",
   "idx" : 85,
   "name" : "Bomba Calor",
   "nvalue" : 0,
   "stype" : "Switch",
   "switchType" : "On/Off",
   "unit" : 1
}

09:01:04 DOM: idx 85, nvalue 0
09:01:04 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "149",
   "svalue2" : "2680503.2",
   "unit" : 1
}

09:01:04 DOM: idx 97, nvalue 0
09:01:04 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "251",
   "svalue2" : "5021961.5",
   "unit" : 1
}

09:01:04 DOM: idx 96, nvalue 0
09:01:05 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 100,
   "EnergyMeterMode" : "1",
   "RSSI" : 4,
   "description" : "",
   "dtype" : "General",
   "id" : "00082082",
   "idx" : 82,
   "name" : "Solar",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "189",
   "svalue2" : "1380541.2",
   "unit" : 1
}

[... 349 lines cut ...]
   "stype" : "kWh",
   "svalue1" : "149",
   "svalue2" : "2680504.0",
   "unit" : 1
}

09:01:23 DOM: idx 97, nvalue 0
09:01:23 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "250",
   "svalue2" : "5021962.0",
   "unit" : 1
}

09:01:23 DOM: idx 96, nvalue 0
```

meingraham, 2019-10-01:
@gajotnt 

What is the Domoticz configuration of the device(s) that don't have this problem?

I don't use Domoticz so I cannot decipher the output. But I do see 96/97 indeces in the commands but your device is configured for 90/91. Other than that, I can't help further.

@arendst?

arendst, 2019-10-01:
Pls provide cosole logging (4) of a restart. Also a screen dump of Domoticz parameters.

gajotnt, 2019-10-01:
Sorry for the time it took, had classes today. Hope this is what you asked for:

```
00:00:00 CFG: Loaded from flash at FA, Count 6596
00:00:00 Project sonoff CozQuintal Version 6.6.0.14(0e240bb-classic)-STAGE
00:00:00 WIF: Connecting to AP1 TNT in mode 11N as CozQuintal...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on CozQuintal with IP address 192.168.1.66
19:32:14 MQT: Attempting connection...
19:32:14 MQT: Connected
19:32:14 MQT: tele/CozQuintal/LWT = Online (retained)
19:32:14 MQT: cmnd/CozQuintal/POWER = 
19:32:14 MQT: tele/CozQuintal/INFO1 = {"Module":"Sonoff T1 3CH","Version":"6.6.0.14(0e240bb-classic)","FallbackTopic":"cmnd/DVES_8A7018_fb/","GroupTopic":"sonoffs"}
19:32:14 MQT: tele/CozQuintal/INFO2 = {"WebServerMode":"Admin","Hostname":"CozQuintal","IPAddress":"192.168.1.66"}
19:32:14 MQT: tele/CozQuintal/INFO3 = {"RestartReason":"Software/System restart"}
19:32:14 MQT: stat/CozQuintal/RESULT = {"POWER1":"OFF"}
19:32:14 MQT: stat/CozQuintal/POWER1 = OFF
19:32:14 MQT: stat/CozQuintal/RESULT = {"POWER2":"ON"}
19:32:14 MQT: stat/CozQuintal/POWER2 = ON
19:32:14 MQT: stat/CozQuintal/RESULT = {"POWER3":"OFF"}
19:32:14 MQT: stat/CozQuintal/POWER3 = OFF
19:32:15 MQT: domoticz/in = {"idx":88,"nvalue":0,"svalue":"","Battery":86,"RSSI":8}
19:32:15 MQT: domoticz/in = {"idx":2,"nvalue":1,"svalue":"","Battery":86,"RSSI":8}
19:32:15 MQT: domoticz/in = {"idx":89,"nvalue":0,"svalue":"","Battery":86,"RSSI":8}
19:32:15 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
19:32:15 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
19:32:15 MQT: stat/CozQuintal/RESULT = {"Command":"Unknown"}
```
[image]
[image]

gajotnt, 2019-10-01:
And this is from one of the ones that dont give the unknown command warning:
```
00:00:00 CFG: Loaded from flash at FA, Count 7106
00:00:00 SRC: Restart
00:00:00 Project sonoff CozinhaCorredor Version 6.6.0.14(d7d0316-sonoff)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 TNT in mode 11N as CozinhaCorredor...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on CozinhaCorredor with IP address 192.168.1.67
19:38:20 NTP: Drift 0, (UTC) Tue Oct 01 20:38:20 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
19:38:21 MQT: Attempting connection...
19:38:21 MQT: Connected
19:38:21 MQT: tele/CozinhaCorredor/LWT = Online (retained)
19:38:21 MQT: cmnd/CozinhaCorredor/POWER = 
19:38:21 MQT: Subscribe to cmnd/CozinhaCorredor/#
19:38:21 MQT: Subscribe to cmnd/sonoffs/#
19:38:21 MQT: Subscribe to cmnd/DVES_A131D1_fb/#
19:38:21 MQT: Subscribe to domoticz/out/#
19:38:21 MQT: tele/CozinhaCorredor/INFO1 = {"Module":"Sonoff T1 2CH","Version":"6.6.0.14(d7d0316-sonoff)","FallbackTopic":"cmnd/DVES_A131D1_fb/","GroupTopic":"sonoffs"}
19:38:21 MQT: tele/CozinhaCorredor/INFO2 = {"WebServerMode":"Admin","Hostname":"CozinhaCorredor","IPAddress":"192.168.1.67"}
19:38:21 MQT: tele/CozinhaCorredor/INFO3 = {"RestartReason":"Software/System restart"}
19:38:21 MQT: stat/CozinhaCorredor/RESULT = {"POWER1":"OFF"}
19:38:21 MQT: stat/CozinhaCorredor/POWER1 = OFF
19:38:21 MQT: stat/CozinhaCorredor/RESULT = {"POWER2":"OFF"}
19:38:21 MQT: stat/CozinhaCorredor/POWER2 = OFF
19:38:22 DOM: idx 90, nvalue 0
19:38:22 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "819",
   "svalue2" : "5027984.5",
   "unit" : 1
}

19:38:22 DOM: idx 96, nvalue 0
19:38:22 MQT: Received Topic domoticz/out, Data Size 273, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082097",
   "idx" : 97,
   "name" : "Casa",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "458",
   "svalue2" : "2684425.0",
   "unit" : 1
}

19:38:22 DOM: idx 97, nvalue 0
19:38:22 MQT: Received Topic domoticz/out, Data Size 284, Data {
   "Battery" : 255,
   "EnergyMeterMode" : "1",
   "RSSI" : 12,
   "description" : "",
   "dtype" : "General",
   "id" : "00082096",
   "idx" : 96,
   "name" : "ConsumoEfectivo",
   "nvalue" : 0,
   "stype" : "kWh",
   "svalue1" : "819",
   "svalue2" : "5027984.5",
   "unit" : 1
}
```

cyablo, 2019-10-02:
I've got the same problem on FHEM. Can't switch/toggle my devices over MQTT. 6.6.0.11 was fine, updated to 6.6.0.14 yesterday.

arendst, 2019-10-02:
Nice one.

Do you have a rule difference between CozQuintal and CozinhaCorredor?

gajotnt, 2019-10-02:
Don't use rules, apart from device names and idx the config is the same.

arendst, 2019-10-02:
Well for some reason the domoticz data falls trough a test.

I changed some thing around this test so pls try the latest dev release.

gajotnt, 2019-10-02:
Have to compile it or the latest http://thehackbox.org/tasmota/ binares will do?

arendst, 2019-10-02:
latest binaries should do now.

You might want to check first that ``mqttlog`` is set to 0

gajotnt, 2019-10-02:
Flashed the latest, if i set the mqttlog to 0 then the messages of command unknown stop showing

arendst, 2019-10-02:
So on some devices MQTTlog is set to non-zero and will fail. As the MQTTlog parameter was introduced only a few commits ago chances are the MQTTlog value variable was used before in previous code. That's why some devices do show this problem and others don't

I will have to reconsider the MQTTLog code.

cyablo, 2019-10-02:
Works now with mqttlog set to 0

arendst, 2019-10-02:
Solved in latest commit https://github.com/arendst/Sonoff-Tasmota/commit/8c34b9edbd9efc3405c1c29324e347f6accd8de1

closed 2019-10-02 by arendst