arendst/Tasmota #6530: Home Assistant discovery: switch and buttons are not transmitted.

effelle, 2019-10-01:
### BUG DESCRIPTION
As per Wiki, Home Assistant discovery should create button and switches under `/binary_sensor/%s/config` automatically, independently by the presence of a relay. Actually the switches and buttons aren't created at all. 
As a first inspection using some `LOG_LEVEL_DEBUG`, seems `void HAssAnnounceButtonSwitch()` is never reached even if `void HAssAnnounceSwitches(void)` and `void HAssAnnounceButtons(void)` are.
A more simple setup like switch1/relay1 will show the same issue: just the relay will be sent to Home Assistant.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): D1 Mini and NodeMCU
- [X] Tasmota binary firmware version number used: 6.6.0.14
  - [X] Pre-compiled 
  - [X] Self-compiled
    - [X] IDE / Compiler used: Gitpod
- [X] Flashing tools used: Esptools.py
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
03:45:43 CMD: Backlog Template; Module; GPIO
03:45:43 SRC: WebConsole from 192.168.15.199
03:45:43 SRC: Backlog
03:45:43 MQT: hadiscovery_dev/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
03:45:43 SRC: Backlog
03:45:43 MQT: hadiscovery_dev/stat/RESULT = {"Module":{"18":"Generic"}}
03:45:44 SRC: Backlog
03:45:44 MQT: hadiscovery_dev/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"9":"Switch1"},"GPIO5":{"10":"Switch2"},"GPIO12":{"17":"Button1"},"GPIO13":{"18":"Button2"},"GPIO14":{"19":"Button3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}    
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
```
 No rules. 
```
- [X] Provide the output of this command: ``Status 0``:
```
03:50:17 CMD: status 0
03:50:17 SRC: WebConsole from 192.168.15.199
03:50:17 MQT: hadiscovery_dev/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["hadiscovery_dev1"],"Topic":"hadiscovery_dev","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:50:17 MQT: hadiscovery_dev/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:19","StartupUTC":"2019-10-01T02:43:58","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":22,"SaveAddress":"F6000"}}
03:50:17 MQT: hadiscovery_dev/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.14(0e240bb-sonoff)","BuildDateTime":"2019-09-30T23:10:16","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EffE-Domus",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","280500000100060000005A64000000000000","00000200"]}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS5 = {"StatusNET":{"Hostname":"hadiscovery_dev-4216","IPAddress":"192.168.15.30","Gateway":"192.168.15.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"CC:50:E3:47:10:78","Webserver":2,"WifiConfig":4}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.15.102","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_471078","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 01 02:50:18 2019","Local":"Tue Oct 01 03:50:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:49","Sunset":"18:30"}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-01T03:50:18","Switch1":"ON","Switch2":"ON"}}
03:50:18 MQT: hadiscovery_dev/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-01T03:50:18","Uptime":"0T00:06:20","UptimeSec":380,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"EffE-Domus","BSSId":"94:EA:EA:97:2A:A7","Channel":1,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:04"}}} 
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
```
03:47:04 CMD: setoption19 1
03:47:04 SRC: WebConsole from 192.168.15.199
03:47:04 MQT: hadiscovery_dev/stat/RESULT = {"SetOption19":"ON"}
03:47:05 MQT: homeassistant/light/471078_LI_1/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_1/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_2/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_2/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_3/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_3/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_4/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_4/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_5/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_5/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_6/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_6/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_7/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_7/config =  (retained)
03:47:05 MQT: homeassistant/light/471078_LI_8/config =  (retained)
03:47:05 MQT: homeassistant/switch/471078_RL_8/config =  (retained)
03:47:05 MQT: homeassistant/sensor/471078_status/config = {"name":"hadiscovery_dev1 status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"471078_status","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]],"name":"hadiscovery_dev1","model":"Generic","sw_version":"6.6.0.14(0e240bb-sonoff)","manufacturer":"Tasmota"},"~":"hadiscovery_dev/tele/"} (retained)
``` 

### TO REPRODUCE
Setup a device with just swithces and buttons and enable `setoption19`.

### EXPECTED BEHAVIOUR
All the switches and buttons should be created on Home Assistant as `binary_sensor`.

effelle, 2019-10-01:
@ascillato2 Adrian why mark it as a feuture request? We have the code to manage button and switches already in place on `xdrv_12_home_assistant.ino` starting from [L298](https://github.com/arendst/Sonoff-Tasmota/blob/c311632b7a4ad3ae89e3f2984ac53c30ecb0d16e/sonoff/xdrv_12_home_assistant.ino#L298) and the [wiki ](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#add-tasmota-devices-to-home-assistant) report it as a normal behavior.

effelle, 2019-10-02:
@arendst, I've found the issue. 
Buttons and switches are not created when using the default `switchtopic` / `buttontopic`.
Just set them with any other value than `0` and the entities will be created as expected:
```
16:01:57 CMD: backlog switchtopic; buttontopic; gpio;
16:01:57 MQT: HA_DEV7/stat/RESULT = {"SwitchTopic":"HA_DEV7"}
16:01:57 MQT: HA_DEV7/stat/RESULT = {"ButtonTopic":"HA_DEV7"}
16:01:57 MQT: HA_DEV7/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"18":"Button2"},"GPIO12":{"9":"Switch1"},"GPIO13":{"10":"Switch2"},"GPIO14":{"11":"Switch3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
16:02:01 CMD: setoption19 1
16:02:01 MQT: HA_DEV7/stat/RESULT = {"SetOption19":"ON"}
16:02:01 MQT: homeassistant/light/471078_LI_1/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_1/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_2/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_2/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_3/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_3/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_4/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_4/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_5/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_5/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_6/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_6/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_7/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_7/config =  (retained)
16:02:01 MQT: homeassistant/light/471078_LI_8/config =  (retained)
16:02:01 MQT: homeassistant/switch/471078_RL_8/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_BTN_1/config = {"name":"HA_DEV7 BTN","stat_t":"~cmnd/POWER","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_BTN_1","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"HA_DEV7/","frc_upd":true,"pl_off":"OFF"} (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_BTN_2/config = {"name":"Sonoff2 BTN","stat_t":"~cmnd/POWER","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_BTN_2","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"HA_DEV7/","frc_upd":true,"pl_off":"OFF"} (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_BTN_3/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_BTN_4/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_1/config = {"name":"HA_DEV7 SW","stat_t":"~cmnd/POWER","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_SW_1","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"HA_DEV7/","frc_upd":true,"pl_off":"OFF"} (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_2/config = {"name":"Sonoff2 SW","stat_t":"~cmnd/POWER","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_SW_2","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"HA_DEV7/","frc_upd":true,"pl_off":"OFF"} (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_3/config = {"name":"Sonoff3 SW","stat_t":"~cmnd/POWER","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_SW_3","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"HA_DEV7/","frc_upd":true,"pl_off":"OFF"} (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_4/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_5/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_6/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_7/config =  (retained)
16:02:01 MQT: homeassistant/binary_sensor/471078_SW_8/config =  (retained)
16:02:01 MQT: homeassistant/sensor/471078_status/config = {"name":"HA_DEV7 status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"471078_status","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]],"name":"HA_DEV7","model":"Generic","sw_version":"6.6.0.14(sonoff)","manufacturer":"Tasmota"},"~":"HA_DEV7/tele/"} (retained))
```
However, this will create issues with normal behavior.  
Now, my question: is there a real interest on HA integration? I know you don't like it much and I can't blame you for that in some way, but we have the possibility to transform it in a fully functional tool for BOTH Home Assistant and OpenHab, since the latter can now identify MQTT device with HA format, even if some adjustment are needed for JsonPath.  
I know the code size and ram usage are an issue, then how about start to think to remove it form the main .bins to free space for more interesting stuff and create another firmware sonoff-ha.bin?
@meingraham @ascillato2 @Jason2866 @jziolkowski what are your thoughts?

ascillato, 2019-10-02:
As there is a important number of users that have Home Assistant as its home automation software, I don't think to be a good idea to disable it by default.

meingraham, 2019-10-02:
I think if there are new features that begin to push against the flash and iRAM limits of the ESP82xx, then HA could be considered a "user option". We can consider providing "pre-made" `user_config_override.h` for HA users to easily compile using Gitpod... or, now that @andrethomas has optimized firmware compilation using PlatfromIO caching on thehackbox, add another branch of binary variants for HA. When 2.6 comes out, there may be an opportunity to have HA variants which are only 2.6. If a user want HA on another core, they can compile themselves.

andrethomas, 2019-10-02:
It's important to recognise that the firmware capabilities are expanding in different directions at the same time and that the time of having a sonoff.bin "one fits all solution" is going to become a reality in the not so distant future.

To me, its not just about HA vs. Non-HA support but also that we're adding drivers for chips which will never be connected to let's say a tuya based light bulb and as nice as it might be to have an LM75AD I2C temperature sensor inside the bulb it's not going to be a very likely combination.

I don't mind adding additional variants but it should be planned carefully otherwise choosing a binary will need a wizard form of some sort on its own... which of course is not ideal.

effelle, 2019-10-02:
I like Mike's idea and Andre has a strong point. Tasmota is growing with an impressive pace.

PR #6548 (2019-10-02): Home Assistant integration BTN/SW
## Description:
Removed a `strcmp` for `buttontopic` and `switchtopic` name. HA integration must enable buttons and switches independently from the name.

**Related issue (if applicable):** fixes #6530

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [C [... cut]
closed 2019-10-02 by arendst