arendst/Tasmota #6531: Hiking DDS238-2 ZN/S energy meter not displaying all options.

myounges, 2019-10-01:
I have flashed the latest Tasmota-sensors.bin version to a WeMos D1 mini and everything seems to work fin but, some of the modbus registers are not appearing neither on the web UI or the MQTT data. They show as 0, but on the meter you **can** see this information on the LCD screen.
I would assume that the issue is related to the ModBus [registers](https://gist.github.com/alphp/95e1efe916c0dd6df7156f43dd521d53#hiking-dds238-2-zns-energy-meter) are not correct for these data points that appear as 0. 

[image]

Apologies if this bug report is not correct, I tried searching the forum and checked if anyone already came up with this problem with no results.

I did see the info of the pull [here](https://github.com/arendst/Sonoff-Tasmota/pull/6384) Not sure at what extent this was tested.

I'm happy to troubleshoot the problem as I have the hardware and not all have it I gueess the file to check is xnrg_09_dds2382.ino but not sure how to test other registers.

Best regards and thank you in advance.

Jason2866, 2019-10-01:
Tasmota does not support all possible options from all devices.
It supports a common set of data. 
What you see in webfrontend and the transmitted data is as designed.

ascillato2, 2019-10-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

myounges, 2019-10-01:
Thank you for the reply just checking the module for this meter and I see the following variables declared:
`TasmotaSerial *DDS2382Serial;

uint32_t dds2382_state = DDS2382_STATE_READY;
uint8_t dds2382_rx_buffer[48];
uint8_t dds2382_nodata_count = 0;

struct {
  uint16_t voltage = 0;
  uint16_t current = 0;
  int16_t active_power = 0;
  uint16_t reactive_power = 0;
  uint16_t power_factor = 0;
  uint16_t frequency = 0;
  uint32_t energy_total = 0;
  uint16_t import_active = 0;
  uint16_t export_active = 0;
} dds2382_data;
`
This is why I was asking because clearly they are in the code to be retrieved from the modbus registers.

Best regards.

pablozg, 2019-10-01:
Hi, What tasmota version are you using? 

Try with this version: http://thehackbox.org/tasmota/pre-2.6/sonoff-sensors.bin

myounges, 2019-10-01:
> Hi, What tasmota version are you using?
> 
> Try with this version: http://thehackbox.org/tasmota/pre-2.6/sonoff-sensors.bin

Version 6.6.0.14(Sensors). Sep 29th

pablozg, 2019-10-01:
After take a look to the code, I don't see any reason to not work.

I use a sdm120 and the code is very similar, working it without issues, maybe theo can take a look. 

Maybe you can try to read the meter with a pc and python or php code, to be sure the meter is sending all data.

myounges, 2019-10-01:
Yes I see that the registers declared for SDM120 SDM220 SDM230 SDM630 seem to work but there has to be something different with the DDS238-2 ZN/S registers, is there any way I can confirm this info is correct in the code?
I'm mainly interested in these two:
0008h-0009h | export energy | 1/100 kWh | unsigned dword | R¹
000Ah-000Bh | import energy | 1/100 kWh | unsigned dword | R¹

All registrers are below.
`
Register(s) | Meaning | Scale Unit | Data format | R/W
-- | -- | -- | -- | --
0000h-0001h | total energy | 1/100 kWh | unsigned dword | R¹
0002h-0003h | reserved |   | unsigned dword |  
0004h-0005h | reserved |   | unsigned dword |  
0006h-0007h | reserved |   | unsigned dword |  
0008h-0009h | export energy | 1/100 kWh | unsigned dword | R¹
000Ah-000Bh | import energy | 1/100 kWh | unsigned dword | R¹
000Ch | voltage | 1/10 V | unsigned word | R
000Dh | current | 1/100 A | unsigned word | R
000Eh | active power | 1 W | signed word | R
000Fh | reactive power | 1 VAr | unsigned word | R
0010h | power factor | 1/1000 | unsigned word | R
0011h | frequency | 1/100 Hz | unsigned word | R
0012h | reserved |   | unsigned word |  
0013h | reserved |   | unsigned word |  
0014h | reserved |   | unsigned word |  
0015h:high | station address | 1-247 | unsigned char | R/W
0015h:low | baud rate | 1-4² | unsigned char

`

Thank for the help!

Cheers

pablozg, 2019-10-01:
Hi, If you are able to compile tasmota, please change this lines in xnrg_09_dds2382.ino:

````
Energy.export_active = (float)((buffer[11] << 24) + (buffer[12] << 16) + (buffer[13] << 8) + buffer[14]) / 100.0;  // 429496729.0 W
float import_active = (float)((buffer[15] << 24) + (buffer[16] << 16) + (buffer[17] << 8) + buffer[18]) / 100.0;  // 429496729.0 W
````
by
````
Energy.export_active = (float)((buffer[19] << 24) + (buffer[20] << 16) + (buffer[21] << 8) + buffer[22]) / 100.0;  // 429496729.0 W
float import_active = (float)((buffer[23] << 24) + (buffer[24] << 16) + (buffer[25] << 8) + buffer[26]) / 100.0;  // 429496729.0 W
````

and test it.

myounges, 2019-10-01:
Will test in a while and let you know.

Thank you very much.

myounges, 2019-10-01:
> Hi, If you are able to compile tasmota, please change this lines in xnrg_09_dds2382.ino:
>
> ```
> Energy.export_active = (float)((buffer[11] << 24) + (buffer[12] << 16) + (buffer[13] << 8) + buffer[14]) [...]

**We got a winner!** Worked like a charm! @pablozg **Thank you very much**, all my respects to you. I would suggest to add this modification to future versions for the dds238-2 meter as it is reading all values correctly now.
[image]

ascillato2, 2019-10-01:
Hi,

Thanks for keeping working on this.

@pablozg

Could you make a Pull Request with this change? Thanks.

pablozg, 2019-10-01:
Hi, tomorrow I will make a PR, but I need to talk with @arendst about it.

We have to be sure that not exist two versions of the same hardware but with different registers before merge it.

All info I was found show the same register config as the showed by @myounges in previous messages.

arendst, 2019-10-02:
I agree with @pablozg to make sure this change doesn't break the previous implementation provided by #6384 with result https://github.com/arendst/Sonoff-Tasmota/pull/6384#issuecomment-529960006

So there might be different hardware versions around...

myounges, 2019-10-02:
I can provide further info of the meter if required. But based on the manufacturer, they did not state any changes in hardware for this model. Chinese model... I know. :)
Let me know if you need something from my side, happy to return the favor.

Thanks,

pablozg, 2019-10-02:
Maybe @matteo-campanella can help us and send his device modbus protocol.

@arendst I see you used the original implementation from matteo so don't take account my first question in the PR.

arendst, 2019-10-02:
This is the current state:
```
      //           0     1     2     3     4     5     6     7     8     9     A     B     C     D     E     F    10    11           = ModBus register
      //  0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40  = Buffer index
      // SA FC BC EnergyTotal             ExportActiv ImportActiv                         Volta Curre APowe RPowe PFact Frequ Crc--  = DDS238-2 ZN/S version 1 (#6384)
      // SA FC BC EnergyTotal                                     ExportActiv ImportActiv Volta Curre APowe RPowe PFact Frequ Crc--  = DDS238-2 ZN/S version 2 (#6531)
```

arendst, 2019-10-02:
Use new command ``SetOption71 0 or 1`` to switch between different Modbus registers for Active Energy.

ascillato2, 2019-10-02:
Closing this issue as it has been fixed. Thanks

closed 2019-10-02 by ascillato2

RedCat30, 2020-01-07:
Hello

I have Hiking DDS238-2 ZN/S energy meter too. I have flashed the latest Tasmota-sensors.bin version (7.2.0)
I found this page yesterday. So far for me, energy today, yesterday, total, export active showed zero. Yesterday I tried SetOption71 1, it worked partially. Instead of zeros there are values, only wrong values.
Take a look at the two pictures. One is the inverter, today's output is 1.5 kWh, the other is Tasmota, today production is 0.001 kWh. Yesterday's values were also bad, according to the inverter 7.92 kWh.

[image]
[image]

Voltage, power, current, frequency values are good.
Is it possible that this is a third version of DDS238? Can anyone help me how to get the right values (Today, yesterday, total)?
Thanks!

thiagoarreguy, 2020-06-05:
> **We got a winner!** Worked like a charm! @pablozg **Thank you very much**, all my respects to you. I would suggest to add this modification to future versions for the dds238-2 meter as it is [...]

Can you share the GPIO configuration for a Generic Module on Tasmota?

thiagoarreguy, 2020-06-05:
> **We got a winner!** Worked like a charm! @pablozg **Thank you very much**, all my respects to you. I would suggest to add this modification to future versions for the dds238-2 meter as it is [...]

I am trying to work in this device: https://a.aliexpress.com/_dVtfr0j

I am getting Error 7: 

DDS2382 response error 7

what means "Not enough data" on modbus communication, ReceiveBuffer function.

**5:53:10** CMD: gpio 255
**05:53:10** SRC: WebConsole from 192.168.0.108
**05:53:10** CMD: Group 0, Index 1, Command "GPIO", Data "255"
**05:53:10** RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"176":"DDS238-2 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"177":"DDS238-2 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

**04:18:41** CMD: gpios

**04:18:41** RSL: stat/tasmota/RESULT = {"GPIOs1":{"0":"None","17":"Button1","90":"Button1n","122":"Button1i","126":"Button1in","18":"Button2","91":"Button2n","123":"Button2i","127":"Button2in","19":"Button3","92":"Button3n","124":"Button3i","128":"Button3in","20":"Button4","93":"Button4n","125":"Button4i","129":"Button4in","9":"Switch1","82":"Switch1n","10":"Switch2","83":"Switch2n","11":"Switch3","84":"Switch3n","12":"Switch4","85":"Switch4n","13":"Switch5","86":"Switch5n","14":"Switch6","87":"Switch6n","15":"Switch7","88":"Switch7n","16":"Switch8","89":"Switch8n","21":"Relay1","29":"Relay1i"}}

**04:18:41** RSL: stat/tasmota/RESULT = {"GPIOs2":{"22":"Relay2","30":"Relay2i","23":"Relay3","31":"Relay3i","24":"Relay4","32":"Relay4i","25":"Relay5","33":"Relay5i","26":"Relay6","34":"Relay6i","27":"Relay7","35":"Relay7i","28":"Relay8","36":"Relay8i","52":"Led1","56":"Led1i","53":"Led2","57":"Led2i","54":"Led3","58":"Led3i","55":"Led4","59":"Led4i","157":"LedLink","158":"LedLinki","37":"PWM1","46":"PWM1i","38":"PWM2","47":"PWM2i","39":"PWM3","48":"PWM3i","40":"PWM4","49":"PWM4i","41":"PWM5","50":"PWM5i","42":"Counter1","94":"Counter1n","43":"Counter2","95":"Counter2n","44":"Counter3"}}

**04:18:41** RSL: stat/tasmota/RESULT = {"GPIOs3":{"96":"Counter3n","45":"Counter4","97":"Counter4n","160":"Buzzer","161":"Buzzeri","148":"Serial Tx","149":"Serial Rx","5":"I2C SCL","6":"I2C SDA","1":"DHT11","2":"AM2301","3":"SI7021","193":"DHT11o","4":"DS18x20","192":"DS18x20o","7":"WS2812","143":"MY92x1 DI","144":"MY92x1 DCKI","140":"SM16716 CLK","141":"SM16716 DAT","142":"SM16716 PWR","180":"SM2135 Clk","181":"SM2135 Dat","107":"Tuya Tx","108":"Tuya Rx","201":"MOODL Tx","8":"IRsend","51":"IRrecv","105":"RFSend","106":"RFrecv","116":"RF Sensor","73":"SR04 Tri/TX","74":"SR04 Ech/RX"}}

**04:18:41** RSL: stat/tasmota/RESULT = {"GPIOs4":{"79":"TM16 CLK","80":"TM16 DIO","81":"TM16 STB","102":"HX711 SCK","103":"HX711 DAT","130":"HLWBL SEL","131":"HLWBL SELi","132":"HLWBL CF1","133":"HLW8012 CF","134":"BL0937 CF","156":"ADE7953 IRQ","145":"CSE7766 Tx","146":"CSE7766 Rx","135":"MCP39F5 Tx","136":"MCP39F5 Rx","137":"MCP39F5 Rst","62":"PZEM0XX Tx","63":"PZEM004 Rx","98":"PZEM016 Rx","99":"PZEM017 Rx","75":"SDMx20 Tx","76":"SDMx20 Rx","77":"SDM630 Tx","78":"SDM630 Rx","178":"DDSU666 Tx","179":"DDSU666 Rx","71":"SerBr Tx","72":"SerBr Rx","60":"MHZ Tx","61":"MHZ Rx","64":"SAir Tx"}}

**04:18:41** RSL: stat/tasmota/RESULT = {"GPIOs5":{"65":"SAir Rx","101":"SDS0X1 Tx","70":"SDS0X1 Rx","189":"HPMA Tx","188":"HPMA Rx","203":"PMS5003 Tx","69":"PMS5003 Rx","100":"MP3 Player","138":"PN532 Tx","139":"PN532 Rx","167":"RDM6300 RX","168":"iBeacon TX","169":"iBeacon RX","195":"HM10 TX","194":"HM10 RX","119":"MX31855 CS","120":"MX31855 CLK","121":"MX31855 DO","154":"HRE Clock","155":"HRE Data","182":"DeepSleep","200":"HRXL Rx","202":"AS3935"}}

thiagoarreguy, 2020-06-05:
Talking with chinese guy from Hiking, Peter, he send me the protocol header:

[image]

@arendst maybe you can check if its all right, please.

kugelkopf123, 2020-06-05:
>Talking with chinese guy from Hiking, Peter, he send me the protocol header: ...

@thiagoarreguy with this answer you’ve got everything you need for the sml driver from Tasmota. https://tasmota.github.io/docs/Smart-Meter-Interface/#sdm530 
Just have a look at this stuff. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

thiagoarreguy, 2020-06-05:
[image]

better detail on registers

kugelkopf123, 2020-06-05:
> [image]
> 
> better detail on registers

It is on yourself to compile a Version with Scripter and SML Support. Everything you need is in the Docs. The link I gave to you was just an example of how it works with a sdm530. There are a lot more examples for Meters with RS485.

thiagoarreguy, 2020-06-05:
15:53:31 DBG: DDS2382 response error 7
15:53:33 DMP: 01 55 AA 03 07 00 08 13 02 00 04 00 01 22 E0 2D
15:53:33 DBG: DDS2382 response error 9
15:53:35 DMP: 01 22 62 AF
15:53:35 DBG: DDS2382 response error 7
15:53:36 DMP: 01 21 DA 26
15:53:36 DBG: DDS2382 response error 7
15:53:37 DMP: 
15:53:37 DBG: DDS2382 response error 7
15:53:39 DMP: 01 21 74 C0
15:53:39 DBG: DDS2382 response error 7
15:53:41 DMP: 
15:53:41 DBG: DDS2382 response error 7
15:53:44 WIF: Checking connection...
15:53:44 DMP: 01 20 F9 44
15:53:44 DBG: DDS2382 response error 7
15:53:46 DMP: 01 20 7D C8
15:53:46 DBG: DDS2382 response error 7
15:53:50 DMP: 01 20 E8 33 55 AA 03 07 00 08 65 02 00 04 00 00 0A CC 52
15:53:50 DBG: DDS2382 response error 9
15:53:51 DMP: 01 21 6B B7
15:53:51 DBG: DDS2382 response error 7
15:53:53 DMP: 01 21 F4 40
15:53:53 DBG: DDS2382 response error 7
15:53:55 DMP: 
15:53:55 DBG: DDS2382 response error 7
15:53:56 DMP: 01 22 D0 1D
15:53:56 DBG: DDS2382 response error 7
15:53:58 DMP: 01 21 33 7F
15:53:58 DBG: DDS2382 response error 7
15:53:59 DMP: 01 20 1C 67
15:53:59 DBG: DDS2382 response error 7
15:54:00 DMP: 01 1F 9F E9
15:54:00 DBG: DDS2382 response error 7
15:54:02 DMP: 01 1F 09 53
15:54:02 DBG: DDS2382 response error 7
15:54:03 DMP: 01 1F A3 ED
15:54:03 DBG: DDS2382 response error 7
15:54:04 WIF: Checking connection...
15:54:05 DMP: 
15:54:05 DBG: DDS2382 response error 7
15:54:06 DMP: 01 20 10 5B
15:54:06 DBG: DDS2382 response error 7
15:54:07 DMP: 01 21 15 61
15:54:07 DBG: DDS2382 response error 7
15:54:08 DMP: 01 18 82 C5
15:54:08 DBG: DDS2382 response error 7
15:54:09 DMP: 01 19 55 99
15:54:09 DBG: DDS2382 response error 7
15:54:10 DMP: 01 1B ED 33
15:54:10 DBG: DDS2382 response error 7
15:54:11 DMP: 01 1F 8C D6
15:54:11 DBG: DDS2382 response error 7
15:54:12 DMP: 01 21 B7 03
15:54:12 DBG: DDS2382 response error 7
15:54:14 DMP: 01 21 28 74
15:54:14 DBG: DDS2382 response error 7
15:54:15 DMP: 01 20 A7 F2
15:54:15 DBG: DDS2382 response error 7
15:54:16 DMP: 01 1F FD 47
15:54:16 DBG: DDS2382 response error 7
15:54:17 DMP: 01 1F 4D 97
15:54:17 DBG: DDS2382 response error 7
15:54:19 DMP: 01 1E A2 EB
15:54:19 DBG: DDS2382 response error 7
15:54:20 DMP: 
15:54:20 DBG: DDS2382 response error 7
15:54:21 DMP: 01 1E 20 69
15:54:21 DBG: DDS2382 response error 7
15:54:22 DMP: 01 1D 9C E4
15:54:22 DBG: DDS2382 response error 7
15:54:24 WIF: Checking connection...
15:54:24 DMP: 
15:54:24 DBG: DDS2382 response error 7
15:54:25 DMP: 01 1E 4D 96
15:54:25 DBG: DDS2382 response error 7
15:54:26 DMP: 01 1D E8 30
15:54:26 DBG: DDS2382 response error 7
15:54:27 DMP: 01 1C EB 32
15:54:27 DBG: DDS2382 response error 7
15:54:28 DMP: 01 1C 64 AB
15:54:28 DBG: DDS2382 response error 7
15:54:30 DMP: 
15:54:30 DBG: DDS2382 response error 7
15:54:32 DMP: 01 1D 08 50
15:54:32 DBG: DDS2382 response error 7
15:54:34 DMP: 
15:54:34 DBG: DDS2382 response error 7
15:54:37 DMP: 01 1C 7E C5
15:54:37 DBG: DDS2382 response error 7
15:54:39 DMP: 
15:54:39 DBG: DDS2382 response error 7
15:54:42 DMP: 01 1B 03 49
15:54:42 DBG: DDS2382 response error 7
15:54:43 DMP: 01 19 53 97
15:54:43 DBG: DDS2382 response error 7
15:54:44 WIF: Checking connection...
15:54:44 DMP: 01 18 40 83 55 AA 03 07 00 08 65 02 00 04 00 00 0A D7 5D
15:54:44 DBG: DDS2382 response error 9
15:54:45 DMP: 01 17 5C 9E
15:54:45 DBG: DDS2382 response error 7
15:54:47 DMP: 01 16 F7 38
15:54:47 DBG: DDS2382 response error 7
15:54:48 DMP: 01 17 8B CD
15:54:48 DBG: DDS2382 response error 7
15:54:49 DMP: 
15:54:49 DBG: DDS2382 response error 7
15:54:52 DMP: 01 17 1E 60
15:54:52 DBG: DDS2382 response error 7
15:54:53 DMP: 01 17 AF F1
15:54:53 DBG: DDS2382 response error 7
15:54:54 DMP: 01 18 51 94 55 AA 03 07 00 08 65 02 00 04 00 00 0A D9 5F
15:54:54 DBG: DDS2382 response error 9
15:54:55 HTP: Console
15:54:56 DMP: 01 18 FD 40
15:54:56 DBG: DDS2382 response error 7
15:54:57 DMP: 01 17 01 43
15:54:57 DBG: DDS2382 response error 7
15:54:58 DMP: 01 14 C6 05
15:54:58 DBG: DDS2382 response error 7
15:54:59 DMP: 01 12 37 74 55 AA 03 07 00 08 65 02 00 04 00 00 0A DA 60
15:54:59 DBG: DDS2382 response error 9
15:55:00 DMP: 01 10 08 43
15:55:00 DBG: DDS2382 response error 7
15:55:02 DMP: 01 0B C8 FE
15:55:02 DBG: DDS2382 response error 7
15:55:03 DMP: 01 08 D2 05
15:55:03 DBG: DDS2382 response error 7
15:55:04 WIF: Checking connection...
15:55:04 DMP: 01 06 94 C5
15:55:04 DBG: DDS2382 response error 7
15:55:05 DMP: 01 05 2E 5E
15:55:05 DBG: DDS2382 response error 7
15:55:06 DMP: 01 04 9F CE
15:55:06 DBG: DDS2382 response error 7
15:55:10 DMP: 01 05 EA 1A
15:55:10 DBG: DDS2382 response error 7
15:55:11 DMP: 01 07 E4 16
15:55:11 DBG: DDS2382 response error 7
15:55:12 DMP: 01 09 70 A4
15:55:12 DBG: DDS2382 response error 7
15:55:14 DMP: 01 08 42 75
15:55:14 DBG: DDS2382 response error 7
15:55:15 DMP: 01 06 88 B9 55 AA 03 07 00 08 65 02 00 04 00 00 0A DD 63
15:55:15 DBG: DDS2382 response error 9

ShurDan, 2020-06-06:
> xnrg_09_dds2382

I can not configure this device
please share the settings

Badwater-Frank, 2021-01-20:
Even this issue is closed, I may point out another issue on the displayed data which can be seen in the initial post. This displays the energy export state; guessing this from reactive power value of 65358 VAr, power and apparent power indicating the same value of 986.
I have the DDS238-2 with CT sensor and compared the same load by turning the CT on the 220V-line; "reference" is Gosund smart plug (Tasmota 8.5.0). DDS238 is connected to a nodemcu via rs485-converter, running Tasmota 9.2.0. Connection/configuration: TX/GPIO1 -> DDS238-2 TX; RX/GPIO3 -> DDS238-2 RX
In case of energy import things look fine:
[image]

In case of energy export the above issue shows up: reactive power is 65xxx, power and apparent power is always the same value:
[image]

I one case the export values are ok: when the power factor is 1.00!

Sniffed data on the rs485:
```
import:
-- request
01 03 00 00 00 12 c5 c7 
-- answer
01 03 24
-- data
-- Total Energy
00 00 01 86 = 390 -> 3.90
--
00 00 00 00 
00 00 00 00 
00 00 00 00 
-- export
00 00 00 0f = 15 ->
-- import
00 00 01 77 = 375 -> 3.75
-- volt
09 30 = 2352 -> 235.2V
-- current
00 0e = 14 -> 0,14A
-- AP
00 1d = 29
-- RP
00 13 = 19
-- PF
03 3f = 831 -> 0.83
-- Freq
13 89 = 5001 -> 50.01
-- CRC
-- CRC
89  9e  
```

```
export:
-- request
01 03 00 00 00 12 c5 c7 
-- answer
01 03 24 
-- data
-- Total Energy
00 00 01 87 = 391 -> 3.91
--
00 00 00 00 
00 00 00 00 
00 00 00 00
-- 
00 00 00 0f = 15
-- 
00 00 01 78 = 376
--
09 2a = 2346 -> 234.6V
-- current
00 0f = 15 -> 0.15A
-- AP
ff e3 = 65507 -> 29 (65536-65507)
-- RP
ff ed = 65517 -> 19 (65536-65517)
-- PF
03 3a = 826 -> 82.6
-- Freq.
13 88 = 5000 -> 50.00
-- CRC
-- CRC
34  08
```

GeorgeWatch, 2021-12-19:
Hi
Modbus [DDS238-2](https://gist.github.com/alphp/95e1efe916c0dd6df7156f43dd521d53)
Modbus [DTS 238-7](https://github.com/arendst/Tasmota/files/7740684/DTS.238-7.Manual.RS485.Modbus.Protokol.pdf)

iosifnicolae2, 2022-12-03:
I've created some documentation on how to connect DTS238-7 to Tasmota here: https://github.com/arendst/Tasmota/discussions/17248