arendst/Tasmota #6534: FEIT BPA800/RGBW/AG/2(P) Template issues

Resterday, 2019-10-01:
Good morning all, I recently flashed some FEIT BPA800/RGBW/AG/2(P) bulbs I bought at Costco with kueblc's new-api branch of tuya-convert. I then updated to 6.6.0. 

I have tried the following template for FEIT bulbs `{"NAME":"OM60/RGBW","GPIO":[255,255,255,255,140,37,0,0,38,142,141,255,255],"FLAG":0,"BASE":18}` but when the bulb first re-sets and I toggle it on, nothing happens, and there are no errors or other messages in the console. If I put in `color 0,0,50,0,0` into console then teh bulb turns on at 20% brightness and a cool white. At this point, I can toggle it on and off but if I try to change any other settings the bulb just goes dark and I cant do anything until i put `color 0,0,50,0,0` into console again. I have tried other template settings, but so far nothings working.

I opened up the bulb when trying to get the tuya convert to work and noticed that it appears the led driver might be different from other FEIT bulbs, but Im not sure. Maybe this has something to do with my issues? Its really hard to see whats written on the component but I have included the pictures that I took.

[image]
[image]

Hoping someone can help me identify next steps for getting these bulbs to work. Thanks!

jziolkowski, 2019-10-01:
Paging Dr. @shantur :)

shantur, 2019-10-02:
@Resterday : Have you tried https://github.com/arendst/Sonoff-Tasmota/wiki/TuyaMCU and https://github.com/arendst/Sonoff-Tasmota/wiki/MCU-Based-Tuya-Dimmers-and-Switches

digiblur, 2019-10-02:
Don't think that's going to work as they used the SM16716 before.  If that is indeed the LED controller, they definitely shifted to a new chip.  Anyone had any luck googling for datasheets?

shantur, 2019-10-02:
@Resterday : Can you try to get a clearer photo of chip U1 or the name on it.

Resterday, 2019-10-02:
@shantur - I'll borrow a friend's dslr and try and get a better picture tonight. That was the best I could do with my phone. I think the text is:

SM726EC
AFAH RWAJ22

but I have been unable to find anything on Google with that.

mhightower83, 2019-10-02:
I found this comment:
> ... the color is driven by a [SM726EB](http://chinaasic.com/product.jsp#item=series#style=21#id=104) protocol equivalent to the SM16716/SM16726.
https://github.com/xoseperez/espurna/issues/568#issuecomment-455820546

However, I have not had much luck with changing the definitions around.
GPIO4 appears to be the Warm LED pin for PWM
and maybe GPIO13 is the power for the color.  LEDs
Cannot get anything else to work.

Photo64, 2019-10-02:
I am also seeing this issue with the Costco Feit bulb. I tried 504 combinations for the SM16716's and none of them worked.

mhightower83, 2019-10-03:
This is getting closer:
```
{"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,38,37,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}
SetOption37 0
```

Note on the older models we needed `SetOption37 54` to get the colors in the right order. Now it needs to be set to `0`. With this, the color LEDs work using the `color` command.

Something is still off on the Cold to Warm white LEDs. All the way over for Warm look right, however, as you move toward the center it goes to Cold White then as you move further down it dims.

Photo64, 2019-10-03:
I just tried this and can confirm I am getting color now but as you mentioned the cold white is acting strangely

mhightower83, 2019-10-03:
Looks like the BP800/RGBW/AG/2(P) has two key chips for controlling the WarmWhite and ColdWhite LEDs, the BP5926 and MT7816CS.

The [BP5926](http://www.bpsemi.com.cn/uploads/file/20181017105105_356.pdf) appears to control the ratio of Cold White to Warm White by PWM input. Here is a relevant clip from google translate: 
>The duty cycle is used to adjust the ratio of the illumination of the two LED sources.
To achieve the purpose of coloring. Two LED output currents complement each other in toning
The total current in the process is constant and equal to the current of the constant current source.

The [MT7816CS](http://www.maxictech.com/UpLoadFile/Files/2018/10/19/113511293fb454363-f.pdf)  appears to be a PWM dimmable constant current LED driver.

If you add [`SetOption68 1`](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Setoption68), then use the color command. Channel 4 will control the Cold White to Warm White ration, 0xFF all cold white, 0x00 all Warm White. Channel 5 controls the brightness, at and below 0x0E seems to turn off. At and above 14 appears to turn on and increases in brightness as you go up. Don't know how you tie this into the sliders on the top screen.

>In a way, this change is an improvement over the previous version. With the old bulbs, you could use the color command to turn on Warm White and Cold White each at full brightness. The bulb became very bright and drew 18 watts instead of its rated 9 watts. When I saw the power draw I quickly changed the settings to avoid overheating the power supply. With this new design that cannot happen.

Photo64, 2019-10-03:
> If you add SetOption68 1, then use the color command. Channel 4 will control the Cold White to Warm White ration, 0xFF all cold white, 0x00 all Warm White. Channel 5 controls the brightness, at and [...]

After setting setoption68 1  I now have 5 toggles on the main screen however they don't control anything until I enter a color command in the console. Is this the expected behavior? If a enter color 0,0,255,0,255 I can then click on/off the 3rd channel button for the blue color or I can click on/off for the 5th channel button for full WW but I can't turn on/off any other color channel. Is the intent to send a color command like color 255,255,255,255,255 so it allows control through the buttons and sliders on the main page? Although if I want to vary the color I'd still have to send it via the color command.

mhightower83, 2019-10-03:
That is not the behavior I got. You seem to have a different screen than I. I only have one toggle. I am going to do a "Backup Configuration" and "Reset Configuration". Then reapplied the following to be sure I know where I started from.

I applied these to "6.6.0(release-sonoff)".
```
{"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,38,37,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}
SetOption68 1
```
I still only see 1 Toggle on my main menu. And a brightness and CT slider.
BTW: I would avoid color 255,255,255,255,255 that turns on all color LEDs and 1 set of White LEDs which will go over the 9 Watt rating on the bulb. Set to non-zero just the 1st three for a color blend or just set the last two for a White CT and brightness. It appears that if you set the 1st three the 4th and 5th are set to zero. 

Results starting after a `color 0,0,0,0,0`
* `channel5 50` gets me medium brightness, Warm White
* `channle4 100` changes to Cold White
* `channel5 100` changes  Cold White to full brightness
* `channel4 0` changes Cold White to Warm White full brightness
* `channel1 50` turns on red and zeros channels 4 and 5
* `channel2 50` turns on green, red stays on
* `channel3 50` turns on blue, red and green stay on

I only have one Toggle button and that appears to turn on an off all channels at once.

Does anyone know of an option that allows for using one channel for White Brightness and a separate channel to adjust White Color Temperature w/o affecting brightness?

Photo64, 2019-10-03:
That's weird because I definitely got 5 toggles. I switched back to Setoption68 0 and see the one toggle now again. I am also using this within Home Assistant and everything appears to be working correctly using this sample template -> https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-‐-Manual-Config-(Legacy)#rgbw-lights
I am also using 6.6.0.7 release

Resterday, 2019-10-03:
@mhightower83 thanks for referencing the other issue.

Here is what I have

`{"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,38,37,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}`

`SetOption68 1
SetOption37 0
SetOption17 1
SetOption59 1`

Home Assistant Config
`## FEIT RGBW Bulbs from https://github.com/digiblur/Sonoff-Tasmota/wiki/Smart-Bulbs  ##
  - platform: mqtt
    name: "feit_1"
    command_topic: "cmnd/feit_1/POWER"
    state_topic: "tele/feit_1/STATE"
    state_value_template: "{{value_json.POWER}}"
    availability_topic: "tele/feit_1/LWT"
    brightness_command_topic: "cmnd/feit_1/Dimmer"
    brightness_state_topic: "tele/feit_1/STATE"
    brightness_scale: 100
    on_command_type: "brightness"
    brightness_value_template: "{{value_json.Dimmer}}"
    color_temp_command_topic: "cmnd/feit_1/CT"
    color_temp_state_topic: "tele/feit_1/STATE"
    color_temp_value_template: "{{value_json.CT}}"
    rgb_command_topic: "cmnd/feit_1/Color2"
    rgb_state_topic: "tele/feit_1/STATE"
    rgb_value_template: "{{value_json.Color.split(',')[0:3]|join(',')}}"
    effect_command_topic: "cmnd/feit_1/Scheme"
    effect_state_topic: "tele/feit_1/STATE"
    effect_value_template: "{{value_json.Scheme}}"
    effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false`

I can confirm that I am now able to change all colors, initiate effects, and begin to slide the toggle towards 'cool white', but as noted by @mhightower83 as I get about halfway to cool white it starts dimming and the bulb goes dark when about 75% of the way to 'cool white'

mhightower83, 2019-10-04:
Until better settings are available I am using this:
```
template  {"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,37,29,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}
SetOption68 1
```
This creates 4 channels, RGBW, and assigns GPIO5(Warm/Cold) to inverted relay1.
Toggle1 is Warm/Cold, On is Warm
Toggle2 is the rest of the RGBW. 
Slider works; however, the range has an early dropout.

A side note, the white brightness range seems to have shrunk on me. In channel terms, the lowest setting before turn off for channel4 is ~40%. It seemed to be lower when 1st turned on.

Since the Warm/Cold PWM duty cycle is used to alternately select between Warm or Cold LEDs, I set GPIO5 to relay1 so I could select one or the other with Toggle1.

mhightower83, 2019-10-04:
@Photo64 Sorry, it looks like I was misreading `SetOption68`. It doesn't work the way I was thinking. If I am reading it correctly now, it would create multiple toggle buttons as you observed. It also is not implemented in the release I am running. It was added at 6.6.0.3.

s-hadinger, 2019-10-04:
This bahevior for White color temp is similar to Xiaomi Philips bulb from Module 48. Can you try seperately module 48 to only CW and WW GPIO (momentarily dropping RGB) to check it works. If so I can add a SetOption to toggle the cold/Warm mode since we're seeing more bulbs like this.

DSuds, 2019-10-04:
@s-hadinger can you give more instructions on what you want to see?
Using module 48 and just the PWM GPIOs I can't get any output.

We need a model for transmitting color BBGGRRDDCT instead of BBGGRRWWCW.

Diramu, 2019-10-04:
As suggested above, probably just an option that treat the excepted (WW,WC) channels as (DD (white dimmer),CT (color temp)) is all that is necessary. I think it should fix both issues.

Aside, will HA integration need changes to control this new method?

s-hadinger, 2019-10-05:
@DSuds Can you please test with the following template based on Module 48:

`{"NAME":"Test_CWWW","GPIO":[0,0,0,0,37,38,0,0,0,0,0,0,0],"FLAG":0,"BASE":48}`

If I read well what's above, GPIO4 controls white brightness and GPIO5 the white temperature color.

This template disables RGB all together and will not solve your issue. It's just to confirm that White control works correctly before doing the changes in the code.

Diramu, 2019-10-05:
I've tried this template modified for my bulb:
{"NAME":"Arlec WW/CW","GPIO":[0,0,0,0,0,37,0,0,0,38,0,0,0],"FLAG":0,"BASE":48}

I would says it works from  the sliders.  I have the bulb open.  You can see the transition between leds. 
I'll try integration to HA later.

s-hadinger, 2019-10-05:
Ok. Thanks for testing. I will see how to integrate in the code. Stay tuned

cwhits, 2019-10-05:
> @DSuds Can you please test with the following template based on Module 48:
> 
> `{"NAME":"Test_CWWW","GPIO":[0,0,0,0,37,38,0,0,0,0,0,0,0],"FLAG":0,"BASE":48}`

Using this template I'm unable to turn the light on/off using the toggle

cwhits, 2019-10-05:
the closest I've been able to get is with this template where the toggle button works, but sliding from warm to cold does make it bluer, but when you get all the way over to cold it dims to off:
```
{"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,38,37,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}
```

Diramu, 2019-10-06:
Integration of my bulb (just the two channels dimmer and temp) worked fine into HA using digiblur 2 channel mqtt code : https://github.com/digiblur/Sonoff-Tasmota/wiki/Smart-Bulbs

@cwhits I had similar behavour with my bulb (just cold and warm light leds). The solution was "base:48" and having the pwm's in the correct order.  My final solution is now up:
https://github.com/digiblur/Sonoff-Tasmota/wiki/Smart-Bulbs

s-hadinger, 2019-10-06:
Waiting for Theo to merge, you should be able to use Module 18 (Generic) and `SetOption73 1` to control ColorTemp through PWM2. PWM1 is still White Brightness.

```
{"NAME":"BPA800/../2(P)","GPIO":[255,255,255,255,38,37,255,255,141,142,140,255,255],"FLAG":0,"BASE":18}
SetOption73 1
```

closed 2019-10-06 by arendst

s-hadinger, 2019-10-06:
Everything should work correctly using Module 48, and the latest build from http://thehackbox.org/tasmota/

cwhits, 2019-10-06:
@s-hadinger can you send me what template you're using?
using this, toggle 1 will turn the light on and off, and toggle 2 will switch between cool/warm.
```
{"NAME":" BPA800","GPIO":[0,0,0,0,21,22,0,0,0,0,0,0,0],"FLAG":0,"BASE":48}
```
setting those as PWM37 and PWM38 respectively ends up with me being unable to turn the light on.
```
{"NAME":" BPA800","GPIO":[0,0,0,0,37,38,0,0,0,0,0,0,0],"FLAG":0,"BASE":48}
```
this is close in that the cool/warm and dark/bright sliders work, but the toggle doesn't actually turn the light on or off:
```
{"NAME":" BPA800","GPIO":[255,255,255,255,37,38,255,255,141,142,140,255,255],"FLAG":0,"BASE":48}
```

cwhits, 2019-10-06:
`Setoption37 0` is the correct RGBW order for Home Assistant, however when choosing an "Effect" the C/W LED does not turn off, so the bulb will be white with the color effect happening at the same time.

I'm still running this template,
```json
{"NAME":" BPA800","GPIO":[255,255,255,255,37,38,255,255,141,142,140,255,255],"FLAG":0,"BASE":48}```
with this version of Tasmota:
```
Cold/Warm and Dark/Bright sliders work correctly in Tasmota WebUI and from Home Assistant.

BUT, the power toggle still doesn't work. Any ideas for figuring that one out?

|Program Version | 6.6.0.15(24142b8-sonoff)|
|--|--|
|Build Date & Time | 2019-10-06T21:03:12|
|Core/SDK Version | 2_4_2/2.2.1(cfd48f3)|

DSuds, 2019-10-07:
@cwhits - you seem to have 37 & 38 swapped.

cwhits, 2019-10-07:
@DSuds they're in the correct order. if I swap 37 and 38 then the controls are backward: cool/warm controls the dimmer, and dark/bright controls the color temp. 

I'm thinking the power might be wrong? I don't know enough about Tasmota to figure it out.

Photo64, 2019-10-08:
I downloaded the Dev build with the 2.3 core and added this template -> {"NAME":" BPA800","GPIO":[255,255,255,255,37,38,255,255,141,142,140,255,255],"FLAG":0,"BASE":48}. The CW & WW slider is working correctly however I noticed that when the bulb is in either CW or WW the dimmer slider shuts the bulb off at about half-way down but when the bulb is on a color the dimmer goes all the way to the end before shutting off. Is that expected or what you are seeing also?

s-hadinger, 2019-10-08:
This looks similar to a bug reported weeks ago but I couldn't reproduce. 

It is definitely a bug. Just to be sure can you try with 2.5.2 if the problem is still there.

cwhits, 2019-10-08:
dimming works correctly for me:
Program Version 6.6.0.15(7f2b6c7-sonoff)
Build Date & Time 2019-10-07T22:03:10
Core/SDK Version 2_4_2/2.2.1(cfd48f3)

Photo64, 2019-10-08:
Still seeing the same results with the dimming after moving to 2.5.2, half way down it goes out when on CW or WW. When on a color it works as expected.

Program Version | 6.6.0.16(6d31608-sonoff)
Build Date & Time | 2019-10-08T14:06:29
Core/SDK Version | 2_5_2/2.2.2-dev(c0eb301)

2019-10-08T14:06:29
2_5_2/2.2.2-dev(c0eb301)

s-hadinger, 2019-10-08:
Btw, I forgot to mention that [White-blend mode](https://github.com/arendst/Sonoff-Tasmota/wiki/White-Blend-Mode) does not work with Module 48 (yet).

Photo64, 2019-10-08:
@s-hadinger Should I be entering a template with module 18?   I just tried changing to Generic 18 but those commands aren't working for me

s-hadinger, 2019-10-08:
Hmmm. Can you also try

Color 00000080FF

Color 00000040FF

They should give similar results with cold white.

Photo64, 2019-10-08:
Color 00000080FF   - Light (CW) on

Color 00000040FF - Light off

s-hadinger, 2019-10-08:
Thanks for testing. I may have an explanation. If the bulb is doing Gamma correction (and I have no clue how it would do that from a pwm output) that could explain this behavior. 

Can you get back to the working template (the one with module 48) and try `LedTable 0`

This disables gamma correction. And try again with the slider.

cwhits, 2019-10-08:
@s-hadinger since this all appears to be currently working for me. is there anything you'd like me to test?

Photo64, 2019-10-08:
Yes the dimmer appears to be working correctly now after LedTable 0

s-hadinger, 2019-10-08:
@Photi64 good news for now. I still need to know if it's a bulb thing or a bug. One thing is that White blended mode does not work yet. You didn't use `rgbtable` right?

@cwhits what is your setting for `LedTable `? Just type the command with no parameter. Is it the exact same bulb as @Photo64?

Photo64, 2019-10-08:
I did not use rgbtable.

cwhits, 2019-10-09:
@s-hadinger  {"LedTable":"OFF"}

Bulb that's listed in the title; from Costco. 🤷

DSuds, 2019-10-09:
@cwhits - there are multiple bulbs with that part number.  We're working on the newer ones where the FCC ID ends in G2

For me Version 6.6.0(release-sonoff)-2_3_0 is working pretty well.  When set to CT 500 the light comes on at 0000000011 and stays on until it drops below 000000000d.  At CT 175 and below the light is out.

mhightower83, 2019-10-09:
> @cwhits - there are multiple bulbs with that part number. We're working on the newer ones where the FCC ID ends in G2

@DSuds, I think you're referring to FCC ID SYW-A19RGBWAGT2 which from my sampling of bulbs carries a model number on the bulb of BPA800/RGBW/AG/2(P). The older bulbs from Costco do not have the "(2)" at the end and have an FCC ID of FCC ID SYW-A19RGBWAGT1. The print is small and it is easy to mistake what you are looking at. 

It should also be pointed out that the default for [`LedTable`](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#ledtable) changed on release 6.5.0.9 and above to default to "On". So I suspect that some of us that started with the older firmware such as 6.5.0 and upgraded saw one result. And those of us who started with newer or did a config reset saw a different result. For me, `LedTable 0` made things work as expected.

Currently, I am running `Project sonoff Sonoff Version 6.6.0.15(7f2b6c7-sonoff)-2_4_2`

Just did these console commands to start over.
```
reset 4
```
Wait for restart
```
template {"NAME":"BPA800/../2(P)","GPIO":[0,0,0,0,37,38,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}
module 0
```
Wait for restart
```
LedTable 0
```
At this point, all 5 channels appear to be tracking the Dim/Bright slider. 
After touching the Warm/Cold slider, the RGB LEDs are no longer tracking the Dim/Bright slider. 
And both sliders have the expected result. 
For me, 9% brightness appears to be the turn-on point for CWWW.

Additional observations:

Setting Channel1 through Channel3 resets Channel4 & 5 - IMO this is good. However, setting channel4 and channel5 does not reset channels1 through 3.

Regarding RGBW option: The bulb is rated at 9 Watts. The Cold/Warm LED blend draws 9 Watts at full brightness. The RGB LEDs draw 3 Watts at full brightness. The two combined is 12 Watts, exceeding the power rating. The cone base gets pretty hot at 9 Watts. I don't have the original Feit software installed to compare against/reverify; however, I don't think RBG and White combined was an option. I think it was RGB or Warm/Cold. Can anyone confirm this?

s-hadinger, 2019-10-09:
@mhightower83 you nailed it. We know now for sure that this bulb has hardware Gamma correction. Ledtable should be off. 

Using CT command should automatically turn RGB off. I understand you used Channel command for 4 and 5. 

Piece of warning, using the Color command gives you unrestricted access to channel values. I.e. `Color FFFFFFFFFF` will turn full power to all channels and might well fry your bulb. Don't do it!

Resterday, 2019-10-09:
Even though its closed, it seems like this discussion is still active. Since I haven't been able to find photos online and i bricked my device, I decided to open it up to take some pictures and share. Hope these help! 

Managed to get the LED board off without any damage using a paint can opener down the hole in the middle. I thought it might crack while I was pulling it out, but it turns out its a solid piece of aluminum and not a PCB. here are some pictures:

[image]
[image]
[image]
[image]
[image]

cwhits, 2019-10-09:
> @DSuds, I think you're referring to FCC ID SYW-A19RGBWAGT2 which from my sampling of bulbs carries a model number on the bulb of BPA800/RGBW/AG/2(P). The older bulbs from Costco do not have the [...]

The FCC ID on my bulb is `SYW-A19RGBWAGT1`

mhightower83, 2019-10-09:
@cwhits then you want this [template](https://blakadder.github.io/templates/feit_rgb_a19.html).

>Piece of warning, using the Color command gives you unrestricted access to channel values. I.e. Color FFFFFFFFFF will turn full power to all channels and might well fry your bulb. Don't do it!

@cwhits this goes double for your bulb. With a fresh config reset, template ..., module 0, etc. The bulb starts off with {"Color":"1A1A1AFFFF"}. This is a bad default setting. Channels 4 and 5 at FF, places the Warm and Cold LEDs at full brightness. At full brightness, you have a 9 Watt string of Warm LEDs and a 9 Watt string of Cold LEDs on at the same time. That would be 18 Watts, I measure 16 Watt so the power supply is already being strained. A change in Warm/Cold slider will correct this or a `CT` command, eg. `CT 500`.

To avoid this, after flashing a new bulb or a config reset. Make sure if reports power `Off`. Then apply the template, etc. Then use a CT command to turn the bulb on. This appears to put things properly. 

I think this latter approach should also be used with the BPA800/RGBW/AG/2(P) (FCC ID SYW-A19RGBWAGT2) as well. I observed that bulb starting off at 12-13 Watts after the `module 0` command.

s-hadinger, 2019-10-09:
@mhightower83 Thanks for the dive deep.

It's a surprise to me that the bulbs starts with {"Color":"1A1A1AFFFF"}. I will try to find out why and potentially fix it.

cwhits, 2019-10-09:
> @cwhits then you want this [template](https://blakadder.github.io/templates/feit_rgb_a19.html).

That's the template I started with, but it didn't work with my bulb. Only template 48 worked correctly (and everything does work correctly with that template now)

mhightower83, 2019-10-10:
Hmm, I guess people will have to take the approach if one template doesn't work try the other.

@cwhits what is the date code on your bulb. My ...GT2 bulb is 228-19-24, which I take to be the 24th week of the year 2019. My ...GT1 is 228-19-21

cwhits, 2019-10-10:
mine is also 228-19-24

s-hadinger, 2019-10-10:
@mhightower83 #6162 fixes the initial value of Color after flashing the firmware or doing a `Reset 5` or `Reset 6`.

Default value is now `{"Color":"1A1A1A0000"}`, playing soft on the bulb power supply.

Thanks for reporting the issue.

raptordemon, 2019-12-11:
so i have yet another version of this bulb. its a OM60/RGBW/15K/AG sold at lowes in Dec 2019.
FCC-ID SYW-A19RGBW15KA
it seems to be different but i have managed to make it work!
[image]
[image]
[image]
[image]

Esp is a TYWE2L with seems to be a tuya module.
RGB driver is a SM726EC chip
Chip on the white top plate seems to manage the WW CW BP5778 J8D22VE SB34J could not find on google.

and here is an approx pinout of the sm726EC and white pin header on the board:
[image]

some things to note is cold and warm seem to be reversed, and i have yet to get color mapping working correctly but i can control RGB and the WW and CW.
hope this helps someone

mhightower83, 2019-12-12:
> ... could not find on google.

@raptordemon, I sometimes have to resort to using a Chinese search engine and rely on google translate to help me get through. I found this datasheet for the [BP5778DJ](http://bbs.elecfans.com/jishu_1309896_1_1.html). Excerpt from Google translate:
> ...  BP5778DJ is a dual-channel dimmable LED linear constant current driver chip that supports PWM and analog dimming signal input. It is mainly used to drive various types of dimming light sources and lamps for city power input. The BP5778DJ based on linear constant current technology can eliminate electrolytic capacitors and magnetic components, help LED drivers achieve small size, long life, and meet EMI standards.
...

cyberplant, 2020-04-06:
Hello!

I'm here to comment that the bulb Nexxt AHIBUSO4U1 also works fine with the template:
```
template {"NAME":"BPA800/../2(P)","GPIO":[0,0,0,0,37,38,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}
```
linked here in the comments.

rezadast2, 2020-07-13:
Is it to be expected that the white brightness level greater than 50 doesnt actually change the brightness with gamma correction off, LEDTABLE 0? With LEDTABLE 1, my brightness was opposite, 100 > 50 worked, but anything less than 50 it turned off. Now with LEDTABLE 1, 100 > 50 doesnt change the brightness at all and stays at 100% brightness. 50 > 0 changes the brightness.

However, when I am using the RGB colors, the brightness scale seems to be fully operational from 100 > 0.

[1 more reply]

PR #6581 (2019-10-06): Fix better handling of PWM White Temperature mode for Module 48 (#6534)
## Description:

Classical bulb have separate channels for WarmWhite and ColdWhite. New bulbs (like Xiaomi Philips) use 1 PWM for Brightness and 1 PWM for ColorTemp.
This PR allows to have the same behavior with RGBCW bulbs.

**Related issue (if applicable):** fixes #6534 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [... cut]
PR #6586 (2019-10-07): Fix #6534 Power off not working
## Description:

Fix Power off not working for FEIT bulbs.

**Related issue (if applicable):** fixes #6534 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-licen [... cut]
commit 09eb3b98 (2019-10-07): Merge pull request #6586 from s-hadinger/fix_5434_2

Fix #6534 Power off not working
commit f6416aa2 (2019-10-07): Fix #6534 Power off not working