arendst/Tasmota #6535: Shutter Code Stat Message and Rules

flashtel, 2019-10-01:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Dual_R2
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Sonoff-Tasmota 6.6.0.14 by Theo Arends

21:53:48 CMD: shutterclose1
21:53:48 SHT: Start shutter in direction -1
21:53:48 MQT: stat/myplace/house/master_bed1/RESULT = {"POWER2":"ON"}
21:53:48 MQT: stat/myplace/house/master_bed1/POWER2 = ON
21:53:48 MQT: stat/myplace/house/master_bed1/RESULT = {"ShutterClose2":0}

Sonoff-Tasmota 6.6.0.14 stb-1.11 by Theo Arends

21:53:24 CMD: shutterclose
21:53:24 Start shutter in direction -1
21:53:24 MQT: stat/myplace/house/master_bed2/RESULT = {"POWER2":"ON"}
21:53:24 MQT: stat/myplace/house/master_bed2/POWER2 = ON
21:53:24 MQT: stat/myplace/house/master_bed2/RESULT = {"shutterclose1":0}

**(Please, remember to close the issue when the problem has been addressed)**

flashtel, 2019-10-01:
This is the rules issue

Sonoff-Tasmota 6.6.0.14 by Theo Arends

21:59:01 CMD: rule1
21:59:01 MQT: stat/21cameron/house/master_bed1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":435,"Rules":"on SHUTTER-1#position==100 DO Publish /status/sonoff/level {"%value%"} ENDON"}
21:59:09 CMD: shutteropen1
21:59:09 SHT: Start shutter in direction 1
21:59:09 MQT: stat/21cameron/house/master_bed1/RESULT = {"POWER1":"ON"}
21:59:09 MQT: stat/21cameron/house/master_bed1/POWER1 = ON
21:59:09 MQT: stat/21cameron/house/master_bed1/RESULT = {"ShutterOpen1":100}
21:59:10 SHT: Shutter 0: Real Pos: 1500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.8  [s]
21:59:11 SHT: Shutter 0: Real Pos: 3500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.8  [s]
21:59:12 SHT: Shutter 0: Real Pos: 5500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.7  [s]
21:59:13 SHT: Shutter 0: Real Pos: 7500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.7  [s]
21:59:14 SHT: Shutter 0: Real Pos: 9500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.7  [s]
21:59:15 SHT: Shutter 0: Real Pos: 11500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.7  [s]
21:59:16 SHT: Shutter 0: Real Pos: 13500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.7  [s]
21:59:17 SHT: Shutter 0: Real Pos: 15500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.7  [s]
21:59:18 SHT: Shutter 0: Real Pos: 17500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 8.8  [s]
21:59:19 SHT: Shutter 0: Real Pos: 19500, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 9.8  [s]
21:59:19 MQT: stat/21cameron/house/master_bed1/SHUTTER1 = 100
21:59:19 MQT: stat/21cameron/house/master_bed1/RESULT = {"POWER1":"OFF"}
21:59:19 MQT: stat/21cameron/house/master_bed1/POWER1 = OFF
21:59:19 MQT: tele/21cameron/house/master_bed1/RESULT = {"Shutter1":{"Position":100,"direction":0}}

Sonoff-Tasmota 6.6.0.14 stb-1.11 by Theo Arends

21:58:44 CMD: rule1
21:58:44 MQT: stat/21cameron/house/master_bed2/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":435,"Rules":"on SHUTTER-1#position==100 DO Publish /status/sonoff/level {"%value%"} ENDON"}
21:58:52 CMD: shutteropen1
21:58:52 Start shutter in direction 1
21:58:52 MQT: stat/21cameron/house/master_bed2/RESULT = {"POWER1":"ON"}
21:58:52 MQT: stat/21cameron/house/master_bed2/POWER1 = ON
21:58:52 MQT: stat/21cameron/house/master_bed2/RESULT = {"shutteropen1":100}
21:58:53 Shutter 0: Real Pos: 1700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.9  [s]
21:58:54 Shutter 0: Real Pos: 3700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 1.9  [s]
21:58:55 Shutter 0: Real Pos: 5700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 2.9  [s]
21:58:56 Shutter 0: Real Pos: 7700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 3.9  [s]
21:58:57 Shutter 0: Real Pos: 9700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 4.8  [s]
21:58:58 Shutter 0: Real Pos: 11700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 5.8  [s]
21:58:59 Shutter 0: Real Pos: 13700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 6.8  [s]
21:59:00 Shutter 0: Real Pos: 15700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 7.8  [s]
21:59:01 Shutter 0: Real Pos: 17700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 8.9  [s]
21:59:02 Shutter 0: Real Pos: 19700, Target 20000, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 9.9  [s]
21:59:02 MQT: stat/21cameron/house/master_bed2/SHUTTER1 = 100
21:59:02 MQT: stat/21cameron/house/master_bed2/RESULT = {"POWER1":"OFF"}
21:59:02 MQT: stat/21cameron/house/master_bed2/POWER1 = OFF
21:59:02 MQT: tele/21cameron/house/master_bed2/RESULT = {"SHUTTER-1":{"position":100, "direction":0}}
21:59:02 RUL: SHUTTER-1#POSITION==100 performs "Publish /status/sonoff/level {"100"}"
21:59:02 MQT: /status/sonoff/level = {"100"}

ascillato, 2019-10-01:
Please, complete the template. What is your issue?

flashtel, 2019-10-01:
The issue with the stat message from the merged code is incorrect.

21:53:48 MQT: stat/myplace/house/master_bed1/RESULT = {"ShutterClose2":0}
vs
21:53:24 MQT: stat/myplace/house/master_bed2/RESULT = {"shutterclose1":0}

The issue with Rule on the merged code is it doesn't trigger.

meingraham, 2019-10-01:
`6.6.0.14 stb-1.11` - this is the Stefan Bode fork. Please address your issue there.

meingraham, 2019-10-01:
`6.6.0.14 stb-1.11` 

is this in the Tasmota branch being enhanced with the shutter feature or in Stefan's fork?

arendst, 2019-10-01:
Regarding rules, the message has changed so you will need to update your rule to trigger on the correct message syntax.

Original message:
```
21:59:02 MQT: tele/myplace/house/master_bed2/RESULT = {"SHUTTER-1":{"position":100, "direction":0}}
```
New message:
```
21:59:19 MQT: tele/myplace/house/master_bed1/RESULT = {"Shutter1":{"Position":100,"direction":0}}
```
So just update your rule to handle the new message syntax.

flashtel, 2019-10-01:
I can see it now and the rules are confirmed as working, thanks.

23:28:45 MQT: tele/21cameron/house/master_bed1/RESULT = {"Shutter1":{"Position":100,"direction":0}}
23:28:45 RUL: SHUTTER1#POSITION==100 performs "Publish /status/sonoff/level {"100"}"
23:28:45 MQT: /status/sonoff/level = {"100"}

ascillato2, 2019-10-01:
Great!

Closing this issue as it has been fixed.

closed 2019-10-01 by ascillato2

flashtel, 2019-10-01:
The stat message is fixed? I'll recompile and test.