arendst/Tasmota #6546: Add smartWifi / minimize electrosmog

hummerdash, 2019-10-02:
**The Problem:** 
Tasmota devices' wifi is always on. 

I know that for a lot of tasks this is neccessary and simply defines a device as an IOT device.

**My feature request:** 
To have the possibility to (automatically) turn off wifi on certain conditions.

**But why?**
I use some smart powerplugs to monitor the energy consumption for the devices that are connected to them. For example I use it to send myself a notification once the washing machine has finished.
In that case there is no need to switch the plug on and off.

And there is also no need to have wifi on for the periods between the washing cycles.

**My suggestion:**
If the current that is drawn throungh the plug exceeds a certain ( adjustable ) threshold wifi is switched on and it starts publishing (mqtt) the measures so that I can monitor the power consumption. 
Once the current drops again under a threshold for a given amount of time, it turns off wifi.

This is also reducing electrosmog a lot.

Jason2866, 2019-10-02:
Tasmota is designed to use mqtt. If you switch of wifi you loose mqtt connection to broker.
This is a not wanted behaviour.

hummerdash, 2019-10-02:
As I described above,  for some usecases it is not necessary to have a connection at every time.
Once wifi is switched on again, it connects to the broker and starts sending its stuff.
Why is that (in this particular case) not a wanted behavior?

**I would rather say that it is a not wanted behavior if it continously sends data even though nothing happens... :-)** 

It is a "one way" usecase. We do not expect data/input from any other (mqtt) device. We are only sending information (if there is any)...

ascillato2, 2019-10-02:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-02 by ascillato2

joba-1, 2019-10-02:
see e.g. #4330

ascillato, 2020-06-16:
Now it is available the command `WIFI` to turn ON and OFF the wifi from the Tasmota Console. Please, use latest development version.