arendst/Tasmota #6552: Template extension

Diramu, 2019-10-03:
Just an idea to extend templates to include "setOption" commands

Optional addition fields could be added  like:
  "SetOption37":6
and
 "SetOption":FFFFFFFF
 a 32 bit hex value for the set option 0 - 31, since they are only a bit bit each.

This will complete the implementation of a device is a single string.

meingraham, 2019-10-03:
Specific setup for a particular device can be placed in the template notes. There are many other settings and rules besides SetOptions that are needed sometimes.

Mike

P.S. I'm unclear what you mean by 
> "SetOption":FFFFFFFF
a 32 bit hex value for the set option 0 - 31, since they are only a bit bit each.

andrethomas, 2019-10-03:
I would keep setoption out of the template.

The user who created a template may have used other "setoptions" that are not related to the specific device thereby creating unexpected behavior for a new user who uses that template.

jziolkowski, 2019-10-03:
Yes, plus the setoptions are spread across three registers, so either include them all or none.

arendst, 2019-10-03:
Initially I thought it would make sense but indeed many setoptions could/would be overwritten making use uncontrollable.

jziolkowski, 2019-10-03:
Agreed. Templates are a way to describe the hardware config of the device. SetOptions describe part of functional part of the device. While in certain scenarios these areas overlap, suggesting that a device _requires_ all of the setoptions to be set in a certain way will lead to a lot of confusion.

ascillato2, 2019-10-03:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-03 by ascillato2

meingraham, 2019-10-03:
> plus the setoptions are spread across **_three_** registers

This is the answer to my question. 32 bits was not enough to specify every SetOption... I was just asking for clarification. Also, some SetOptions are not a simple binary setting. Those would require the bitmask to specify it's use plus the value of the SetOption for those that aren't binary.