arendst/Tasmota #6560: Configuration lost on upgrade

assachs, 2019-10-04:
Hello,
while upgrading from 6.6.0.11 to 6.6.0.15 my device lost the configuration. (Wifi, Template, MQTT,...)

Is this expected behaviour?
I updated 2 devices (Obi 2 Socket). Same problem on both. 

Firmware: http://thehackbox.org/tasmota/sonoff.bin
Update: OTA

Commands (after resetting the configuration):

12:01:11 CMD: Backlog Template; Module; GPIO`
12:01:11 MQT: stat/obi1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
12:01:11 MQT: stat/obi1/RESULT = {"Module":{"61":"OBI Socket 2"}}
12:01:11 MQT: stat/obi1/RESULT = {"GPIO":"Not supported"}

12:02:24 CMD: Status 0
12:02:24 MQT: stat/obi1/STATUS = {"Status":{"Module":61,"FriendlyName":["Obi1"],"Topic":"obi1","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:02:24 MQT: stat/obi1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:21","StartupUTC":"2019-10-04T10:46:03","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":12,"SaveAddress":"F8000"}}
12:02:24 MQT: stat/obi1/STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(22dbe59-sonoff)","BuildDateTime":"2019-10-04T11:03:10","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
12:02:24 MQT: stat/obi1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005AFF000000170700","00100000"]}}
12:02:24 MQT: stat/obi1/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":472,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashMode":3,"Features":["00000809","8FDAE397","243683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:02:24 MQT: stat/obi1/STATUS5 = {"StatusNET":{"Hostname":"obi1-4259","IPAddress":"192.168.10.151","Gateway":"192.168.10.6","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.200","Mac":"B4:E6:2D:33:B0:A3","Webserver":2,"WifiConfig":4}}
12:02:24 MQT: stat/obi1/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"obi1","MqttClient":"obi1","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:02:24 MQT: stat/obi1/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 04 11:02:24 2019","Local":"Fri Oct 04 12:02:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:53","Sunset":"18:24"}}
12:02:24 MQT: stat/obi1/STATUS10 = {"StatusSNS":{"Time":"2019-10-04T12:02:24"}}
12:02:24 MQT: stat/obi1/STATUS11 = {"StatusSTS":{"Time":"2019-10-04T12:02:24","Uptime":"0T00:16:21","UptimeSec":981,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:06"}}}

I have more devices to upgrade. I will wait for your answer, so i can log more details if you need them.

Thanks
Andi

arendst, 2019-10-04:
As the CRC calculation was changed between 6.6.0.7 and 6.6.0.11 you might expect some configuration issues although I haven't seen them before.

In that case I suggest you do:
- Open GUI and make a configuration backup
- Execute command ``upload 1`` from the GUI console
- Perform a configuration restore from the GUI if possible

assachs, 2019-10-04:
I updated more devices now.

Device 1 + 2 (Obi Socket 2): 
6.6.0.11 to 6.6.0.15(22dbe59-sonoff)-2_4_2
Configuration Lost

Device 3,4,5,6 (Obi Socket 2 and Tuya):
6.6.0.11 to  6.6.0.15(6997626-sonoff)-2_4_2
Configuration not lost.

I am not sure if all 6.6.0.11 had the same build.

On previous updates i never had this problem.

andrethomas, 2019-10-04:
I tried reproducing by flashing 2e9f06f to a blank device, resetting it to defaults and performing basic configuration. Then upgraded to 6997626 OTA and it loaded fine.

```
00:00:00 CFG: Loaded from flash at 3FB, Count 9
00:00:00 Project sonoff Sonoff Version 6.6.0.15(6997626-sonoff)-2_4_2
```

Not sure what could have caused it but if it was one of the changes made between .11 and .12 then this test should have revealed it.

assachs, 2019-10-04:
I had problems with .11 to .15 (22dbe59).
I had no problems with .11 to .15 (6997626) .

andrethomas, 2019-10-04:
Thx, let me test that.

andrethomas, 2019-10-04:
Nope, sorry.

6.6.0.11(2e9f06f-sonoff)-2_4_2 -> 6.6.0.15(22dbe59-sonoff)-2_4_2

No problem detected.

assachs, 2019-10-04:
I close the issue since i cannot reproduce it with the newest 6.6.0.15. Thank's for testing.

closed 2019-10-04 by assachs

andrethomas, 2019-10-04:
No prob - doesn't mean there isn't something causing it somewhere... just a pity it does not seem obvious when observing the code changes and of course, a manual test passed, which in itself is also not 100% conclusive.

Thanks for reporting though - definitely something to keep an eye on and you've successfully reminded me that I am somehow not archiving the actual build bins at the moment which is appreciated.