arendst/Tasmota #6561: Cannot reset Energy Today/Total

rt400, 2019-10-04:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
I try to reset Energy Total/Today unsuccessfully, after I run the EnergyReset3 0 command, I get the same values but do not get 0. If I run the EnergyReset1 0 command then TODAY gets 0 and after a few seconds suddenly it gets a value that was previously.
Also run Reset5 , still got the same values , flash again with "Erase All Sketch" got the same results.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [X] Tasmota binary firmware version number used: 6.6.0.15 2.6/2.5/2.4.2/2.3 Core
  - [X] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Arduino,PlatformIO
- [X] Flashing tools used: NodeMcU PyFlasher 4.0
- [X] Provide the output of command: 
``
15:07:26 MQT: stat/3PhazeMonitor/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
  ```
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [X] Provide the output of this command: ``Status 0``:
  ```
  15:09:25 MQT: stat/3PhazeMonitor/STATUS = {"Status":{"Module":18,"FriendlyName":["3 Phaze","Sonoff2","Sonoff3","Sonoff4"],"Topic":"3PhazeMonitor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"MejahezHome","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T02:08:27","StartupUTC":"2019-10-04T10:00:58","Sleep":0,"CfgHolder":4617,"BootCount":2,"SaveCount":1119,"SaveAddress":"3F8000"}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS2 = {"StatusFWR":{"Version":"6.6.0.14(sonoff)","BuildDateTime":"2019-10-02T16:01:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mejahez",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["0000C029","280500000100060000005A64000000000000","80000000"]}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS4 = {"StatusMEM":{"ProgramSize":533,"Free":2536,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FCA6387","043783A1","22BE17CD","01001BC0","00000201"],"Drivers":"1,2,3,4,5,6,8,9,10,12,16,17,18,19,20,21,22,24,28","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,16,17,18,20,22,26,34"}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS5 = {"StatusNET":{"Hostname":"3PhazeMonitor","IPAddress":"192.168.0.56","Gateway":"192.168.1.253","Subnetmask":"255.255.254.0","DNSServer":"192.168.1.254","Mac":"3C:71:BF:29:2B:DE","Webserver":2,"WifiConfig":4}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.254","MqttPort":1883,"MqttClientMask":"3PhazeMonitor","MqttClient":"3PhazeMonitor","MqttUser":"homeassistant","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 04 12:09:25 2019","Local":"Fri Oct 04 15:09:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"08:53","Sunset":"20:24"}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS10 = {"StatusSNS":{"Time":"2019-10-04T15:09:25","ANALOG":{"A0":184},"ENERGY":{"TotalStartTime":"2019-10-04T15:09:23","Total":308.467,"Yesterday":0.000,"Today":238.737,"Power":[64,2731,12],"ApparentPower":[112,2773,102],"ReactivePower":[92,480,101],"Factor":[0.57,0.98,0.12],"Frequency":[50,50,50],"Voltage":[229,230,230],"Current":[0.490,12.051,0.444]},"AM2301":{"Temperature":30.6,"Humidity":36.7},"TSL2561":{"Illuminance":34.020},"TempUnit":"C"}}
15:09:25 MQT: stat/3PhazeMonitor/STATUS11 = {"StatusSTS":{"Time":"2019-10-04T15:09:25","Uptime":"0T02:08:27","UptimeSec":7707,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":1,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:07"}}}
  
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
[image]

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pablozg, 2019-10-04:
If you are using a external meter, tasmota auto update the energy total to be in sincro with the hardware counter. If you want reset it you must reset the meter counter.

But if @arendst is agree, we can add a new setoption command to enable or disable the autoupdate function.

rt400, 2019-10-05:
> If you are using an external meter, tasmota auto update the energy total to be in sincro with the hardware counter. If you want reset it you must reset the meter counter.
> 
> But if @arendst is agree, we can add a new "setoption" command to enable or disable the autoupdate function.

i used with PZEM04-T 3.0v , i think that he doesn't have a counter and don't have a reset button on board like version 1.0/2.0 .
 the "Total" isn't belong to tasmota code ???

pablozg, 2019-10-05:
Hi, I have take a look to the pzem code and yes, it have a energy total counter, and is used to update the tasmota counter.

In a previous commit I made, I introduced an autoupdate function to get tasmota counter in sync with the hardware counter. 

I have made a new PR to can choose between use the hardware counter or the tasmota counter, using the setoption72 command, but I'm waiting to Theo approval.

ascillato2, 2019-10-06:
Closing this issue as it has been added

closed 2019-10-06 by ascillato2

rt400, 2019-10-09:
@pablozg sorry .. but still its not fixed..
i try the SetOption72 with no result.
i got measure number that not real measure
[image]
with 2 days, also try to reset again but all the time he jump to number that i dont know from where he give it.

rt400, 2019-10-09:
[image]
look what happen on every reset command

rt400, 2019-10-09:
look...maybe i have a Ghost in my PZEM...
```
22:49:17 CMD: SetOption72
22:49:17 MQT: stat/3PhazeMonitor/RESULT = {"SetOption72":"OFF"}
22:49:29 CMD: EnergyReset3 0
22:49:29 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":467.033,"Yesterday":0.000,"Today":467.033,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:49:37 CMD: EnergyReset3 0
22:49:37 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":467.034,"Yesterday":0.000,"Today":467.034,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:49:38 MQT: tele/3PhazeMonitor/STATE = {"Time":"2019-10-09T22:49:37","Uptime":"0T00:00:50","UptimeSec":50,"Heap":27,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:49:38 MQT: tele/3PhazeMonitor/SENSOR = {"Time":"2019-10-09T22:49:38","ANALOG":{"A0":238},"ENERGY":{"TotalStartTime":"2019-10-09T00:00:00","Total":467.034,"Yesterday":0.000,"Today":467.034,"Period":7,"Power":[152,493,89],"ApparentPower":[241,501,177],"ReactivePower":[187,88,153],"Factor":[0.63,0.98,0.51],"Frequency":[50,50,50],"Voltage":[237,238,237],"Current":[1.016,2.108,0.744]},"AM2301":{"Temperature":30.2,"Humidity":47.8},"TSL2561":{"Illuminance":5.406},"TempUnit":"C"}
22:49:46 CMD: SetOption72 1
22:49:46 MQT: stat/3PhazeMonitor/RESULT = {"SetOption72":"ON"}
22:49:51 CMD: EnergyReset3 0
22:49:51 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":467.035,"Yesterday":0.000,"Today":467.035,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:49:59 CMD: EnergyReset3 0
22:49:59 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":467.039,"Yesterday":0.000,"Today":467.039,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:10 MQT: tele/3PhazeMonitor/STATE = {"Time":"2019-10-09T22:50:10","Uptime":"0T00:01:23","UptimeSec":83,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:50:10 MQT: tele/3PhazeMonitor/SENSOR = {"Time":"2019-10-09T22:50:10","ANALOG":{"A0":238},"ENERGY":{"TotalStartTime":"2019-10-09T00:00:00","Total":428.313,"Yesterday":0.000,"Today":467.040,"Period":6,"Power":[153,491,90],"ApparentPower":[243,499,180],"ReactivePower":[188,87,156],"Factor":[0.63,0.98,0.50],"Frequency":[50,50,50],"Voltage":[237,237,237],"Current":[1.023,2.102,0.757]},"AM2301":{"Temperature":30.0,"Humidity":47.8},"TSL2561":{"Illuminance":5.406},"TempUnit":"C"}
22:50:14 CMD: EnergyReset1 0
22:50:14 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":428.315,"Yesterday":0.000,"Today":0.000,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:18 CMD: EnergyReset3 0
22:50:18 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.690,"Yesterday":0.000,"Today":399.690,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:28 CMD: EnergyReset3 0
22:50:28 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.692,"Yesterday":0.000,"Today":399.692,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:34 CMD: EnergyReset1 0
22:50:34 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":428.317,"Yesterday":0.000,"Today":0.000,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:36 CMD: EnergyReset3 0
22:50:36 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.694,"Yesterday":0.000,"Today":399.694,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:43 MQT: tele/3PhazeMonitor/STATE = {"Time":"2019-10-09T22:50:43","Uptime":"0T00:01:56","UptimeSec":116,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:50:43 MQT: tele/3PhazeMonitor/SENSOR = {"Time":"2019-10-09T22:50:43","ANALOG":{"A0":238},"ENERGY":{"TotalStartTime":"2019-10-09T22:50:42","Total":428.321,"Yesterday":0.000,"Today":399.696,"Period":0,"Power":[152,490,88],"ApparentPower":[240,498,174],"ReactivePower":[186,88,150],"Factor":[0.63,0.98,0.51],"Frequency":[50,50,50],"Voltage":[237,237,237],"Current":[1.012,2.101,0.733]},"AM2301":{"Temperature":30.2,"Humidity":47.8},"TSL2561":{"Illuminance":5.406},"TempUnit":"C"}
22:50:50 CMD: SetOption72 0
22:50:50 MQT: stat/3PhazeMonitor/RESULT = {"SetOption72":"OFF"}
22:50:56 CMD: EnergyReset1 0
22:50:56 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":428.322,"Yesterday":0.000,"Today":0.000,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:50:58 CMD: W
22:50:58 MQT: stat/3PhazeMonitor/RESULT = {"Command":"Unknown"}
22:51:00 CMD: EnergyReset3 0
22:51:01 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.699,"Yesterday":0.000,"Today":399.699,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:51:16 MQT: tele/3PhazeMonitor/STATE = {"Time":"2019-10-09T22:51:16","Uptime":"0T00:02:29","UptimeSec":149,"Heap":28,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}
[... cut ...]
22:51:28 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.705,"Yesterday":0.000,"Today":399.705,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:51:34 CMD: EnergyReset1 0
22:51:34 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.706,"Yesterday":0.000,"Today":0.000,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
22:51:37 CMD: EnergyReset3 0
22:51:37 MQT: stat/3PhazeMonitor/RESULT = {"EnergyReset":{"Total":399.706,"Yesterday":0.000,"Today":399.706,"Usage":[0.000,3448.858],"Export":[0.000,0.000]}}
```

pablozg, 2019-10-10:
By your logs, I see you have three pzem, the "resetEnergy" issue is fixed with the "setOption72", but @arendst did some changes to the pzem code, see issue #6585.

arovak, 2019-10-19:
I recently build a setup with 3 PZEM004T-V3 modules and also have the issue that EnergyTotal is not reset when using EnergyReset3. So I updated the device today to 6.6.0.20 but this version does not seem to fix the weird values.
Furthermore, it look to me that after every restart the EnergyTotal of the PZEM modules is added to the EnergyToday and EnergyTotal value.  

Image below is just after restart with SetOption72 = OFF and EnergyReset1/2/3 = 0
[image]

When I restart the values are multiplied.
[image]

I run this setup on a sonoff basic module. Below is the Status 0 of my device.
```
13:48:11 MQT: stat/pzem/STATUS = {"Status":{"Module":0,"FriendlyName":["pzem"],"Topic":"pzem","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:48:11 MQT: stat/pzem/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://10.4.0.5/esp8266/tasmota/6.6.0.20/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:37","StartupUTC":"2019-10-19T11:43:34","Sleep":50,"CfgHolder":4617,"BootCount":133,"SaveCount":675,"SaveAddress":"F5000"}}
13:48:11 MQT: stat/pzem/STATUS2 = {"StatusFWR":{"Version":"6.6.0.20(sonoff)","BuildDateTime":"2019-10-19T11:35:59","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
13:48:11 MQT: stat/pzem/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":3,"LogHost":"10.4.0.8","LogPort":514,"SSId":["skarenet","SecIOT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C8000100060000005A00000000000000","00000000"]}}
13:48:11 MQT: stat/pzem/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:48:11 MQT: stat/pzem/STATUS5 = {"StatusNET":{"Hostname":"pzem-2146","IPAddress":"10.4.0.180","Gateway":"10.4.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.4.0.7","Mac":"5C:CF:7F:7A:68:62","Webserver":2,"WifiConfig":4}}
13:48:11 MQT: stat/pzem/STATUS6 = {"StatusMQT":{"MqttHost":"10.4.0.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7A6862","MqttUser":"sonoff5","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:48:11 MQT: stat/pzem/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 19 11:48:11 2019","Local":"Sat Oct 19 13:48:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:16","Sunset":"18:54"}}
13:48:11 MQT: stat/pzem/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:48:11 MQT: stat/pzem/STATUS10 = {"StatusSNS":{"Time":"2019-10-19T13:48:11","ENERGY":{"TotalStartTime":"2019-10-19T13:39:53","Total":170.446,"Yesterday":0.000,"Today":170.446,"Power":[125,24,414],"ApparentPower":[133,135,569],"ReactivePower":[44,132,391],"Factor":[0.94,0.18,0.73],"Frequency":[50,50,50],"Voltage":[237,236,235],"Current":[0.560,0.571,2.422]}}}
13:48:11 MQT: stat/pzem/STATUS11 = {"StatusSTS":{"Time":"2019-10-19T13:48:11","Uptime":"0T00:04:37","UptimeSec":277,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SecIOT","BSSId":"F0:9F:C2:FD:2E:72","Channel":11,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

arendst, 2019-10-20:
I cannot reproduce but I have only one PZEM004T v2 (I think).

What happens when you connect just one PZEM?

On the other hand, what is your GPIO configuration? As far as I know the PZEM004T does not do Frequency but your display shows frequency. Pls report output of command ``gpio``.

arendst, 2019-10-20:
Ah just googled the V3 and it seems it now supports modbus protocol so must be using the PzemAc driver. That explains the frequency.

I will have to dig up my Pzem014 for testing.

arendst, 2019-10-20:
Pls try the latest commit and report back.

arovak, 2019-10-20:
First, thank you for the quick response and great software!

> What happens when you connect just one PZEM?
> 
With just 1 PZEM v3 connected value is still correct after energyreset1/2/3 0 and reboot.
 
[image]

> Pls report output of command `gpio`.

```
13:45:21 CMD: gpio
13:45:21 MQT: stat/pzem/RESULT = {"GPIO2":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
```

The template I am using on this sonoff basic is

```
13:48:34 MQT: stat/pzem/RESULT = {"NAME":"PZEM","GPIO":[17,62,255,98,255,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":1}
```

> 
> 
> Pls try the latest commit and report back.

With the new build results are the same.
After reconnectiong the 3 PZEM values it loads a weird value for EnergyToday and EnergyTotal and these are still multiplied after a reboot

[image]

After reboot
[image]

arendst, 2019-10-20:
Thx for testing. Could you pls try with just two connected.

arendst, 2019-10-20:
Ah don't bother. I think what's going on. Now I have to find a way to solve it.

arendst, 2019-10-20:
Do you compile yourself?

arendst, 2019-10-20:
And another possible fix. Pls report back.

arovak, 2019-10-20:
Yes I use Platformio to compile, and it looks like you have fixed the problem!

Energyreset1/2/3 0, setoption72 OFF and after multiple reboots I do not have the weird values anymore.

[image]

Again, thank you for the quick response and fix!

arendst, 2019-10-20:
Great! Thx for your feedback.

rt400, 2019-10-21:
@arendst is that fix add to the code ? cause didn't see this in _changelog.ino..
anyway i thinks that i have one defect module. i log it with SysLog and upload it , hope you can understand the HEX Data cause i am not 
[https://paste.ofcode.org/dG8guTQRnaLvpz6BSeuaS5](url)
thanks...

rt400, 2019-10-22:
@arendst look here https://github.com/mandulaj/PZEM-004T-v30
it's look that he made a Energy reset by sending serial command (cause in 3.0v there is no reset button)
so maybe you can add this to tasmota command so we can reset the hardware enregy..

rt400, 2019-10-27:
@arendst It still gets worse and brings crazy values
Have an idea what to do with it? You may be able to add a reset command to these modules because version 3 does not have a reset button and reset is through software.
Thanks so much for all the much appreciated help

gromar88, 2020-05-25:
Is there a possibility to split energy for each phase?