arendst/Tasmota #6563: WCWW bulb with dimmer and color pwms

Diramu, 2019-10-04:
I've tuya-converted a warm/cold white bulb https://blakadder.github.io/templates/arlec_GLD112HA.html

But main page does not control the bulb correctly because it is assumed that each PWM control either color led.
The bulb  use a 2 fet driver that uses a single pwm to multiplex between the two colors.
Here a pdf in Chinese of this driver chip, not that I can read it, but the sample circuit infers this behaviour: [http://www.bpsemi.com.cn/uploads/file/20181017105105_356.pdf](url)

It is possible to a have a set option to describe this (overload or use an invalid setoption37 when only 2 pwms are defined, or yet another setoption)

mhightower83, 2019-10-04:
This chip and method are also being used in the more recent Feit bulbs, BP800/RGBW/AG/2(P). https://github.com/arendst/Sonoff-Tasmota/issues/6534#issuecomment-537771525

s-hadinger, 2019-10-04:
The Philips Xiaomi works the same way, 2 PWM. One for brightness and one for color tone. Can you please try with Template 48?

mhightower83, 2019-10-04:
For the Feit BP800/RGBW/AG/2(P) that _only_ works if the RGB LEDs are disabled. Otherwise, it does appear to work well. Hopefully, @Diramu will have good luck with that Template, since his device is also using the same PWM Warm/Cold chip and does not have RGB LEDs with an additional controller.

@s-hadinger Is there another template base that combines 48 with an RBG controller like the SM16716?

s-hadinger, 2019-10-04:
Module 48 is the only way to trigger the Xiaomi Philips mode. I need to double check the code though. 

I'm not sure you can combine both PWM and SM16716 at the same time. Again need to double check. 

Does it work for your bulb, and are CW/WW in the right order?

s-hadinger, 2019-10-04:
I just read the other issue. I guess it's time for a new SetOption for Color Temp PWM.

mhightower83, 2019-10-04:
After a "Reset Configuration" I set this from the "Console":
```
template {"NAME":"BPA800/../2(P)","GPIO":[0,0,0,0,37,38,0,0,0,0,0,0,0],"FLAG":0,"BASE":48}
```
Using the slider controls Cold/Warm works as expected. Dark/Bright works with the light staying off between about 28 - 39%.  Dimmer command with 28 - 39 will turn light on. My initial impression is this value changes (goes up) as the bulb warms up. I have applied this to two bulbs now with similar results. It is looking like there may be a need to set a floor level for Dark.

BTW: I really did need the "Reset Configuaration" some previous unknown settings/options that I use with the older model were stopping it from working.

DSuds, 2019-10-04:
@mhightower83 Can you continue this on the Feit v2 RBG bulb thread #6534 instead of here?

closed 2019-10-05 by ascillato2