arendst/Tasmota #6565: beside sunrise and sunset solar noon is needed

salihys, 2019-10-04:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?** 
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Please add solar noon beside sunrise and sunset to be used in rules
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-10-04:
What is your use case that requires a precise solar noon? What timing do you have that is critical with +/- 2-3 minutes?

ascillato, 2019-10-04:
In timers you can substract or add minutes to sunrise and sunset and then trigger a rules or relays.

meingraham, 2019-10-04:
But, doesn't solar noon drift with respect to sunrise/sunset throughout the year?

s-hadinger, 2019-10-04:
On first approximation it should be at the same clock time everyday and dependubg only on location, right ? if you need sub-minute accuracy it's more complicated.

meingraham, 2019-10-04:
@s-hadinger yes, approximately 12:00 :wink:

s-hadinger, 2019-10-04:
Well that's a big approximation. It depends on your longitude in the time zone. 
The average of sunrise and sunset should do it.

salihys, 2019-10-04:
I want to use that to accurately turn on a light when it's time for Muslim prayer to alert a def person when it's time to pray. I hope that someone can help with this issue and thanks for your comments.

meingraham, 2019-10-04:
@salihys 

I think I'd rather depend on a HAB running locally on the LAN (no Internet required). A Raspberry Pi suffices. It will keep it's time in sync with NTP when an Internet connection is available. It will capture astronomical information then as well. The HAB server can even implement logic for whatever events you need if necessary to have more autonomy. With all the necessary information, the HAB can command the Tasmota device to trigger the light exactly when needed. Heck, you could even re-write  a Tasmota rule that gets update on the device to perform the action at the appropriate time should the HAB be unavailable for any reason.

Mike

ascillato2, 2019-10-05:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-05 by ascillato2

joba-1, 2019-10-05:
some more thoughts about this:

@salihys : 
1) I found the time needs to be when sun is at zenith. 
That is a very precise point in time. You can mathematically prove, that if you have an infinite number of points like those on the path of the sun, the likelihood you will hit exactly that one point on the zenith is zero. So there is probably some allowance +/-. What interval around noon is acceptable for the reminder? Or if you dont know: what do non deaf muslims use?

2) How precise is the time on your device, how precise can it be set without internet?

@ tasmota experts: 
1) what is the precision of tasmotas sunrise/sunset calculation used? It is usually +/- some minutes.

2) I think @s-hadinger 's rule should do it with enough precision: (sunrise - sunset) / 2
Could you show an example rule (I still dont grok them...)

arendst, 2019-10-05:
> what do non deaf muslims use

Church bells?

jziolkowski, 2019-10-05:
More likely the mu'azzin call

salihys, 2019-10-05:
They hear the call for prayer "athan"

meingraham, 2019-10-05:
Is it possible to perform %sunset%-%sunrise% arithmetic in a rule?

salihys, 2019-10-06:
Good news, I was able to do it through node-red by utilizing the Bigtimer node. The node has all the timers that represents the call for prayer times. Thanks to you all.