arendst/Tasmota #6567: SetOption80 1 -> Shutter command ignored. Not working (for me)

Coroglio, 2019-10-05:
Hello guys. I’m unable to activate shutter option on Tasmota fresh flashed.
I’m using a Wemos D1 mini with a 2 channel relays board ad 2 buttons.
I flashed (via Nodemcu win) sonoff-4m.bin. After this, I setup the module “GENERIC (18)” and after reboot I wast set BUTTON1,2 RELAY1,2 on free GPIOs (exactly GPIO number 4-5-15-16). After reboot again in web-console (or via serial terminal too) I set “SetOption80 1”. 
Command it was been surely accepted : via terminal I got the message: 
`RSL: stat/sonoff/RESULT = {"SetOption80":"ON"}`
Now, I reboot another time but slider no appear, and setting up ShutterMode 0 I got the message
`RSL: stat/sonoff/RESULT = {"Command":"Unknown"}` 
Please note that I got the same result using sonoff-sensors.bin, sonoff.bin and/or selecting Sonoff Dual R2
Please could you explain me where is my error ?
Thanks in advance

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: sonoff-4m.bin (version 6.0.15 6997626-sonoff,  2019-10-04T18:08:55, core 2.5.5)
  - [x] Pre-compiled
  - [] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: NODEMCU (win-1.05)
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ` 11:15:12 CMD: Backlog Template; Module; GPIO
11:15:12 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:15:12 RSL: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
11:15:13 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"18":"Button2"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"22":"Relay2"}}
 `
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
 
STATUS 0 output here:
```
11:26:42.868 -> 00:00:00 CFG: Loaded from flash at 3F6, Count 22
11:26:42.868 -> 00:00:00 Project sonoff Sonoff Version 6.6.0.15(6997626-sonoff)-2_5_2
11:26:43.196 -> 00:00:00 WIF: Connecting to AP1 *-*-*-*-*-* in mode 11N as sonoff-2674...
11:26:49.703 -> 00:00:06 WIF: Connected
11:26:49.703 -> 00:00:06 HTP: Web server active on sonoff-2674 with IP address 192.168.1.79
11:27:08.459 -> 00:00:25 CMD: ShutterMode 0
11:27:08.459 -> 00:00:25 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
11:31:42.229 -> 10:31:42 RSL: tele/sonoff/STATE = {"Time":"2019-10-05T10:31:42","Uptime":"0T00:04:59","UptimeSec":299,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*-*-*-*-*-*","BSSId":"E0:B9:E5:D7:D8:49","Channel":11,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:06"}}
11:36:42.503 -> 10:36:42 RSL: tele/sonoff/STATE = {"Time":"2019-10-05T10:36:42","Uptime":"0T00:09:59","UptimeSec":599,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*-*-*-*-*-*","BSSId":"E0:B9:E5:D7:D8:49","Channel":11,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:06"}}
11:36:43.477 -> 10:36:43 APP: Serial logging disabled
12:15:12.156 -> 11:15:12 CMD: Backlog Template; Module; GPIO
12:15:12.428 -> 11:15:12 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:15:12.710 -> 11:15:12 RSL: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
12:15:12.991 -> 11:15:13 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"18":"Button2"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"22":"Relay2"}}
12:15:50.949 -> 11:15:50 CMD: Backlog Template; Module; GPIO
12:15:51.194 -> 11:15:51 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:15:51.475 -> 11:15:51 RSL: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
12:15:51.814 -> 11:15:51 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"17":"Button1"},"GPIO5":{"18":"Button2"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"22":"Relay2"}}
12:16:43.991 -> 11:16:44 RSL: tele/sonoff/STATE = {"Time":"2019-10-05T11:16:44","Uptime":"0T00:50:01","UptimeSec":3001,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"*-*-*-*-*-*","BSSId":"??:??:??:??:??","Channel":11,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:06"}}
12:18:05.472 -> 11:18:05 CMD: Status 0
12:18:05.472 -> 11:18:05 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:18:05.506 -> 11:18:05 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:51:22","StartupUTC":"2019-10-05T09:26:43","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":23,"SaveAddress":"3F5000"}}
12:18:05.545 -> 11:18:05 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(6997626-sonoff)","BuildDateTime":"2019-10-04T18:08:55","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
[... 45 lines cut ...]
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-2674 with IP address 192.168.1.??
11:25:46 NTP: Drift 0, (UTC) Sat Oct 05 10:25:46 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:25:46 HTP: Main Menu
11:25:49 APP: Boot Count 13
11:25:49 CFG: Saved to flash at 3FA, Count 26, Bytes 4096
11:25:57 HTP: Console
11:26:02 CMD: ShutterMode 0
11:26:02 SRC: WebConsole from 192.168.1.18
11:26:02 CMD: Group 0, Index 1, Command "SHUTTERMODE", Data "0"
11:26:02 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
11:26:05 WIF: Checking connection...
11:26:05 WIF: Connected  
```

meingraham, 2019-10-05:
@Coroglio 

The feature is disabled by default. You will have to compile your own firmware to use the Shutter feature.

Add the following compiler directives to `user_config_override.h`:
```
#ifndef USE_SHUTTER
#define USE_SHUTTER      //  (+6k code)
#endif
```
It is a simple process to compile. Use the online [Gitpod IDE](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) (you do not have to install any software).

This feature was just recently ported to the Tasmota main branch. I apologize, but I missed this and did not have it in the documentation. Thank you for bringing this to my attention.

Regards.

Mike

ascillato2, 2019-10-05:
Closing this issue as it has been answered.

If you need further assistant on compiling, please find us in the Tasmota support chat. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-05 by ascillato2

Coroglio, 2019-10-05:
thanks a lot !

L.

Inviato da Posta per Windows 10

Da: Michael Ingraham
Inviato: sabato 5 ottobre 2019 13:31
A: arendst/Sonoff-Tasmota
Cc: Coroglio; Mention
Oggetto: Re: [arendst/Sonoff-Tasmota] SetOption80 1 -> Shutter commandignored. Not working (for me) (#6567)

@Coroglio
The feature is disabled by default. You will have to compile your own firmware to use the Shutter feature.
Add the following compiler directives to user_config_override.h:
#ifndef USE_SHUTTER
#define USE_SHUTTER      //  (+6k code)
#endif
It is a simple process to compile. Use the online Gitpod IDE (you do not have to install any software).
This feature was just recently ported to the Tasmota main branch. I apologize, but I missed this and did not have it in the documentation. Thank you for bringing this to my attention.
Regards.
Mike

pirx7, 2020-11-16:
Unfortunately here 
         https://github.com/arendst/Tasmota/blob/development/BUILDS.md1
it is said that "USE_SHUTTER" is active inside normal tasmota.bin.
I lost a lot of time to find out the reason why the command ShutterMode2 was not working: 
	CMD: ShutterMode2
	MQT: stat/Shutter/RESULT = {"Command":"Unknown"}

although
	ShutterRelay1 1
	MQT: stat/Enten-Shutter/RESULT = {"ShutterRelay1":1}
seems working.

Please change the misleading document.
Thanks