arendst/Tasmota #6568: Shutter Position MQTT on Switch use

onkeldittmayer, 2019-10-05:
Hi Theo,
i'm verry interressted in the Shutter funktion from StefanBode and your integration in your code works fine, but i can't find a good solution to transmit the Shutter Position via MQTT to IoBroker (Sonoff Module) when using the HardwareSwitch or WebButtons.

When i use the Slider, i got an Event in Console:
14:39:59 MQT: stat/sonoff/RESULT = {"ShutterPosition1":7}

this Massage is missing when manually using the switches or webButtons.

[image]

TelePeriode also another Shutter1_Position DP
But to get a "fresh" Position value, i have to set Teleperiod to 10 and generate unnecessary IO.

it would be a dream to get the result in IoBroker, when using the switches.
And it would be perfekt to get the direction via MQTT when the shutter is moving.

ubi1234, 2019-10-09:
I have 6.6.015 and have as well Shutter position with Shutter1  in this sample 96. as the direction.
**### Use web buttons**

22:05:39 SHT: Start shutter in direction 1
22:05:39 MQT: stat/RolloEZ/RESULT = {"POWER1":"ON"}
22:05:39 MQT: stat/RolloEZ/POWER1 = ON
22:05:39 MQT: stat/RolloEZ/RESULT = {"ShutterPosition1":96}
22:05:40 SHT: Shutter 0: Real Pos: 1400, Target 1840, source: Shutter, start-pos: 0 %, direction: 1, rtcshutter: 0.7  [s]
22:05:40 MQT: stat/RolloEZ/SHUTTER1 = 96
22:05:40 MQT: stat/RolloEZ/RESULT = {"POWER1":"OFF"}
22:05:40 MQT: stat/RolloEZ/POWER1 = OFF
22:05:40 MQT: tele/RolloEZ/RESULT = {"Shutter1":{"Position":96,"direction":0}}

  **### press button on switch.**
22:13:12 MQT: stat/RolloEZ/RESULT = {"POWER1":"ON"}
22:13:12 MQT: stat/RolloEZ/POWER1 = ON
22:13:12 SHT: Shutter 0: Real Pos: 1900, Target 46000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 0.0  [s]
22:13:13 SHT: Shutter 0: Real Pos: 3600, Target 46000, source: Shutter, start-pos: 4 %, direction: 1, rtcshutter: 0.9  [s]
22:13:14 MQT: stat/RolloEZ/RESULT = {"POWER1":"OFF"}
22:13:14 MQT: stat/RolloEZ/POWER1 = OFF
22:13:14 MQT: stat/RolloEZ/SHUTTER1 = 90
22:13:14 MQT: tele/RolloEZ/RESULT = {"Shutter1":{"Position":90,"direction":0}}

ascillato2, 2019-10-09:
Please, update to latest Tasmota version. Thanks. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-09 by ascillato2

onkeldittmayer, 2019-10-10:
@ubi1234 what MQTT server do you use? And how do you get the stat/RolloEZ/SHUTTER1 Value from Tasmota in MQT Server?
As you can see the MQT: stat/RolloEZ/RESULT = {"ShutterPosition1":96} is missing on your "press button on switch".

thanks for your support

ascillato, 2019-10-10:
Hi, for general usage questions, please refer to the Tasmota wiki and to the Tasmota Support Chat. Github issues are for Tasmota software bug reporting. Thanks.