arendst/Tasmota #6573: CurrentLow CurrentHigh :  message that says that the threshold was met is formatted wrong

Pat-78, 2019-10-06:
### BUG DESCRIPTION
I open this issue after chat with meingraham

I set CurrentLow and CurrentHigh to a value : it is ok
The message that says that the threshold was met is bad formatted : 
Two exemples : 
MQT: SHP6_2/tele/MARGINS = {ON"CurrentHigh":""}
MQT: SHP6_2/tele/MARGINS = {ON"CurrentLow":",",OFF"CurrentHigh":""}
So I can't process as usual them with Node-Red.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): BW-SHP6
- [x ] Tasmota binary firmware version number used: 6.6.0.15
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: OTA upgrade from 6.6.0
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  10:49:33 MQT: SHP6_2/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
10:49:33 MQT: SHP6_2/stat/RESULT = {"Module":{"45":"BlitzWolf SHP"}}
10:49:33 MQT: SHP6_2/stat/RESULT = {"GPIO1":{"0":"None"},"GPIO3":{"0":"None"}}
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
 11:00:31 MQT: SHP6_2/stat/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"SHP6_2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:00:31 MQT: SHP6_2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T14:05:28","StartupUTC":"2019-10-05T18:55:03","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":103,"SaveAddress":"F9000"}}
11:00:31 MQT: SHP6_2/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(6997626-sonoff)","BuildDateTime":"2019-10-04T18:03:10","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:00:31 MQT: SHP6_2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxxxx",""],"TelePeriod":0,"Resolution":"558780C0","SetOption":["00008009","280500000100060000005AFF000000000000","00000000"]}}
11:00:31 MQT: SHP6_2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","243683A0","22B617CD","01001BC0","00000081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
11:00:31 MQT: SHP6_2/stat/STATUS5 = {"StatusNET":{"Hostname":"SHP6_2-4893","IPAddress":"xxx.xxx.xxx.xxx","Gateway":"xxx.xxx.xxx.xxx","Subnetmask":"xxx.xxx.xxx.xxx","DNSServer":"xxx.xxx.xxx.xxx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4}}
11:00:31 MQT: SHP6_2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx.xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_34331D","MqttUser":"DVES_USER","MqttCount":35,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:00:31 MQT: SHP6_2/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 06 09:00:31 2019","Local":"Sun Oct 06 11:00:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"07:56","Sunset":"19:20"}}
11:00:31 MQT: SHP6_2/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":45,"PowerHigh":150,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":230,"CurrentHigh":235}}
11:00:31 MQT: SHP6_2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-06T11:00:31","ENERGY":{"TotalStartTime":"2019-09-26T10:00:26","Total":14.630,"Yesterday":2.872,"Today":1.243,"Power":213,"ApparentPower":377,"ReactivePower":311,"Factor":0.56,"Voltage":246.700,"Current":1.529}}}
11:00:31 MQT: SHP6_2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-06T11:00:31","Uptime":"0T14:05:28","UptimeSec":50728,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":35,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxx","BSSId":"xx:xxx:xxx:xx:xx:xx","Channel":6,"RSSI":52,"LinkCount":35,"Downtime":"0T00:03:41"}}}
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  10:59:35 MQT: SHP6_2/tele/MARGINS = {ON"CurrentHigh":""}
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
a welle formatted message as : 
/tele/MARGINS = {"CurrentHigh":"ON"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-10-06:
Thx for testing. Verified. Solving...

arendst, 2019-10-06:
Solved.

Pat-78, 2019-10-06:
It's ok for me
Thx

Jason2866, 2019-10-06:
@Pat-78  Please close issue. Thx!

closed 2019-10-06 by Pat-78