arendst/Tasmota #6574: tasmota 6.6 on tuya DS102 1 Gang

recu, 2019-10-06:
Hi Team

I successfully flashed tasmota on several sonoff devices, and a ESP8266 module. I adquiered a  tuya DS102 1 Gang device in order to evaluate, flashed tasmota with tuya-convert, upgraded to sonoff.bin version 6.6.0 firmware, and configured with the DS102 1 Gang template. I am able to switch On/Off successfully from the module web server, and by mqtt from domoticz, but... the basic functionality is not working, when pressing the switch button on the device, nothing happens, no relay noise, no led on/off/flashing.
The template is the one from the repository: {"NAME":"DS-102 1 Gang","GPIO":[57,0,0,17,0,0,0,0,0,21,56,0,0],"FLAG":1,"BASE":18}

Jason2866, 2019-10-06:
Please try latest development version. There are MANY changes since the version 6.6.0

recu, 2019-10-06:
Thanks Jason2866 
Just tried. Same behavior

Jason2866, 2019-10-06:
@jziolkowski  I think you use this switch? Can you help?

recu, 2019-10-06:
Hi again, I solved by myself.. but don't know how. trying to guess the template, i test several configurations and the device became unresponsive, Relay was on/off every second. assumed the device was on access point mode, then and connected. Configured again the SSID and password, and connected as infrastructure mode, applied the same template and... voila!!! now is working. Thanks

closed 2019-10-06 by ascillato2

M-Ahmad-Naeem, 2020-10-14:
> The template is the one from the repository: {"NAME":"DS-102 1 Gang","GPIO":[57,0,0,17,0,0,0,0,0,21,56,0,0],"FLAG":1,"BASE":18}

@recu 
Hi!
I have also Tasmotized I Gand Switch, both functionalities are working fine.
Thanks for sharing the template.
I am now trying to Hack WiFi Smart Socket. Any sugesstions?

recu, 2020-10-14:
Hi

I have no experience on WiFi mart Socket, sorry I cannot help you.

Regards