arendst/Tasmota #6577: Rule for analog temperature triggering before defined temperature is met

no1knows, 2019-10-06:
When creating a rule to trigger on the Analog Temperature hitting a certain value, the rule triggers before the value is met.

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): D1 Mini
- [ ] Tasmota binary firmware version number used: 6.6.0
  - [ ] Pre-compiled - yes
- [ ] Flashing tools used: NodeMCU PyFlasher
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  Configuration output here:
`17:11:04 MQT: sonoff-guestufhtemp/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:11:04 MQT: sonoff-guestufhtemp/stat/RESULT = {"Module":"18 (Generic)"}
17:11:05 MQT: sonoff-guestufhtemp/stat/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"21 (Relay1)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
17:11:05 MQT: sonoff-guestufhtemp/tele/HASS_STATE = {"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Generic","RestartReason":"Power on","Uptime":"0T07:35:01","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":13,"SaveCount":41,"IPAddress":"192.168.1.229","RSSI":"80","LoadAvg":23}`

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  `17:11:35 MQT: sonoff-guestufhtemp/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":396,"Rules":"on Analog#Temperature>45.0 DO Backlog Power OFF; publish notify-emergency Guest_UFH; RuleTimer1 900; Rule1 0  ENDON"}
17:11:36 MQT: sonoff-guestufhtemp/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":478,"Rules":"ON Rules#Timer=1 DO Rule1 1 ENDON"}
17:11:36 MQT: sonoff-guestufhtemp/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}`
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  `17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Guest UFH Temp"],"Topic":"sonoff-guestufhtemp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T07:36:05","StartupUTC":"2019-10-06T08:36:04","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":41,"SaveAddress":"FB000"}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["27SR",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["000A8009","280500000100060000000000000000000000","00000200"]}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS5 = {"StatusNET":{"Hostname":"GuestUFHTemp","IPAddress":"192.168.1.229","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"4C:11:AE:0D:83:B5","Webserver":2,"WifiConfig":4}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.99","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0D83B5","MqttUser":"no1knows","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 06 16:12:09 2019","Local":"Sun Oct 06 17:12:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:56","Sunset":"18:20"}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-06T17:12:09","ANALOG":{"Temperature":39.8},"TempUnit":"C"}}
17:12:09 MQT: sonoff-guestufhtemp/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-06T17:12:09","Uptime":"0T07:36:05","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"POWER":"OFF","Wifi":{"AP":1,"SSId":"27SR","BSSId":"60:E3:27:DE:AA:94","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:06"}}}`
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
> `17:05:07 CMD: Rule1 on Analog#Temperature>40.0 DO Backlog Power OFF; publish notify-emergency Family_UFH; RuleTimer1 900; Rule1 0  ENDON
17:05:07 MQT: sonoff-familyufhtemp/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":395,"Rules":"on Analog#Temperature>40.0 DO Backlog Power OFF; publish notify-emergency Family_UFH; RuleTimer1 900; Rule1 0  ENDON"}
17:05:20 RUL: ANALOG#TEMPERATURE>40.0 performs "Backlog Power OFF; publish notify-emergency Family_UFH; RuleTimer1 900; Rule1 0"
17:05:21 MQT: sonoff-familyufhtemp/tele/STATE = {"Time":"2019-10-06T17:05:21","Uptime":"0T07:40:40","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"27SR","BSSId":"9C:3D:CF:E0:6E:E6","Channel":11,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:05:21 MQT: sonoff-familyufhtemp/stat/RESULT = {"POWER":"OFF"}
17:05:21 MQT: sonoff-familyufhtemp/stat/POWER = OFF
17:05:21 MQT: notify-emergency = Family_UFH`

  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-10-06:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-06 by ascillato2

no1knows, 2019-10-06:
This is clearly a bug. The Rule is supposed to trigger above 40.0 but it's triggering at 39.2.

ascillato2, 2019-10-06:
The tele message that you see is sent at teleperiod interval. So, the temperature that you see is Before the trigger of the rule. Your rule is triggering ok at > 40

no1knows, 2019-10-06:
The temperature never gets to 40 though, before or after the rule is triggered. Would you suggest that it's jumping above 40 briefly and then back down?

ascillato, 2019-10-06:
In the teleperiod message you see just the "instant value" of that moment. A photo of the value of temperature at that particular moment (the teleperiod). For the rule, it is constantly seeing the value. Also it is seeing the smalls jumps (or noise) in the measurement. So, yes, your sensor is sensing >40 (at least for an instant) and it is triggering your rule.

meingraham, 2019-10-06:
As @ascillato mentioned, `Analog#Temperature` reached the trigger value. If you want to check out the difference, user `tele-Analog#Temperature`. This is the value when TelePeriod occurs.

no1knows, 2019-10-06:
Righto, will do. Thank you both.